project-chip/connectedhomeip #33444: Content Launcer feature map shows up as indexes instead of bitmask

andy31415, 2024-05-14:
Definition of

```xml
<bitmap name="Feature" type="bitmap32">
     <cluster code="0x050a"/>
     <field name="ContentSearch" mask="0x1"/>
     <field name="URLPlayback" mask="0x2"/>
     <field name="AdvancedSeek" mask="0x3"/>
     <field name="TextTracks" mask="0x4"/>
     <field name="AudioTracks" mask="0x5"/>
   </bitmap>
   ```
   
   is wrong because masks should be `0x01, 0x02, 0x04, 0x08, 0x10, ...` instead of using indexes. This prevented us from migrating to feature bits for these features.
   
   If we use these, there may be sideffects. Need to update the data type and validate if there are any sideffects in code (that thinks these are indexes instead of bitmasks).

bzbarsky-apple, 2024-05-14:
Looks like the buggy XML was added in https://github.com/project-chip/connectedhomeip/pull/30691 @lazarkov @hawk248 @Brian-Resideo

chrisdecenzo, 2024-05-15:
These are not actually used in the code today, but this should be fixed to prevent future issues in case developers reference the output from these.

bzbarsky-apple, 2024-05-15:
> These are not actually used in the code today

They're shipped in the Android and Python APIs, no?  (And Darwin, but we made sure to not expose this to API consumers so far....)

brdandu, 2024-06-03:
We should also look into ZAP have an upgrade mechanism where things like this can be fixed. The latest ZAP can have upgrade rules to fix old .zap files opened with latest GSDK.