project-chip/connectedhomeip #33464: Solitary Ack error loop when controller abandons BDX log download

ksperling-apple, 2024-05-15:
I have observed a number of instances where abandonment of a BDX log download by the controller causes the device app to go into a "loop" of attempting to send a solitary ACK. I don't think it reproduces 100% of the time though.

[bdx-shutdown-server-loop.txt](https://github.com/project-chip/connectedhomeip/files/15315602/bdx-shutdown-server-loop.txt)

bzbarsky-apple, 2024-05-15:
So this is a case where the _session_ is gone for some reason?

This looks a _lot_ like https://github.com/project-chip/connectedhomeip/issues/16805.... but maybe now we have a clearer way to see what's going on there!

ksperling-apple, 2024-05-15:
Hm yeah maybe it's the same issue. This happened when I stopped the controller side, does that close the session somehow?

bzbarsky-apple, 2024-05-15:
I would not expect it to, generally.....