project-chip/connectedhomeip #33467: One problem about the definition of Vendor ID in Matter Specification Version 1.3 documentation.

whqxbs, 2024-05-15:
### Build issue(s)

### Documentation issues

**Page 828 in Matter Specification Version 1.3 documentation has the definition in the figure for `AdminVendorID Field`:**
[image]
**I would like to know what is mentioned here "This value SHALL NOT be one of the reserved Vendor ID values defined in Table 1, 'Vendor ID Allocations '."** 

- **How to understand it?** 
- **And why can't I use the Vendor ID in Table 1?**

**Page 53 in Matter Specification Version 1.3 documentation has the Table 1, and all Vendor IDs in Table 1 are "reserved" as described in the documentation:**
[image]

- **I would like to know what happens if the Vendor ID from Table 1 is used in the AdminVendorID Field?**

Thank you very much for your answering!

whqxbs, 2024-05-15:
I used the CHIP Tool's `--commissioner-vendor-id` parameter for the test of Vendor ID definition and I can specify any value for the Vendor ID. So I don't understand what the documentation means by "SHALL NOT be one of the reserved Vendor ID values defined in Table 1".

bzbarsky-apple, 2024-05-15:
I think the intent is that it can't be 0 or one of the test values; it has to be an actual vendor id identifying the commissioner....   But either way, spec issues need to go in the spec repository (unfortunately not public).

I filed https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/9212 to track this; thank you for raising it!

closed 2024-05-15 by bzbarsky-apple

whqxbs, 2024-05-16:
Thank you! If there is any progress, I would appreciate it if you could reply here with an explanation of the documentation because of the not public spec issues. :)

bzbarsky-apple, 2024-05-16:
I'll try, but I am not following the spec closely...