project-chip/connectedhomeip #33475: [Test Failed] Python tests write acl with "fabricIndex=0" field in Write request

Apollon77, 2024-05-15:
### Test issue(s)

While executing python test ACE_1_2.py I found that the acl entries written by Python contained an fabricIndex field with value 0. see https://github.com/project-chip/matter.js/actions/runs/9092348098/job/24988983271#step:4:1080

> [{"privilege":5,"authMode":2,"subjects":["112233","112234"],"targets":[{"cluster":null,"endpoint":0,"deviceType":null}],"fabricIndex":0}]

The value is really encoded in the Tlv data.

According to specifications 7.13.7. FabricIndex Field:

> This field SHALL NOT be present in a write interaction. For a write interaction, the server SHALL provide the value of the accessing fabric-index as the FabricIndex field value to processing logic, after receipt of the interaction. 

I do not know if this is a python test issue or a more general issue.

### Platform

all

bzbarsky-apple, 2024-05-15:
Under "7.19.1.9. Struct", in the "Fabric-Scoped Struct" section, the text says:

> The global FabricIndex field of a fabric-scoped struct SHOULD NOT be indicated in a write interaction.

> The global FabricIndex field of a fabric-scoped struct SHALL be ignored in a write interaction.

It looks like this has been inconsistent all along, actually.  The intent of the spec is what 7.19.1.9 says, so 7.13.7 needs to be fixed.  I filed https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/9213 to track this, but this is not an SDK issue so closing this.

Apollon77, 2024-05-22:
Thank you @bzbarsky-apple ... ok in fact I add logic then in matter.,js to always overwrite the fabricIndex, irrelevant from what comes in