project-chip/connectedhomeip #33484: [1.3] chip-tool buil fail

KEEPHEADU, 2024-05-16:
### Reproduction steps

source scripts/activate.sh 

./scripts/examples/gn_build_example.sh examples/chip-tool out/host/

This error will occur if you build any project.

```
ERROR at //.gn:15:1: Unable to load "/home/xh/silabs/matter/examples/chip-tool/build_overrides/build.gni".
import("//build_overrides/build.gni")

```

### Bug prevalence

any time

### GitHub hash of the SDK that was being used

no

### Platform

efr32

### Type

Test Improvement

### Anything else?

[log.txt](https://github.com/project-chip/connectedhomeip/files/15331879/log.txt)

bzbarsky-apple, 2024-05-16:
Well, does the file `/home/xh/silabs/matter/examples/chip-tool/build_overrides/build.gni` exist?

In general, `examples/chip-tool/build_overrides/build.gni` does exist in the Matter repository (see https://github.com/project-chip/connectedhomeip/blob/22b9c96b691d3bb5a5d3db510db353b3b2bc2f74/examples/chip-tool/build_overrides and https://github.com/project-chip/connectedhomeip/blob/22b9c96b691d3bb5a5d3db510db353b3b2bc2f74/examples/build_overrides/build.gni) , so if it doesn't exist for you the first step is figuring out why not.

KEEPHEADU, 2024-05-17:
> Well, does the file `/home/xh/silabs/matter/examples/chip-tool/build_overrides/build.gni` exist?
>
> In general, `examples/chip-tool/build_overrides/build.gni` does exist in the Matter repository (see [...]

Thank you for your reply. This question is due to the failure of my first construction, I deleted the content in `Examples/Chip-Tool/Build_overrides`. After I revoke the operation, I can build it normally.

closed 2024-05-17 by KEEPHEADU