project-chip/connectedhomeip #33487: [CERT-TEST-FAILURE] Test_TC_ACL_2_3 implicitely requires chip chunked array writes

Apollon77, 2024-05-16:
### Feature Area

Other

### Test Case

Test_TC_ACL_2_3

### Reproduction steps

Formally the last successful write is in "step 9" of the test ... but Step 18 expects another value. 

It seems Step 18 expects the "first of the second entries" from step 17 ... which gave an ConstraintError because tried to write 2 entries while only one is allowed.

I assume that this is because of the chunked write handling added in https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/access-control-server/access-control-server.cpp#L353 which is very chip specific because formally there is no requiremenst that I know that any array needs to be transported in a chunked way ... and also not that writes needs to be handled chunked at all.

In fat it is really unexpected in case of acl (and especially in case of ACL where consistence should be the most important topic) that a write command like in step 17 returns an error, BUT had an effect by writing the first of the two chunked parts ...

It would be great to get some background information for this, thank you!

In general the test will also have issues as soon as a controller implementation is not sending the list "always chunked" as chip does, because the logic in the access-control-server.cpp would return an Constrainerror without saving any value when it is >1 entry in the array. So this is then not consistent.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

master

### Platform

core

cecille, 2024-05-16:
@mlepage-google @hasty

Apollon77, 2024-05-16:
Ok the more I look into it the more it becomes clear to me that the expectation with chunked writes is to really process them also in their chunks one by one. Matter.js is currently not doing this that way. I can change that.

With this the question (that is unanswered for me also after consulting specs - or I am too blind) I added above is the one left: If on one chunk an error happens are more chunks for the same path also processed or not? I assume yes. In fact the result would be then one response per chunk, e.g. one ok, one error and one ok ... and it is up to the caller to decide whats now up ;-) Correct?

Thank you for your support.

Apollon77, 2024-05-16:
Another follow up question: how exactly data version protected writes work with chinked writes? On acl cluster the specs tell the administrator to ideally protect writes with data version filters. So… when each chunked part is handled as own write then dataversion changes after each. So after first write that will change unless the controller makes assumptions on how be re ion increases. Is that also correct? So data version protected writes should never be chunked ?!

mlepage-google, 2024-05-16:
Hello Appollon.

Some clarifications...

First, the **extension** attribute is _optional_ and if present, must be of length 1. That is where there is a constraint error when attempting to write a second element to that list.

The **ACL** attribute is _mandatory_, and must support at least 3 entries per fabric. Therefore, writing 3 entries will never result in a constraint error due to exhausting list length; writing 4 or more entries might (depending on actual supported length) result in such errors, but they would always be at the end, and never have a successful entry after exhausting the supported length.

An ACL entry may fail other data validity checks, and fail to write for that reason.

In the SDK, when a list is written (say with 3 elements), it is done as follows:

1. List cleared via list **replace** (with empty list) action.
2. First element added via list item **append** action.
3. Second element added via list item **append** action.
4. Third element added via list item **append** action.

In that case, if the second item resulted in a failure (e.g. due to invalid item data), the list would still be of length 1 after having attempted to append the second item. Processing would continue, and the third item would be appended, resulting in a list of length 2. It is as if only the two valid items were sent, except that a write error occurred for the second item.

Apollon77, 2024-05-16:
All right. Thank you for confirmation. In fact this is then also no testing issue and could be closed or moved. 

Nevertheless the specification lack details on chunked write operations and how this should work. It was completely unclear to me till now.

 In my eyes this behavior is maximum inconsistent because if I send an unchunked array with the 3 items where second is invalid then nothing is written and the constraint error happens for all three. But unassigned that this ship has sailed ;-)

Would you also have an info how chunked writes work with dataversion check on the write? (Also because acl specs propose to use that). In my understanding each processed chunk increase the version so the data version check only works for the first chunk. 
I know that chip always send array full and so resets the array with the first chunk but the protocol does not dictate that. So someone could really just add items to an existing list. ;-)

Apollon77, 2024-05-16:
Ps: (and slightly off topic) is there a reason why Chip always chunks the data in data reports and writes? If the array would fit into the data report then sending it unchanged would need less bytes then always chunked. 
Would that not be important for low power devices? 

But yes then the test here would get issues because they could behave differently ;-))

Apollon77, 2024-05-16:
Re your infos on dataversion: ok thank you for the reference. Puuhh that makes things interesting. 

So we have kind of a „transaction“ (summarize multiple changes and „commit“ at the end to increase version) but without having a real „transaction“ (aka rollback everything in case of an error). Good. Will be fun to implement that. ;-)

Apollon77, 2024-05-16:
When I think deeper then I also ask myself why the spec 1.3 in "10.6.4.3.1. Lists" just tals about "full list" or "clear the list and send appends" ... Combined with a dataVersion security approach it should be easiely possible to just send list modifications. Also the spec examples kind of contain all these examples, just the spec tells is "SHOULD" not be done ... 
This should be even less bytes needed and should be (because of the dataVersion security guard) end in the same final list ... or?! But yes thats more "spec and protocol feedback" and I have no idea if chip has implemented that ... I assume not becuse eg acl just handled additions beside "full array" and last time I checked in the c++ code I only saw places that handle these two cases, nothing else... Sorry to place it here, but I have no other ways to bring such thoughts/questions to any attention ;-)

mlepage-google, 2024-05-16:
> … as example in acl code is there for both options but the „full array“ code path should always ever be executed with zero or one element …

In access-control-server.cpp for writing the Extension attribute with an entire list, it only ever handles a list of length 0 or 1 because that attribute list can only ever be length 0 or 1.

https://github.com/project-chip/connectedhomeip/blob/22b9c96b691d3bb5a5d3db510db353b3b2bc2f74/src/app/clusters/access-control-server/access-control-server.cpp#L350

For the ACL attribute, the length of the list being written is not assumed.

https://github.com/project-chip/connectedhomeip/blob/22b9c96b691d3bb5a5d3db510db353b3b2bc2f74/src/app/clusters/access-control-server/access-control-server.cpp#L252

Apollon77, 2024-05-17:
I did most adjustments needed and are currently finalizing all the tests but I need to tell you that the tests here (especially ACE 2_5, 2_6 and such) are veeery chip implementation specific. I now need to mimic the event order based on the code how chip is managing the entries in the relevant cases and send out events. Also the events have the "latestValue" which is not really described in the specs, but in the end is a different value depending on if it is a Removal or other action. All this is nowhere mentioned in the specification. In the end thats all valid and ok to require it, but if such things like the exact event behaviour, orders and such are checked in certification in this depth then also the specification should contain these implementation details as requirements. Thats at least my opinion :-)

In fact I spent at least 4-8h today to just run tests, analyze why it broke, adjust code, and do this in loops ;) Same for the specific write request handling for "splitted/chunked list processing". The specification (at least not clear to me) is not containing all the details whats requitred and tested. I think all details in specification would have saved me a lot of this time.

Please see this as feedback from a developer that implements the protocol mainly after specs (as it should work out).

Apollon77, 2024-06-06:
In addition to above I also see a spec issue here ...

Spec tells in 6.6.4. Access Control Cluster update side-effects

> Updates to the Access Control Cluster SHALL take immediate effect in the Access Control system.

and then having the example of "1. Action: Write (1st path)" ... but with chunked writing the first write path action is the complete reset of the array to be empty! so the first real value gets added in the _second_ Write action. So if this would become effective really directly then all other acl actions will indeed fail because the acl list is empty.

So it would be great to know how this should be really handled with all chunked writes. Does ACL list changes really become effective directly, but "ignoring" the initial write action to clear the list? Or is the new entry just effective after all writes are done? In both cases specification would require an update in my pov.

Also in general this "proceed the writes entries one by one" also could have effects in e.g. bindings or groups. In both cases lists are used to handle things. If we also here clear the lists and then "write one by one with immediately becoming effective" means that if I have 5 groups/bindings and I add a sixths then the first write clears the list and then the first entry gets added and formally all others removed. if this becomes effective directly too that would mean we stop listening to all these groups or disconnect from all bound devices except the first and then we rebuild all these connections and listening sockets again newly and add one more. or am I wrong or miss a point?

Apollon77, 2024-06-07:
Ok, I verified one of the things by experimenting with chip:
It seems to me that a "write to the acl attribute" is processed completely in the write message and becomes effective then. This is the reason that clearing the array as first write path action or adding acl entries in strange order is now screwing up the whole acl system. So with this assumption is that chunked list writes are processed as a sequence in chip until the "write path" (regarding endpoint/cluster/attribute) changes and are then being updated in the acl system. At least so it seems to me.

cjandhyala, 2024-08-19:
@Apollon77 can we close this issue ?

Apollon77, 2024-08-30:
In my eyes not tbh. Because we were in the middle of a certification we now adjusted matter.js to process list writes the exact same way as chip does by never writing the full list but always "chunking" it. If I would change this back to the original way to write the full list at once then by the defined logic of the "how events are needed to be sent defined by spec" we would have different events triggered.

Example:
* currently: the first event is always an update of the first element, followed by delete events (in backwards order which is also nowhere defined in the specs, but implemented that way by chip) for all other former list entries, then it generates more add-events. 
* with a read full list write there would be update events for all "overlapping" list indices and then add or delete events for the rest depending on if the new list is longer then the old list

As you can see, based on the event logic defined by spec these two "way to process lists" is different and will cause a different order of events by nature as soon as the list is longer then 1 entry. Additionally the fact that add/update cases are processed in ascending order, but delete cases are process in descending order is undefined by the specs, so also the order of delete events itself would be undefined, but the yaml tests are explicitly requiring them to be in "the" order that chip implements.

So yes - because of several "workarounds" and "codeflow like chip" in matter.js - it is formally no cert blocker for matter.js devices, but mainly because we did not wanted to have this discussing within the certification process. But as soon as another implementation implements the same (e.g rust) the same issue could arise again.

Meaning, in my eyes, if the current behavior and tests should stay as is, this would require some spec adjustments to define the chip behavior as "the one". In my eyes this would not be the best idea tbh ...
So other options would be to adjust the tests in a way that relaxes the checks in a way that things work or are ok.

Basically this topic also touches the whole topic that chip choose to send "lists always chunked" also if they would match into one message ...

Apollon77, 2024-09-06:
Ok, I thought a bit on how we could ideally discuss this with a chance to come to solutions and not to run in circles ;-)

In my eyes the very first topic is: If there is an agreement on the fact that normal list write interaction are _always_ done chunked as chip sdk is sending them currently and this stays that way forever that would simplify a lot. So that would be the very first topic to discuss.
* if there is an agreement then this should be simply added to the spec as a "SHALL" requirement and 80% of the topic is solved basically. And formally that's just an "editorial" because noone was able to certify a device without this behaviour because else the above mentioned ACL tests would all have failed. When this is decided it would also be a great idea to reflect this in chip code and just support this case and error if someone sends something different to be clear here.
* if this is no agreement then we have a funny discussion :-)

When the above topic is agreed only one main topic is left: order of delete events in the ACL case which relies on the sdk implementation where deletions are processed from the highest list index down to 0. If this is the way to go for ACL, define it in the specs and we are also done here. Or we should think about how we could enhance the yaml tests to just verify event existence and not the order :-)

(side note: the acl yaml tests rely on the fact that the latest acl events are still readable ... which (as I understood) might not be the case, so also test could fail if too many events are generated, but thats other topic)

Apollon77, 2025-01-27:
Closing in favor of

https://github.com/CHIP-Specifications/chip-test-plans/issues/4810 and
https://github.com/project-chip/connectedhomeip/issues/37204

closed 2025-01-27 by Apollon77