project-chip/connectedhomeip #33512: [Build] AppTask.cpp error 'Custom0taEntries_t' was not declared in this scope

chajk, 2024-05-18:
### Build issue(s)

Hi 
When I tried to build lighting-app for nxp/k32w/k32w0, The following compile errors come up although the same step is tried. 
Can you please help to fix this issue?

https://github.com/NXP/matter/tree/v1.0-branch-nxp/examples/lighting-app/nxp/k32w/k32w0

1) Download [K32W0 SDK 2.6.9](https://cache.nxp.com/lgfiles/bsps/SDK_2_6_9_K32W061DK6.zip).
2) Start building the application either with Secure Element or without

without Secure Element
$ export NXP_K32W0_SDK_ROOT=/home/$user/DK_2_6_9_K32W061DK6/
$ source ./scripts/activate.sh
$ cd examples/lighting-app/nxp/k32w/k32w0
$ gn gen out/debug --args="k32w0_sdk_root=\"${NXP_K32W0_SDK_ROOT}\" chip_with_OM15082=1 chip_with_ot_cli=0 is_debug=false chip_crypto=\"platform\" chip_with_se05x=0 chip_pw_tokenizer_logging=true"
$ ninja -C out/debug

---------------------------------------------------------------------------------------------------------
arrow@ubuntu:~/connectedhomeip/examples/lighting-app/nxp/k32w/k32w0$ ninja -C out/debug/
ninja: Entering directory `out/debug/'
[0/1] Regenerating ninja files
device: K32W061
board: k32w061dk6
ntag: 0
increased TX power: 0
FRO32k:  0
low power:  0
OTA default firmware processor:  1
OTA default factory data processor:  0
PDM Encryption:  1
Antenna Diversity enabled:  0
ECC crypto lib:  NXP-Ultrafast-P256
chip_with_sdk_package: 1
[20/850] c++ obj/main/chip-k32w0x-light-example.AppTask.cpp.o
FAILED: obj/main/chip-k32w0x-light-example.AppTask.cpp.o 
arm-none-eabi-g++ -MMD -MF obj/main/chip-k32w0x-light-example.AppTask.cpp.o.d -Wconversion -march=armv7e-m -mcpu=cortex-m4 -mfloat-abi=soft -mthumb -Os -g2 -fno-common -ffunction-sections -fdata-sections --specs=nosys.specs --specs=nano.specs -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -Wall -Werror -Wstack-usage=8192 -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-unused -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -Wno-unused-function -Wno-conversion -Wno-sign-compare -Wno-clobbered -Wno-implicit-fallthrough -Wno-shadow -mthumb -MMD -MP -fno-optimize-strlen -Wno-error=address -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/mbedtls/port/matter -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/mbedtls/port/ksdk -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/boards/k32w061dk6/wireless_examples/openthread/reed/bm -isystem../../third_party/connectedhomeip/src/platform/nxp/k32w/k32w0 -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/CMSIS/Include -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/components/serial_manager -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/components/uart -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/devices/K32W061 -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/utilities/debug_console/str -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/utilities/debug_console -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/devices/K32W061/utilities -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/devices/K32W061/utilities/debug_console -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/devices/K32W061/utilities/str -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/ble_controller/interface -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/bluetooth/application/common -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/bluetooth/application/common/gatt_db -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/bluetooth/application/common/gatt_db/macros -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/bluetooth/host/config -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/bluetooth/host/interface -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/Common -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/Flash/External/Interface -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/Flash/Internal -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/FunctionLib -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/GPIO -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/Logging/Interface -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/Keyboard/Interface -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/LED/Interface -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/Lists -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/LowPower/Interface/k32w061dk6 -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/MemManager/Interface -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/Messaging/Interface -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/ModuleInfo -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/OSAbstraction/Interface -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/OtaSupport/Interface/ -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/Panic/Interface -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/PDM/Include -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/RNG/Interface -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/SecLib -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/SerialManager/Interface -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/SerialManager/Source -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/TimersManager/Interface -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/TimersManager/Source -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/XCVR/DK6 -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/framework/XCVR/DK6/Build/Include -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/ieee-802.15.4/MacDynamic/Include -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/middleware/wireless/ieee-802.15.4/uMac/Include -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/rtos/amazon-freertos/lib/FreeRTOS/portable/GCC/ARM_CM3 -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/rtos/amazon-freertos/lib/include -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/rtos/amazon-freertos/lib/include/private -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/rtos/amazon-freertos/lib/third_party/unity/src -isystem../../../../../../../../SDK_2_6_9_K32W061DK6/devices/K32W061/drivers -Wno-type-limits -fmacro-prefix-map=../../third_party/connectedhomeip/= -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -DNDEBUG -DgMainThreadPriority_c=5 -DCPU_JN518X -DCPU_JN518X_REV=2 -DJENNIC_CHIP_FAMILY_NAME=_JN518x -DMAC_PROTO_TAG=1 -DJENNIC_CHIP_FAMILY_JN518x -DgPWR_LDOMEM_0_9V_PD=0 -DSDK_DEBUGCONSOLE=DEBUGCONSOLE_REDIRECT_TO_SDK -DPRINTF_ADVANCED_ENABLE -DNO_SYSCORECLK_UPD=0 -DUSE_RTOS=1 -DUSE_SDK_OSA=0 -DgSerialManagerMaxInterfaces_c=2 -DFSL_RTOS_FREE_RTOS=1 -DgTotalHeapSize_c=0xF000 -DDEBUG_SERIAL_INTERFACE_INSTANCE=0 -DAPP_SERIAL_INTERFACE_INSTANCE=1 -DgOTA_externalFlash_d=1 -DgEepromPostedOperations_d=1 -DgOtaEepromPostedOperations_d=1 -DgOtaVerifyWrite_d=0 -DgExternalFlashIsCiphered_d=1 -DPDM_USE_DYNAMIC_MEMORY=1 -DPDM_SAVE_IDLE=1 -DgBootData_None_c=1 -DPROGRAM_PAGE_SZ=256 -DconfigFRTOS_MEMORY_SCHEME=4 -DosCustomStartup=1 -DENABLE_RAM_VECTOR_TABLE=1 -DgTMR_Enabled_d=1 -DgTimestamp_Enabled_d=0 -DCHIP_ENABLE_OPENTHREAD=1 -DgOtaMemPoolId_c=1 -DPoolsDetails_c=_block_size_\ 32\ _number_of_blocks_\ 6\ _pool_id_\(0\)\ _eol_\ _block_size_\ 256\ _number_of_blocks_\ 3\ _pool_id_\(0\)\ _eol_\ _block_size_\ 512\ _number_of_blocks_\ 2\ _pool_id_\(0\)\ _eol_\ _block_size_\ 768\ _number_of_blocks_\ 1\ _pool_id_\(0\)\ _eol_\ _block_size_\ 268\ \ _number_of_blocks_\ 8\ _pool_id_\(gOtaMemPoolId_c\)\ _eol_ -DSUPPORT_FOR_15_4=1 -DgAppMaxConnections_c=1 -DgAppUseBon [... cut ...]
cc1plus: note: unrecognized command-line option '-Wno-unknown-warning-option' may have been intended to silence earlier diagnostics
[25/850] c++ obj/third_party/connectedhomeip/zzz_generate...n/app-common/zap-generated/attributes/zap.Accessors.cpp.o
ninja: build stopped: subcommand failed.
arrow@ubuntu:~/connectedhomeip/examples/lighting-app/nxp/k32w/k32w0$ 
------------------------------------------------------------------------------------------------

### Anything else?

It's also reproducible regardless of Ubuntu version.

bzbarsky-apple, 2024-05-22:
@doru91 can you take a look, please?

doru91, 2024-05-24:
Hi, @chajk 

Please note that SDK 2.6.9 is not the latest SDK.

I would recommend using west for automatically downloading the latest SDK, see detailed instructions here: https://github.com/project-chip/connectedhomeip/tree/master/examples/contact-sensor-app/nxp/k32w/k32w0#building

Thanks,
Doru