project-chip/connectedhomeip #33517: [BUG] chip-tool commissioning ICD device then commissioning non-ICD device will fail

wilbur-se, 2024-05-19:
### Reproduction steps

1.run chip-tool under interactive mode
2.pair a ICD device -> OK
3.then pair a non-ICD device -> FAIL

### Bug prevalence

always

### GitHub hash of the SDK that was being used

matter 1.3

### Platform

core

### Platform Version(s)

all platform that implement commissioner

### Anything else?

From my anaysis, the root cause is no chance to set mNeedIcdRegistration to false (controller/AutoCommissioner.cpp)
When commissioning a ICD device, because ICD cluster exists, mNeedIcdRegistration will be set to true. Following a non-ICD device, mNeedIcdRegistration is still true, then commissioner will go to ICD registration stage. But the regisitration will fail for status UNSUPPORTED_CLUSTER.

One changes to fix maybe:

--- a/src/controller/AutoCommissioner.cpp
+++ b/src/controller/AutoCommissioner.cpp
@@ -752,6 +752,7 @@ CHIP_ERROR AutoCommissioner::CommissioningStepFinished(CHIP_ERROR err, Commissio
 
             if (mParams.GetICDRegistrationStrategy() != ICDRegistrationStrategy::kIgnore)
             {
+                mNeedIcdRegistration = false;
                 if (mDeviceCommissioningInfo.icd.isLIT && mDeviceCommissioningInfo.icd.checkInProtocolSupport)
                 {
                     mNeedIcdRegistration = true;

yunhanw-google, 2024-05-23:
https://github.com/project-chip/connectedhomeip/pull/33574

yunhanw-google, 2024-05-23:
Thanks for reporting this issue, the above is the PR for issue fix.

closed 2024-05-24 by bzbarsky-apple