project-chip/connectedhomeip #33521: [Build] To convert gn, ninja build system to cmake, make build system for Matter Linux platform

gaganpydi, 2024-05-20:
### Build issue(s)

1. My other project uses cmake to generate make files in build process. I want to integrate Matter Linux platform examples (lighting-app) or other to my project, but Matter Linux examples are using gn to generate ninja files. 

### Platform

other

### Anything else?

https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/linux - Directory for build.gn file in lighting-app example.

gerickson, 2024-06-05:
@gaganpydi, I am just getting started with Matter myself and have noted some challenges with the build system as well. Questions you might have encountered that I have encountered:

1. I cannot seem to find a way to specify: "use this C compiler, this C++ compiler, this linker, this archiver, these C flags, these C++ flags, and these linker flags". This is beyond trivial in autotools and straightforward enough in Meson or CMake.
2. I cannot seem to find a way to _stage_ the build such that I get the binary artifacts and the public header files that define the public APIs for those binary artifacts. For example, under <build artifact output>, I see many executables and many *.a's under _lib/_, but not headers to be found.
3. I cannot seem to find a way to specify, "I already have the satisfied dependencies for nlassert, olio, openttread, ot-br-posix, etc. in my project. Build and link against them _here_, _here_, _here_, _here_, and _here_.". This is straightforward enough in CMake, autotools, or Meson.
4. I cannot seem to find a way to specify, "I don't want to use any of the platform definitions contained in the project. I will be supplying these externally. Either trust me and link against weakly-defined stubs or find them _here_." This is somewhat related to (2) in that if there were a `stage-headers` or `install-headers` target / goal, a platform / product integrator could run that first, build his/her platform against that, and then cycle back and build the rest of Matter against that. Building my platform / product integration in the Matter tree is a non-starter.

gerickson, 2024-06-07:
Regarding (3) above, someone pointed me to #31025, which looks like an important answer to the question of how to specify external package dependencies in the _gn_ build of Matter.

It looks like the _gn_ idiom is _<something>_root_ for the root of an external dependency, (for example, _nlio_root_, _nlassert_root_, _lwip_root_). However, it is unclear whether that is to be the source or build artifact root for a dependency. My expectation is the latter. @mspang, any clarification on that front?

mspang, 2024-06-07:
> Regarding (3) above, someone pointed me to #31025, which looks like an important answer to the question of how to specify external package dependencies in the _gn_ build of Matter.
> 
> It looks like the _gn_ idiom is __root_ for the root of an external dependency, (for example, _nlio_root_, _nlassert_root_, _lwip_root_). However, it is unclear whether that is to be the source or build artifact root for a dependency. My expectation is the latter. @mspang, any clarification on that front?

For the most part the expectation is that folks will assemble all of source code and compile it together, or in as few phases as they can manage for their integration, so typically it would either be the source root or the location of a custom build file that compiles the library.

I know this is the not the workflow you're expecting,  but it's the workflow that facilitates the best iteration speed across the dependency graph - eg if you change a single header file in a dependency, downstream sources that include that header will be recompiled rather than recompiling the entire project.

That said, it's not as if GN does not support linking precompiled libraries. This can be done in the build file, it just results in a worse experience for developers.

closed 2024-10-24 by andy31415