project-chip/connectedhomeip #33539: [TC-RR-1.1] INFO Step 14: Setting group key map on fabric 6  Error .

ChengqiangShe, 2024-05-22:
### Test issue(s)

**[TC-RR-1.1] INFO Step 14: Setting group key map on fabric 6  Error .
However, the device configuration has a maximum of 5 fabrics.
The sixth fabrics should not be tested in this test case。**

Part of the test log：

...
`[20240522_11:16:58.591][MatterTest] 05-22 03:16:59.277 INFO <<< [E:38899i S:46231 M:218414525 (Ack:265560162)] (S) Msg TX to 1:0000000012344321 [7E16] [UDP:[fd11:35::f394:a563:db40:73f1%wpan0]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck)
[20240522_11:16:58.595][MatterTest] 05-22 03:16:59.277 INFO Step 14: Setting group key map on fabric 3
[20240522_11:16:58.595][MatterTest] 05-22 03:16:59.280 INFO >>> [E:38899i S:46231 M:265560163 (Ack:218414524)] (S) Msg RX from 1:0000000012344321 [7E16] --- Type 0000:10 (SecureChannel:StandaloneAck)
[20240522_11:16:58.596][MatterTest] 05-22 03:16:59.281 INFO Found an existing secure session to [5:0000000012344321]!
[20240522_11:16:58.605][MatterTest] 05-22 03:16:59.290 INFO <<< [E:38900i S:46233 M:1154383] (S) Msg TX to 5:0000000012344321 [8486] [UDP:[fd11:35::f394:a563:db40:73f1%wpan0]:5540] --- Type 0001:06 (IM:WriteRequest)
[20240522_11:16:59.369][MatterTest] 05-22 03:17:00.051 INFO Retransmitting MessageCounter:1154383 on exchange 38900i Send Cnt 1
[20240522_11:17:00.080][MatterTest] 05-22 03:17:00.764 INFO Retransmitting MessageCounter:1154383 on exchange 38900i Send Cnt 2
[20240522_11:17:00.232][MatterTest] 05-22 03:17:00.916 INFO >>> [E:38900i S:46233 M:162018498 (Ack:1154383)] (S) Msg RX from 5:0000000012344321 [8486] --- Type 0001:07 (IM:WriteResponse)
[20240522_11:17:00.287][MatterTest] 05-22 03:17:00.972 INFO <<< [E:38900i S:46233 M:1154384 (Ack:162018498)] (S) Msg TX to 5:0000000012344321 [8486] [UDP:[fd11:35::f394:a563:db40:73f1%wpan0]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck)
[20240522_11:17:00.288][MatterTest] 05-22 03:17:00.973 INFO Step 14: Setting group key map on fabric 4
[20240522_11:17:00.289][MatterTest] 05-22 03:17:00.975 INFO >>> [E:38900i S:46233 M:162018499 (Ack:1154383)] (S) Msg RX from 5:0000000012344321 [8486] --- Type 0000:10 (SecureChannel:StandaloneAck)
[20240522_11:17:00.290][MatterTest] 05-22 03:17:00.976 INFO Found an existing secure session to [8:0000000012344321]!
[20240522_11:17:00.300][MatterTest] 05-22 03:17:00.986 INFO <<< [E:38901i S:46234 M:222357059] (S) Msg TX to 8:0000000012344321 [A797] [UDP:[fd11:35::f394:a563:db40:73f1%wpan0]:5540] --- Type 0001:06 (IM:WriteRequest)
[20240522_11:17:00.390][MatterTest] 05-22 03:17:01.067 INFO >>> [E:38900i S:46233 M:162018500 (Ack:1154383)] (S) Msg RX from 5:0000000012344321 [8486] --- Type 0000:10 (SecureChannel:StandaloneAck)
[20240522_11:17:01.007][MatterTest] 05-22 03:17:01.692 INFO Retransmitting MessageCounter:222357059 on exchange 38901i Send Cnt 1
[20240522_11:17:01.757][MatterTest] 05-22 03:17:02.439 INFO Retransmitting MessageCounter:222357059 on exchange 38901i Send Cnt 2
[20240522_11:17:02.283][MatterTest] 05-22 03:17:02.959 INFO >>> [E:38901i S:46234 M:259270866 (Ack:222357059)] (S) Msg RX from 8:0000000012344321 [A797] --- Type 0001:07 (IM:WriteResponse)
[20240522_11:17:02.331][MatterTest] 05-22 03:17:03.017 INFO <<< [E:38901i S:46234 M:222357060 (Ack:259270866)] (S) Msg TX to 8:0000000012344321 [A797] [UDP:[fd11:35::f394:a563:db40:73f1%wpan0]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck)
[20240522_11:17:02.333][MatterTest] 05-22 03:17:03.017 INFO Step 14: Setting group key map on fabric 5
[20240522_11:17:02.334][MatterTest] 05-22 03:17:03.019 INFO >>> [E:38901i S:46234 M:259270867 (Ack:222357059)] (S) Msg RX from 8:0000000012344321 [A797] --- Type 0000:10 (SecureChannel:StandaloneAck)
[20240522_11:17:02.335][MatterTest] 05-22 03:17:03.020 INFO >>> [E:38901i S:46234 M:259270868 (Ack:222357059)] (S) Msg RX from 8:0000000012344321 [A797] --- Type 0000:10 (SecureChannel:StandaloneAck)
[20240522_11:17:02.336][MatterTest] 05-22 03:17:03.021 INFO Found an existing secure session to [11:0000000012344321]!
[20240522_11:17:02.345][MatterTest] 05-22 03:17:03.031 INFO <<< [E:38902i S:46235 M:88594030] (S) Msg TX to 11:0000000012344321 [129B] [UDP:[fd11:35::f394:a563:db40:73f1%wpan0]:5540] --- Type 0001:06 (IM:WriteRequest)
[20240522_11:17:03.080][MatterTest] 05-22 03:17:03.764 INFO Retransmitting MessageCounter:88594030 on exchange 38902i Send Cnt 1
[20240522_11:17:03.814][MatterTest] 05-22 03:17:04.494 INFO Retransmitting MessageCounter:88594030 on exchange 38902i Send Cnt 2
[20240522_11:17:04.434][MatterTest] 05-22 03:17:05.118 INFO >>> [E:38902i S:46235 M:20087375 (Ack:88594030)] (S) Msg RX from 11:0000000012344321 [129B] --- Type 0001:07 (IM:WriteResponse)
[20240522_11:17:04.488][MatterTest] 05-22 03:17:05.173 INFO <<< [E:38902i S:46235 M:88594031 (Ack:20087375)] (S) Msg TX to 11:0000000012344321 [129B] [UDP:[fd11:35::f394:a563:db40:73f1%wpan0]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck)
[20240522_11:17:04.497][MatterTest] 05-22 03:17:05.173 INFO Step 14: **Setting group key map on fabric 6**
[20240522_11:17:04.498][MatterTest] 05-22 03:17:05.175 INFO >>> [E:38902i S:46235 M:20087376 (Ack:88594030)] (S) Msg RX from 11:0000000012344321 [129B] --- Type 0000:10 (SecureChannel:StandaloneAck)
[20240522_11:17:04.498][MatterTest] 05-22 03:17:05.176 INFO Found an existing secure session to [14:0000000012344321]!
[20240522_11:17:04.498][MatterTest] 05-22 03:17:05.182 INFO >>> [E:38902i S:46235 M:20087377 (Ack:88594030)] (S) Msg RX from 11:0000000012344321 [129B] --- Type 0000:10 (SecureChannel:StandaloneAck)
[20240522_11:17:04.503][MatterTest] 05-22 03:17:05.187 INFO <<< [E:38903i S:46236 M:242040225] (S) Msg TX to 14:0000000012344321 [88B2] [UDP:[fd11:35::f394:a563:db40:73f1%wpan0]:5540] --- Type 0001:06 (IM:WriteRequest)
[20240522_11:17:05.289][MatterTest] 05-22 03:17:05.971 INFO Retransmitting MessageCounter:242040225 on exchange 38903i Send Cnt 1
[20240522_11:17:06.037][MatterTest] 05-22 03:17:06.721 INFO Retransmitting MessageCounter:242040225 on exchange 38903i Send Cnt 2
[20240522_11:17:07.356][MatterTest] 05-22 03:17:08.039 INFO Retransmitting MessageCounter:242040225 on exchange 38903i Send Cnt 3
[20240522_11:17:13.244][MatterTest] 05-22 03:17:13.928 INFO Retransmitting MessageCounter:242040225 on exchange 38903i Send Cnt 4
[20240522_11:17:22.878][MatterTest] 05-22 03:17:23.562 ERROR Failed to Send CHIP MessageCounter:242040225 on exchange 38903i sendCount: 4 max retries: 4
[20240522_11:17:24.354][MatterTest] 05-22 03:17:25.026 ERROR Exception occurred in test_TC_RR_1_1.
[20240522_11:17:24.354]Traceback (most recent call last):
[20240522_11:17:24.354]  File "/root/python_testing/scripts/sdk/TC_RR_1_1.py", line 474, in test_TC_RR_1_1
[20240522_11:17:24.355]    await self.fill_and_validate_group_key_map(
[20240522_11:17:24.355]  File "/root/python_testing/scripts/sdk/TC_RR_1_1.py", line 583, in fill_and_validate_group_key_map
[20240522_11:17:24.356]    await client.WriteAttribute(
[20240522_11:17:24.356]  File "/usr/local/lib/python3.10/dist-packages/chip/ChipDeviceCtrl.py", line 1047, in WriteAttribute
[20240522_11:17:24.356]    return await future
[20240522_11:17:24.356]asyncio.exceptions.CancelledError
[20240522_11:17:24.356]
[20240522_11:17:24.356]During handling of the above exception, another exception occurred:
[20240522_11:17:24.356]
[20240522_11:17:24.357]Traceback (most recent call last):
[20240522_11:17:24.357]  File "/usr/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
[20240522_11:17:24.357]    return fut.result()
[20240522_11:17:24.357]asyncio.exceptions.CancelledError
[20240522_11:17:24.357]
[20240522_11:17:24.357]The above exception was the direct cause of the following exception:
[20240522_11:17:24.357]
[20240522_11:17:24.357]Traceback (most recent call last):
[20240522_11:17:24.357]  File "/usr/local/lib/python3.10/dist-packages/mobly/base_test.py", line 783, in exec_one_test
[20240522_11:17:24.357]    test_method()
[20240522_11:17:24.357]  File "/root/python_testing/scripts/sdk/matter_testing_support.py", line 1547, in async_runner
[20240522_11:17:24.358]    return asyncio.run(runner_with_timeout)
[20240522_11:17:24.358]  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
[20240522_11:17:24.358]    return loop.run_until_complete(main)
[20240522_11:17:24.358]  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
[20240522_11:17:24.358]    return future.result()
[20240522_11:17:24.358]  File "/usr/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
[20240522_11:17:24.358]    raise exceptions.TimeoutError() from exc
[20240522_11:17:24.359]asyncio.exceptions.TimeoutError
[20240522_11:17:24.369][MatterTest] 05-22 03:17:25.040 INFO            ***** Test Failure : `

### Platform

efr32

ChengqiangShe, 2024-05-22:
This TC_RR_1_1 testacase is failed  in the latest TH : Version: 2.10+spring2024 Sha:7ede6b, SHA chip-cert-bin:11f94c38171b709712106362e902f65253f4968

bzbarsky-apple, 2024-05-23:
@ChengqiangShe The test reads the Fabrics attribute from the device and then tries to do the group key map once per entry in the list it got.  So presumably it got a list with 6 entries...

Can you please attach, not paste, the full log?

ChengqiangShe, 2024-05-23:
I changed the macro definition value to 4 and solved this error.

ChengqiangShe, 2024-05-23:
I solved this error by changing the macro definition value to 4, which was 6 before.
The reason for the occurrence of timeOut is that it takes 29 seconds for the device to complete the execution and response after receiving this command, but the maximum timeout from chip-tool retransmission is 22 seconds. So by the time the device responds, chip-tool has timed out

ChengqiangShe, 2024-05-23:
However, another error occurred in the subsequent test. The DUT will use more than 255 pieces of nvm data, and then an error will be reported when writing to nvm. This may be a problem with the nvm operation of the efr32mg24 chip.
@silabs-Simran @bzbarsky-apple 

My key macro definition is as follows:

`/**
 * @def CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC
 *
 * @brief Defines the number of "endpoint->controlling group" mappings per fabric.
 *
 * Binds to number of GroupMapping entries per fabric
 */
#ifndef CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC
#define CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC 4 //2  //must > 4
#endif

/**
 * @def CHIP_CONFIG_MAX_GROUPS_PER_FABRIC
 *
 * @brief Defines the number of groups supported per fabric, see Group Key Management Cluster in specification.
 *
 * Binds to number of GroupState entries to support per fabric
 */
#ifndef CHIP_CONFIG_MAX_GROUPS_PER_FABRIC
#define CHIP_CONFIG_MAX_GROUPS_PER_FABRIC (4 * CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC)
#endif

#if CHIP_CONFIG_MAX_GROUPS_PER_FABRIC < (4 * CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC)
#error "Please ensure CHIP_CONFIG_MAX_GROUPS_PER_FABRIC meets minimum requirements. See Group Limits in the specification."
#endif

/**
 * @def CHIP_CONFIG_MAX_GROUPS_PER_FABRIC
 *
 * @brief Defines the number of groups key sets supported per fabric, see Group Key Management Cluster in specification.
 *
 * Binds to number of KeySet entries to support per fabric (Need at least 1 for Identity Protection Key)
 */
#ifndef CHIP_CONFIG_MAX_GROUP_KEYS_PER_FABRIC
#define CHIP_CONFIG_MAX_GROUP_KEYS_PER_FABRIC 4// 10 //2
#endif

#if CHIP_CONFIG_MAX_GROUP_KEYS_PER_FABRIC < 1
#error "Please ensure CHIP_CONFIG_MAX_GROUP_KEYS_PER_FABRIC > 0 to support at least the IPK."
#endif
`
[Dut_test.log](https://github.com/project-chip/connectedhomeip/files/15414065/Dut_test.log)

[TC_RR_1_1 Fail.log](https://github.com/project-chip/connectedhomeip/files/15414062/TC_RR_1_1.Fail.log)

KaMeHaMeHaaaa, 2024-11-28:
Hi @bzbarsky-apple, we have a product with six endpoints, all of which belong to device type On/Off Plug-in Unit. This results in six group clusters. According to the test case requirements, I must set CHIP_CONFIG_MAX_GROUPS_PER_FABRIC to 24. However, setting it to 24 causes Setting group key map to take too long(about 11 seconds), resulting in a timeout when sending the response, ultimately leading to failure. Do you have any suggestions? Thank you!

bzbarsky-apple, 2024-12-05:
@KaMeHaMeHaaaa Please check with whoever wrote the test?  I recall there being issues with the tests having too-short timeouts....

That said, it also sounds like something is broken somewhere if writing 24 things takes 11 seconds....