project-chip/connectedhomeip #33630: [BUG] - Sometimes matter SDK crashes when shutdownCommissioning is called

igor90ns, 2024-05-28:
### Reproduction steps

When user try to add several devices in a row, sometimes happens that matter sdk crashes.

In our implementation mobile application is commissioner only, we commission device and then pass parameters to our hub in order to establish case session and control devices.
After **onCommissioningComplete** is received we call **shutdownCommissioning()**

This flow work 90% of time, but sometimes it happens that matter sdk crash, and causes unstable behavior in core app.
Issue mainly prevail when app is built as release rather then debug, and when thread devices are commissioned rather then IP ( but it can happen in debug and with ip devices)

```
Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 8072 (jio.smartliving), pid 8072 (jio.smartliving)
---------------------------- PROCESS STARTED (8623) for package com.jio.smartliving ----------------------------
2024-05-22 14:09:10.097  8621-8621  DEBUG                   crash_dump64                         A  Cmdline: com.jio.smartliving
2024-05-22 14:09:10.097  8621-8621  DEBUG                   crash_dump64                         A  pid: 8072, tid: 8072, name: jio.smartliving  >>> com.jio.smartliving <<<
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #01 pc 0000000002417ed8  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/lib/arm64/libCHIPController.so
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #02 pc 0000000002417e90  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/lib/arm64/libCHIPController.so
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #03 pc 0000000002417de4  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/lib/arm64/libCHIPController.so (chip::Messaging::ExchangeManager::~ExchangeManager()+56)
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #04 pc 00000000024411a8  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/lib/arm64/libCHIPController.so
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #05 pc 0000000002440e4c  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/lib/arm64/libCHIPController.so
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #06 pc 0000000002440bac  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/lib/arm64/libCHIPController.so
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #07 pc 000000000244ff48  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/lib/arm64/libCHIPController.so (chip::Controller::DeviceController::Shutdown()+228)
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #08 pc 0000000002451458  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/lib/arm64/libCHIPController.so (chip::Controller::DeviceCommissioner::Shutdown()+236)
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #09 pc 0000000000d0cfa4  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/lib/arm64/libCHIPController.so
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #10 pc 0000000000d1fe80  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/lib/arm64/libCHIPController.so (Java_chip_devicecontroller_ChipDeviceController_shutdownCommissioning+60)
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #13 pc 00000000002dfee8  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/oat/arm64/base.vdex (chip.devicecontroller.ChipDeviceController.shutdownCommissioning+4)
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #15 pc 000000000087cfda  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/oat/arm64/base.vdex (F4.g.p+102)
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #17 pc 000000000095af36  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/oat/arm64/base.vdex (W4.c.d+10)
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #19 pc 000000000095c858  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/oat/arm64/base.vdex (W5.q.G1+8)
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #21 pc 000000000095d5b4  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/oat/arm64/base.vdex (W5.r.B2+4)
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #23 pc 0000000000adeaf6  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/oat/arm64/base.vdex (f6.d.Q3+110)
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #25 pc 000000000095da78  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/oat/arm64/base.vdex (W5.r.X5+28)
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #27 pc 000000000095bee0  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/oat/arm64/base.vdex (W5.q$d.g+320)
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #29 pc 000000000095bf84  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/oat/arm64/base.vdex (W5.q$d.d+4)
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #31 pc 000000000090abfa  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/oat/arm64/base.vdex (O8.n$a.h+94)
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #33 pc 000000000090ad88  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/oat/arm64/base.vdex (.n$a.run&&+16)
2024-05-22 14:09:10.098  8621-8621  DEBUG                   crash_dump64                         A        #35 pc 000000000085ccb8  /data/app/~~CAEp8Q09T12Vi6CWqvBF_A==/com.jio.smartliving-T38pLZAydK3p2o1oKilaLg==/oat/arm64/base.vdex (.c$b.run complex+4)
---------------------------- PROCESS STARTED (8639) for package com.jio.smartliving ----------------------------
---------------------------- PROCESS ENDED (8072) for package com.jio.smartliving ----------------------------
------------------------

```

Full commissioning log is attached
[shutdown crash.txt](https://github.com/project-chip/connectedhomeip/files/15465350/shutdown.crash.txt)

### Bug prevalence

1/10

### GitHub hash of the SDK that was being used

181b0cb14ff007ec912f2ba6627e05dfb066c008

### Platform

android

### Platform Version(s)

v1.2.0.1

igor90ns, 2024-06-10:
Is there anything new about this issue?

yunhanw-google, 2024-12-04:
@MrLiuYunPing @igor90ns  could you try to revert the change from https://github.com/project-chip/connectedhomeip/pull/29881, and see whether this issue can be relieved?

MrLiuYunPing, 2024-12-09:
> #29881 
I think there is another issue causes the crash. Just for example, the Java finalize() method is deprecated, it will not be called. The AndroidDeviceControllerWrapper object won't call ~AndroidDeviceControllerWrapper, so the DefaultICDClientStorag's shutdown will not be called, too.

closed 2024-12-12 by yunhanw-google