project-chip/connectedhomeip #33657: [CERT-TEST-FAILURE] [TC-TMP-2.1] -  Python Script Failure

Ashwinigrl, 2024-05-29:
### Feature Area

Other

### Test Case

TC-TMP-2.1

### Reproduction steps

1. Execute the below mentioned command to put DUT into a commissionable state : `./chip-all-clusters-app`
2. Once DUT reach the commissionable state sent Python run command on TH : `python3 TC_TMP_2_1.py --commissioning-method on-network --discriminator 3840 --passcode 20202021  --storage-path admin_storage.json --paa-trust-store-path ../../credentials/development/paa-root-certs/ `

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

16d6475ef81f890371ab0d1b1f7e42ea4318d13a

### Platform

raspi

### Anything else?

I tried to execute the **TC-TMP-2.1** Python test, but the test case is failing at Test Step 3 (TH reads the `MinMeasuredValue` attribute from the DUT and saves as `min_measured_value`.  If  `min_measured_value` is not null, set `min_bound` to `min_measured_value`) with below error.

> [MatterTest] 05-28 12:28:37.896 INFO 		***** Test Step 3 : TH reads the MinMeasuredValue attribute from the DUT and saves as `min_measured_value`. If `min_measured_value` is not null, set `min_bound` to `min_measured_value`
> [MatterTest] 05-28 12:28:37.901 INFO Found an existing secure session to [1:0000000012344321]!
> [MatterTest] 05-28 12:28:37.908 INFO <<< [E:43630i S:47791 M:215481016] (S) Msg TX to 1:0000000012344321 [9676] [UDP:[fe80::1869:7dff:fe5c:16e8%vethfaf8632]:5540] --- Type 0001:02 (IM:ReadRequest)
> [MatterTest] 05-28 12:28:38.094 INFO Received unknown unsecure packet for initiator 0x00AB224CF821E90A
> [MatterTest] 05-28 12:28:38.105 INFO >>> [E:43630i S:47791 M:21561934 (Ack:215481016)] (S) Msg RX from 1:0000000012344321 [9676] --- Type 0001:05 (IM:ReportData)
> [MatterTest] 05-28 12:28:38.120 INFO <<< [E:43630i S:47791 M:215481017 (Ack:21561934)] (S) Msg TX to 1:0000000012344321 [9676] [UDP:[fe80::1869:7dff:fe5c:16e8%vethfaf8632]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck)
> [MatterTest] 05-28 12:28:38.124 ERROR Exception occurred in test_TC_TMP_2_1.
> Traceback (most recent call last):
>   File "/home/ubuntu/master_cntrl/connectedhomeip/no/lib/python3.10/site-packages/mobly/base_test.py", line 783, in exec_one_test
>     test_method()
>   File "/home/ubuntu/master_cntrl/connectedhomeip/src/python_testing/matter_testing_support.py", line 1553, in async_runner
>     return asyncio.run(runner_with_timeout)
>   File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
>     return loop.run_until_complete(main)
>   File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
>     return future.result()
>   File "/usr/lib/python3.10/asyncio/tasks.py", line 445, in wait_for
>     return fut.result()
>   File "/home/ubuntu/master_cntrl/connectedhomeip/src/python_testing/TC_TMP_2_1.py", line 59, in test_TC_TMP_2_1
>     min_measured_value = await self.read_single_attribute_check_success(cluster=cluster, attribute=attr.MinMeasuredValue)
>   File "/home/ubuntu/master_cntrl/connectedhomeip/src/python_testing/matter_testing_support.py", line 849, in read_single_attribute_check_success
>     asserts.assert_true(read_ok, read_err_msg)
>   File "/home/ubuntu/master_cntrl/connectedhomeip/no/lib/python3.10/site-packages/mobly/asserts.py", line 353, in assert_true
>     fail(msg, extras)
>   File "/home/ubuntu/master_cntrl/connectedhomeip/no/lib/python3.10/site-packages/mobly/asserts.py", line 475, in fail
>     raise signals.TestFailure(msg, extras)
> mobly.signals.TestFailure: Details=Error reading <class 'chip.clusters.Objects.TemperatureMeasurement'>:<class 'chip.clusters.Objects.TemperatureMeasurement.Attributes.MinMeasuredValue'> = ValueDecodeFailure(TLVValue=None, Reason=InteractionModelError(<Status.UnsupportedCluster: 195>)), Extras=None
> [MatterTest] 05-28 12:28:38.156 INFO 		***** Test Failure : 

**Step that require review:**
Test step 3 

**Reference log:**
[TC-TMP-2.1(DUT).txt](https://github.com/project-chip/connectedhomeip/files/15483396/TC-TMP-2.1.DUT.txt)
[TC-TMP-2.1(CNTRL).txt](https://github.com/project-chip/connectedhomeip/files/15483399/TC-TMP-2.1.CNTRL.txt)

**Spec References:**
- https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/TemperatureMeasurement.adoc

**Test Plan References:**
- https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/temperaturemeasurement.adoc

**Python script References:**
- https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/TC_TMP_2_1.py

cjandhyala, 2024-05-29:
@Ashwinigrl do you know when this test was passing last time ? If you could git bisect and narrow down the commitID it failed, that would be super helpful.

Ashwinigrl, 2024-05-30:
This is a newly added script @cjandhyala  There are no issues with the SDK, and we can run the test case manually as per the test plan. Even though the script matches the test plan, it fails at the 3rd step during execution.

bzbarsky-apple, 2024-05-30:
Why is TH getting back "unsupported cluster" here?  That cluster is definitely supported on all-cluster-app....

(Separately, TH seems to not be doing detailed data model logging here, so it's hard to tell what it really got.)

@cecille

cecille, 2024-06-06:
test defaults to EP0, so try with --endpoint 1 (or wherever that cluster is located on the device)

Ashwinigrl, 2024-06-10:
Tried with `--endpoint 1`, Test case is passed. hence closing this Issue.

closed 2024-06-10 by Ashwinigrl