project-chip/connectedhomeip #33663: [CERT-TEST-FAILURE] [TC-SC-3.6][TC-RR-1.1] - Test Failed

Rajashreekalmane, 2024-05-29:
**Description:**
**1. Tried to validate TC-SC-3.6 and TC-RR-1.1 in V1.3 TH but** 

   After the DUT was paired by a previous script (FAN-3.1),  ran TC-SC-3.6 and TC-RR-1.1, which resulted in failure. However, after running these two scripts, TC-FAN-3.1 passed successfully in V1.3 TH

**PFA error log for TC-SC-3.6** 

> [MatterTest] 05-29 15:33:07.955 INFO Reverted Last Known Good Time to previous value
> [MatterTest] 05-29 15:33:07.958 ERROR Exception occurred in test_TC_SC_3_6.
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.10/dist-packages/mobly/base_test.py", line 783, in exec_one_test
>     test_method()
>   File "/root/python_testing/scripts/sdk/matter_testing_support.py", line 1547, in async_runner
>     return asyncio.run(runner_with_timeout)
>   File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
>     return loop.run_until_complete(main)
>   File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
>     return future.result()
>   File "/usr/lib/python3.10/asyncio/tasks.py", line 445, in wait_for
>     return fut.result()
>   File "/root/python_testing/scripts/sdk/TC_SC_3_6.py", line 155, in test_TC_SC_3_6
>     new_admin_ctrl = new_fabric_admin.NewController(nodeId=dev_ctrl.nodeId)
>   File "/usr/local/lib/python3.10/dist-packages/chip/FabricAdmin.py", line 101, in NewController
>     controller = ChipDeviceCtrl.ChipDeviceController(
>   File "/usr/local/lib/python3.10/dist-packages/chip/ChipDeviceCtrl.py", line 1774, in __init__
>     ).raise_on_error()
>   File "/usr/local/lib/python3.10/dist-packages/chip/native/__init__.py", line 87, in raise_on_error
>     raise self.to_exception()
> chip.exceptions.ChipStackError: src/credentials/FabricTable.cpp:815: CHIP Error 0x0000000B: No memory
> [MatterTest] 05-29 15:33:07.982 INFO 		***** Test Failure : 
> [MatterTest] 05-29 15:33:07.982 INFO Finished test in 1110ms
> [MatterTest] 05-29 15:33:07.983 INFO [Test] test_TC_SC_3_6 ERROR

**PFA error log for TC-RR-1.1**

> MatterTest] 05-29 15:33:36.459 INFO Reverted Last Known Good Time to previous value
> [MatterTest] 05-29 15:33:36.461 ERROR Exception occurred in test_TC_RR_1_1.
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.10/dist-packages/mobly/base_test.py", line 783, in exec_one_test
>     test_method()
>   File "/root/python_testing/scripts/sdk/matter_testing_support.py", line 1547, in async_runner
>     return asyncio.run(runner_with_timeout)
>   File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
>     return loop.run_until_complete(main)
>   File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
>     return future.result()
>   File "/usr/lib/python3.10/asyncio/tasks.py", line 445, in wait_for
>     return fut.result()
>   File "/root/python_testing/scripts/sdk/TC_RR_1_1.py", line 151, in test_TC_RR_1_1
>     new_admin_ctrl = new_fabric_admin.NewController(nodeId=dev_ctrl.nodeId, catTags=[0x0001_0001])
>   File "/usr/local/lib/python3.10/dist-packages/chip/FabricAdmin.py", line 101, in NewController
>     controller = ChipDeviceCtrl.ChipDeviceController(
>   File "/usr/local/lib/python3.10/dist-packages/chip/ChipDeviceCtrl.py", line 1774, in __init__
>     ).raise_on_error()
>   File "/usr/local/lib/python3.10/dist-packages/chip/native/__init__.py", line 87, in raise_on_error
>     raise self.to_exception()
> chip.exceptions.ChipStackError: src/credentials/FabricTable.cpp:815: CHIP Error 0x0000000B: No memory
> [MatterTest] 05-29 15:33:36.471 INFO 		***** Test Failure : 
> [MatterTest] 05-29 15:33:36.472 INFO Finished test in 1164ms
> [MatterTest] 05-29 15:33:36.473 INFO [Test] test_TC_RR_1_1 ERROR
> [MatterTest] 05-29 15:33:36.483 INFO Teardown: shutting down all subscription to avoid racy callbacks

**GitHub hash of the SDK that was being used**
Version: 2.10+spring2024
Sha: 5f28838

**PFA Log below for V1.3 TH:** 
[tc-sc-3.6 and tc-rr-1.1 - V1.3 validation log.txt](https://github.com/project-chip/connectedhomeip/files/15487673/tc-sc-3.6.and.tc-rr-1.1.-.V1.3.validation.log.txt)

**2. Tried to validate TC-SC-3.6 and TC-RR-1.1 in V1.2 TH, TC-SC-3.6 got passed  but The TC-RR-1.1 is got Failed** 

**PFA error log for TC-RR-1.1**
> [MatterTest] 05-28 15:21:02.540 ERROR Exception occurred in test_TC_RR_1_1.
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.10/dist-packages/mobly/base_test.py", line 783, in exec_one_test
>     test_method()
>   File "/root/python_testing/scripts/sdk/matter_testing_support.py", line 1547, in async_runner
>     return asyncio.run(runner_with_timeout)
>   File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
>     return loop.run_until_complete(main)
>   File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
>     return future.result()
>   File "/usr/lib/python3.10/asyncio/tasks.py", line 445, in wait_for
>     return fut.result()
>   File "/root/python_testing/scripts/sdk/TC_RR_1_1.py", line 197, in test_TC_RR_1_1
>     new_controllers = await CommissioningBuildingBlocks.CreateControllersOnFabric(
>   File "/usr/local/lib/python3.10/dist-packages/chip/utils/CommissioningBuildingBlocks.py", line 140, in CreateControllersOnFabric
>     newController = fabricAdmin.NewController(nodeId=nodeId, paaTrustStorePath=paaTrustStorePath, catTags=catTags)
>   File "/usr/local/lib/python3.10/dist-packages/chip/FabricAdmin.py", line 101, in NewController
>     controller = ChipDeviceCtrl.ChipDeviceController(
>   File "/usr/local/lib/python3.10/dist-packages/chip/ChipDeviceCtrl.py", line 1576, in __init__
>     ).raise_on_error()
>   File "/usr/local/lib/python3.10/dist-packages/chip/native/__init__.py", line 67, in raise_on_error
>     raise self.to_exception()
> chip.exceptions.ChipStackError: src/credentials/FabricTable.cpp:802: CHIP Error 0x0000000B: No memory
> [MatterTest] 05-28 15:21:02.551 INFO 		***** Test Failure : 
> [MatterTest] 05-28 15:21:02.552 INFO Finished test in 2731ms
> [MatterTest] 05-28 15:21:02.553 INFO [Test] test_TC_RR_1_1 ERROR
> [MatterTest] 05-28 15:21:02.564 INFO Teardown: shutting down all subscription to avoid racy callbacks

**PFA log below for V1.2 :** 
[TC-RR-1.1 and TC-SC-3.6 -V1.2 Execution.txt](https://github.com/project-chip/connectedhomeip/files/15487712/TC-RR-1.1.and.TC-SC-3.6.-V1.2.Execution.txt)

**3. Tried to validate TC-SC-3.6 and TC-RR-1.1 using master as DUT and CNTR, TC-SC-3.6 got passed  but The TC-RR-1.1 is got Failed**

**PFA error log for TC-RR-1.1**

> [MatterTest] 05-29 15:15:41.398 INFO Avahi group established
> [MatterTest] 05-29 15:15:41.399 ERROR Exception occurred in test_TC_RR_1_1.
> Traceback (most recent call last):
>   File "/home/ubuntu/May27_dut/connectedhomeip/no/lib/python3.10/site-packages/mobly/base_test.py", line 783, in exec_one_test
>     test_method()
>   File "/home/ubuntu/May27_dut/connectedhomeip/src/python_testing/matter_testing_support.py", line 1553, in async_runner
>     return asyncio.run(runner_with_timeout)
>   File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
>     return loop.run_until_complete(main)
>   File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
>     return future.result()
>   File "/usr/lib/python3.10/asyncio/tasks.py", line 445, in wait_for
>     return fut.result()
>   File "/home/ubuntu/May27_dut/connectedhomeip/src/python_testing/TC_RR_1_1.py", line 196, in test_TC_RR_1_1
>     new_controllers = await CommissioningBuildingBlocks.CreateControllersOnFabric(
>   File "/home/ubuntu/May27_dut/connectedhomeip/no/lib/python3.10/site-packages/chip/utils/CommissioningBuildingBlocks.py", line 140, in CreateControllersOnFabric
>     newController = fabricAdmin.NewController(nodeId=nodeId, paaTrustStorePath=paaTrustStorePath, catTags=catTags)
>   File "/home/ubuntu/May27_dut/connectedhomeip/no/lib/python3.10/site-packages/chip/FabricAdmin.py", line 101, in NewController
>     controller = ChipDeviceCtrl.ChipDeviceController(
>   File "/home/ubuntu/May27_dut/connectedhomeip/no/lib/python3.10/site-packages/chip/ChipDeviceCtrl.py", line 1768, in __init__
>     ).raise_on_error()
>   File "/home/ubuntu/May27_dut/connectedhomeip/no/lib/python3.10/site-packages/chip/native/__init__.py", line 87, in raise_on_error
>     raise self.to_exception()
> chip.exceptions.ChipStackError: src/credentials/FabricTable.cpp:815: CHIP Error 0x0000000B: No memory
> [MatterTest] 05-29 15:15:41.410 INFO Avahi group established
> [MatterTest] 05-29 15:15:41.416 INFO 		***** Test Failure : 
> [MatterTest] 05-29 15:15:41.417 INFO Finished test in 2511ms

**PFA log below for master commit**
[tc-sc-3.6 and tc-rr-1.1 master.txt](https://github.com/project-chip/connectedhomeip/files/15487769/tc-sc-3.6.and.tc-rr-1.1.master.txt)

cecille, 2024-05-29:
These errors are different than the initial errors reported on slack, which were related to the initial commissioning.

Can you please try removing the TH controller KVS file and re-trying the tests?

Rajashreekalmane, 2024-05-29:
After removing the TH controller KVS file, I paired the DUT using a previous script (FAN-3.1).  then ran the scripts TC-SC-3.6, TC-RR-1.1, and TC-FAN-3.1, which resulted in failures in V1.3 image 

**error log test_TC_RR_1_1**
> [MatterTest] 05-29 17:40:56.357 ERROR Exception occurred in test_TC_RR_1_1.
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.10/dist-packages/mobly/base_test.py", line 783, in exec_one_test
>     test_method()
>   File "/root/python_testing/scripts/sdk/matter_testing_support.py", line 1547, in async_runner
>     return asyncio.run(runner_with_timeout)
>   File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
>     return loop.run_until_complete(main)
>   File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
>     return future.result()
>   File "/usr/lib/python3.10/asyncio/tasks.py", line 445, in wait_for
>     return fut.result()
>   File "/root/python_testing/scripts/sdk/TC_RR_1_1.py", line 477, in test_TC_RR_1_1
>     Dict[int, Clusters.GroupKeyManagement.Structs.GroupInfoMapStruct]] = await self.add_all_groups(
>   File "/root/python_testing/scripts/sdk/TC_RR_1_1.py", line 655, in add_all_groups
>     add_response: Clusters.Groups.Commands.AddGroupResponse = await client.SendCommand(
>   File "/usr/local/lib/python3.10/dist-packages/chip/ChipDeviceCtrl.py", line 950, in SendCommand
>     return await future
> chip.exceptions.ChipStackError: src/app/CommandSender.cpp:336: CHIP Error 0x00000032: Timeout
> [MatterTest] 05-29 17:40:56.369 INFO 		***** Test Failure : 
> [MatterTest] 05-29 17:40:56.369 INFO Finished test in 29581ms
> [MatterTest] 05-29 17:40:56.370 INFO [Test] test_TC_RR_1_1 ERROR
> [MatterTest] 05-29 17:40:56.378 INFO Teardown: shutting down all subscription to avoid racy callbacks
> [MatterTest] 05-29 17:40:56.383 INFO Summary for test class TC_RR_1_1: Error 1, Executed 1, Failed 0, Passed 0, Requested 1, Skipped 0
> [MatterTest] 05-29 17:40:56.385 INFO Summary for test run MatterTest@05-29-2024_17-40-26-774:
> Total time elapsed 29.60804031600128s
>

PFA log below 
[V1.3 (RR-1.1,SC-3.6).txt](https://github.com/project-chip/connectedhomeip/files/15489327/V1.3.RR-1.1.SC-3.6.txt)

Rajashreekalmane, 2024-05-30:
The test cases passed after removing the KVS file in v1.3 TH.

closed 2024-05-30 by Rajashreekalmane