project-chip/connectedhomeip #33676: The DAC for the end device needs to be certified by an authority, so does it need to be if Commissioner and Controller?[1.3]

yf-fan-org, 2024-05-30:
### Reproduction steps

The end device needs certificate issued by authoritative organization because of DAC, does it need certificate issued by commissioner and controller? Because the commissioner needs to generate NOC, and NOC depends on RCAC, does this RCAC certificate need to be issued by an authoritative organization, or can it be generated by self-signature?

### Bug prevalence

day

### GitHub hash of the SDK that was being used

1

### Platform

other

### Type

Test Improvement

### Anything else?

The end device needs certificate issued by authoritative organization because of DAC, does it need certificate issued by commissioner and controller? Because the commissioner needs to generate NOC, and NOC depends on RCAC, does this RCAC certificate need to be issued by an authoritative organization, or can it be generated by self-signature?

bzbarsky-apple, 2024-05-30:
RCAC can be generated by anyone.

closed 2024-05-30 by bzbarsky-apple

yf-fan-org, 2024-05-31:
> RCAC can be generated by anyone.

So how does this ensure it's authoritative, or is it sufficient here to be able to generate RCACs

bzbarsky-apple, 2024-06-01:
Anyone can be a commissioner, yes, as long as a user wants them to be.