project-chip/connectedhomeip #33688: [BUG] Commissioning failed with CHIP Error 0x000000C9: No shared trusted root

Saravana-kr22, 2024-05-31:
### Reproduction steps

I have used python controller to done a stress test on my RPI sample dut (all-clusters-app) on Pair-Unpair operation for 10000 ietration. After completion of the 10000 ietration i found that the 22 where failed to commission the dut during the case session with the below traceback

> Traceback (most recent call last):
  File "/Users/chaitanyajandhyala/code/python/matter-qa/src/matter_qa/library/helper_libs/matter_testing_support.py", line 1177, in pair_dut
    raise CommissionFailure(f"Pairing with DUT failed Reason {commission_device_result.__str__()} ")
chip.commissioning.CommissionFailure: CommissionFailure(Pairing with DUT failed Reason src/protocols/secure_channel/CASESession.cpp:1984: CHIP Error 0x000000C9: No shared trusted root )

### Bug prevalence

Whenever i do this

### GitHub hash of the SDK that was being used

3718e996b56f36a553c1b4571f77728396590776

### Platform

python, raspi

### Platform Version(s)

RPI-4, 8GB RAM

### Anything else?

**Example of Failed iteartion details:**

The iteartion number 2608 is passed below file for the same 
TH log: [TH_log_2024-04-26_04-31-38.log](https://github.com/user-attachments/files/15515995/log_2024-04-26_04-31-38.log)
DUT log: [Dut_log_2608_2024-04-26T04_31_47_630533.log](https://github.com/user-attachments/files/15516001/Dut_log_2608_2024-04-26T04_31_47_630533.log)

The iteration number 2609 is failed with this error below the logfile
TH log : [TH_log.log](https://github.com/user-attachments/files/15515981/log_2024-04-26_04-31-58.log)
DUT log: [Dut_log_2609_2024-04-26T04_37_18_972898.log](https://github.com/user-attachments/files/15515990/Dut_log_2609_2024-04-26T04_37_18_972898.log)

The iteartion number 2610 is passed below file for the same 
TH log:  [TH_log_2024-04-26_04-37-27.log](https://github.com/user-attachments/files/15516016/log_2024-04-26_04-37-27.log)
Dut log:  [Dut_log_2610_2024-04-26T04_38_36_549561.log](https://github.com/user-attachments/files/15516033/Dut_log_2610_2024-04-26T04_38_36_549561.log)

For the detailed summary please reference this json: [summary.json](https://github.com/user-attachments/files/15516042/summary.json)

gokul-bhalodiya-7, 2024-05-31:
When Matter controller commission any end-device then it will store the operational credentials in it's persistent memory which will be used later for controlling it.

cjandhyala, 2024-06-05:
@gokul-bhalodiya-7 as you there in the attached logs, the previous iteration successfully paired and unpaired the DUT. 

To your question, if the unpair indeed deleted the credentials,  the test case does factory reset the raspi before starting the next iteration i.e. it kills the app, deletes the temp storage on raspi before starting the next iteration, which should have deleted the credentials on the storage.

cjandhyala, 2024-06-18:
closing the ticket for now, following-up with a new ticket..

closed 2024-06-18 by cjandhyala