project-chip/connectedhomeip #33697: [Test Failed] TestDiscovery expects announcement for device type 0xffff which is no valie MEI

Apollon77, 2024-05-31:
### Test issue(s)

I know that TestDiscovery.yaml is a chip internal test but it expects that the announced device type is 0xffff which is not valid MEI definition for a Device type according to chip specs. The highest allowed noin-manufacturerspecific devicetype should be 0xbfff (aka 49151).

Even tho the chip test applications are just reference implementations I think they should not use out of range data and formally invalid IDs

bzbarsky-apple, 2024-06-01:
It's running against all-clusters-app.  There is no "valid" device type that can be used for that thing, obviously....

Apollon77, 2024-06-01:
> Probably it should just use a vendor-scoped device type in one of the test vendor namespaces.

that would be the most simple option, I agree

bzbarsky-apple, 2024-06-03:
One other note: the test expects _whatever device type id is specified on the command line_, defaulting to 0xFFFF.  When running it against anything other than all-clusters-app, the right data should be passed on the command line.

Apollon77, 2024-06-03:
@bzbarsky-apple ahh ok yes I could override that setting ... will look into this too.thank you

closed 2024-06-05 by mergify[bot]