project-chip/connectedhomeip #33706: [1.2] After complete CASE Session establish, AccessControl: denied

lxj199722, 2024-06-03:
### Reproduction steps

I am using an iPhone, HomePod mini, and the Home app to commission a Matter device. 
After the commissioning is completed, the Matter device goes offline immediately.
I found that during the commissioning process, the CASE sessions would be established three times. 
Upon completing the last CASE session, the log of the Matter device prints "AccessControl: denied."

```
[0527_17:34:32_177]#chip[EM] >>> [E:19127r S:0 M:192362848] (U) Msg RX from 0:68311E2F6DA65BAD [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
[0527_17:34:32_184]#chip[IN] CASE Server received Sigma1 message . Starting handshake. EC 0x10013470
[0527_17:34:32_223]#chip[EM] <<< [E:19127r S:0 M:174265717 (Ack:192362848)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::C02:6745:FCD1:BF2D%r00]:49378] --- Type 0000:10 (SecureChannel:StandaloneAck)
[0527_17:34:32_224]#chip[SC] Received Sigma1 msg
[0527_17:34:32_347]#chip[SC] CASE matched destination ID: fabricIndex 1, NodeID 0x00000000B9A53BE7
[0527_17:34:33_382]#chip[EM] <<< [E:19127r S:0 M:174265718 (Ack:192362848)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::C02:6745:FCD1:BF2D%r00]:49378] --- Type 0000:31 (SecureChannel:CASE_Sigma2)
[0527_17:34:33_384]#chip[SC] Sent Sigma2 msg
[0527_17:34:33_394]#chip[DL] Long dispatch time: 1216 ms, for event type 3
[0527_17:34:33_406]#chip[EM] >>> [E:19127r S:0 M:192362848] (U) Msg RX from 0:68311E2F6DA65BAD [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
[0527_17:34:33_419]#chip[EM] <<< [E:19127r S:0 M:174265719 (Ack:192362848)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::C02:6745:FCD1:BF2D%r00]:49378] --- Type 0000:10 (SecureChannel:StandaloneAck)
[0527_17:34:33_428]#chip[EM] >>> [E:19127r S:0 M:192362848] (U) Msg RX from 0:68311E2F6DA65BAD [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
[0527_17:34:33_456]#chip[EM] <<< [E:19127r S:0 M:174265720 (Ack:192362848)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::C02:6745:FCD1:BF2D%r00]:49378] --- Type 0000:10 (SecureChannel:StandaloneAck)
[0527_17:34:33_737]#chip[EM] Retransmitting MessageCounter:174265718 on exchange 19127r Send Cnt 1
[0527_17:34:34_098]#chip[EM] Retransmitting MessageCounter:174265718 on exchange 19127r Send Cnt 2
[0527_17:34:34_737]#chip[EM] Retransmitting MessageCounter:174265718 on exchange 19127r Send Cnt 3
[0527_17:34:35_668]#chip[EM] Retransmitting MessageCounter:174265718 on exchange 19127r Send Cnt 4
[0527_17:34:36_382]#
[0527_17:34:36_553]##chip[EM] >>> [E:19127r S:0 M:192362849 (Ack:174265718)] (U) Msg RX from 0:68311E2F6DA65BAD [0000] --- Type 0000:32 (SecureChannel:CASE_Sigma3)
[0527_17:34:36_576]#chip[EM] <<< [E:19127r S:0 M:174265721 (Ack:192362849)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::C02:6745:FCD1:BF2D%r00]:49378] --- Type 0000:10 (SecureChannel:StandaloneAck)
[0527_17:34:36_577]#chip[SC] Received Sigma3 msg
[0527_17:34:36_687]#chip[SC] The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
[0527_17:34:36_698]#chip[DL] Long dispatch time: 155 ms, for event type 3
[0527_17:34:36_711]#chip[EM] >>> [E:19127r S:0 M:192362850 (Ack:174265718)] (U) Msg RX from 0:68311E2F6DA65BAD [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
[0527_17:34:36_723]#chip[EM] >>> [E:19127r S:0 M:192362851 (Ack:174265718)] (U) Msg RX from 0:68311E2F6DA65BAD [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
[0527_17:34:36_746]#chip[EM] >>> [E:19127r S:0 M:192362852 (Ack:174265718)] (U) Msg RX from 0:68311E2F6DA65BAD [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
[0527_17:34:37_502]#chip[DL] Long dispatch time: 763 ms, for event type 2
[0527_17:34:37_514]#chip[EM] >>> [E:19127r S:0 M:192362849 (Ack:174265718)] (U) Msg RX from 0:68311E2F6DA65BAD [0000] --- Type 0000:32 (SecureChannel:CASE_Sigma3)
[0527_17:34:37_538]#chip[EM] <<< [E:19127r S:0 M:174265722 (Ack:192362849)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::C02:6745:FCD1:BF2D%r00]:49378] --- Type 0000:10 (SecureChannel:StandaloneAck)
[0527_17:34:37_549]#chip[EM] >>> [E:19127r S:0 M:192362849 (Ack:174265718)] (U) Msg RX from 0:68311E2F6DA65BAD [0000] --- Type 0000:32 (SecureChannel:CASE_Sigma3)
[0527_17:34:37_561]#chip[EM] <<< [E:19127r S:0 M:174265723 (Ack:192362849)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::C02:6745:FCD1:BF2D%r00]:49378] --- Type 0000:10 (SecureChannel:StandaloneAck)
[0527_17:34:37_569]#chip[EM] >>> [E:19127r S:0 M:192362849 (Ack:174265718)] (U) Msg RX from 0:68311E2F6DA65BAD [0000] --- Type 0000:32 (SecureChannel:CASE_Sigma3)
[0527_17:34:37_599]#chip[EM] <<< [E:19127r S:0 M:174265724 (Ack:192362849)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::C02:6745:FCD1:BF2D%r00]:49378] --- Type 0000:10 (SecureChannel:StandaloneAck)
[0527_17:34:38_139]#chip[EM] <<< [E:19127r S:0 M:174265725 (Ack:192362849)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::C02:6745:FCD1:BF2D%r00]:49378] --- Type 0000:40 (SecureChannel:StatusReport)
[0527_17:34:38_144]#chip[SC] SecureSession[0x10011c58, LSID:52823]: State change 'kEstablishing' --> 'kActive'
[0527_17:34:38_145]#chip[IN] CASE Session established to peer: <000000002D4A61A6, 1>
[0527_17:34:38_156]#chip[DL] Long dispatch time: 553 ms, for event type 2
[0527_17:34:38_173]#chip[EM] >>> [E:19127r S:0 M:192362853 (Ack:174265718)] (U) Msg RX from 0:68311E2F6DA65BAD [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
[0527_17:34:38_502]#chip[EM] Retransmitting MessageCounter:174265725 on exchange 19127r Send Cnt 1
[0527_17:34:38_596]#chip[EM] >>> [E:19128r S:52823 M:46751897] (S) Msg RX from 1:000000002D4A61A6 [F6D3] --- Type 0001:02 (IM:ReadRequest)
[0527_17:34:38_611]#chip[EM] >>> [E:19129r S:52823 M:46751898] (S) Msg RX from 1:000000002D4A61A6 [F6D3] --- Type 0001:03 (IM:SubscribeRequest)
[0527_17:34:38_617]#chip[DMG] AccessControl: denied
[0527_17:34:38_618]#chip[DMG] AccessControl: denied
[0527_17:34:38_619]#chip[DMG] AccessControl: denied
[0527_17:34:38_619]#chip[DMG] AccessControl: denied
[0527_17:34:38_628]#chip[DMG] AccessControl: denied
[0527_17:34:38_628]#chip[DMG] AccessControl: denied
[0527_17:34:38_629]#chip[DMG] AccessControl: denied
[0527_17:34:38_639]#chip[DMG] AccessControl: denied
[0527_17:34:38_639]#chip[DMG] AccessControl: denied
[0527_17:34:38_640]#chip[DMG] AccessControl: denied
[0527_17:34:38_641]#chip[DMG] AccessControl: denied
[0527_17:34:38_651]#chip[DMG] AccessControl: denied
[0527_17:34:38_652]#chip[DMG] AccessControl: denied
[0527_17:34:38_653]#chip[DMG] AccessControl: denied
[0527_17:34:38_653]#chip[DMG] AccessControl: denied
[0527_17:34:38_662]#chip[DMG] AccessControl: denied
[0527_17:34:38_663]#chip[DMG] AccessControl: denied
[0527_17:34:38_663]#chip[DMG] AccessControl: denied
[0527_17:34:38_665]#chip[DMG] AccessControl: denied
[0527_17:34:38_672]#chip[DMG] AccessControl: denied
[0527_17:34:38_673]#chip[DMG] AccessControl: denied
[0527_17:34:38_673]#chip[DMG] AccessControl: denied
[... 231 lines cut ...]
[0527_17:34:40_084]#chip[DMG] AccessControl: denied
[0527_17:34:40_085]#chip[DMG] AccessControl: denied
[0527_17:34:40_085]#chip[DMG] AccessControl: denied
[0527_17:34:40_093]#chip[DMG] AccessControl: denied
```

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

0ff709f25f6a7af68314c1b0a147a76c30c234f6

### Platform

ameba, darwin

### Platform Version(s)

1.2.0.1, iphone(ios 17.5.1),Homepod Mini(17.5 21L569)

### Type

Platform Issue

bzbarsky-apple, 2024-06-03:
@lxj199722 Please file a feedback assistant ticket (which would include logs from the Home app side, and ideally corresponding logs from the device side) following the directions below.  Please post the feedback assistant ticket ID here and I'll get it followed up on.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

lxj199722, 2024-06-04:
> @lxj199722 Please file a feedback assistant ticket (which would include logs from the Home app side, and ideally corresponding logs from the device side) following the directions below. Please post [...]

Thanks, this is problem matter device log.
[ACL-denied-light-demo.log](https://github.com/user-attachments/files/15543442/ACL-denied-light-demo.log)

bzbarsky-apple, 2024-06-08:
@lxj199722 Did you file the feedback assistant ticket?

lxj199722, 2024-06-11:
> @lxj199722 Did you file the feedback assistant ticket?

Yes, I am. I also feedback on Radar Chirp.

bzbarsky-apple, 2024-06-11:
@lxj199722 What is the feedback ticket ID?

lxj199722, 2024-06-11:
> @lxj199722 What is the feedback ticket ID?

FB13826995

bzbarsky-apple, 2024-06-12:
Thanks; this is being investigated, but very much looks like a bug in Home, not in the Matter SDK: Home is setting up the wrong ACLs and then has no access.

Closing this, and the fix needs to be tracked via the Apple feedback ticket.

closed 2024-06-12 by bzbarsky-apple

yongfa8Wong, 2025-05-13:
Did you solve it ? it also occur to me

lxj199722, 2025-05-27:
> Did you solve it ? it also occur to me

It maybe a bug in apple home app. User can create a new home, homepod and matter device in new home will solve it.