project-chip/connectedhomeip #33727: [BUG] CHIP Error 172

iVoGia, 2024-06-04:
### Reproduction steps

Step:
1/ Scan QR. 
2/ Start pairing device.

### Bug prevalence

Newbie

### GitHub hash of the SDK that was being used

matter device

### Platform

darwin, other

### Anything else?

I'm following the struct guide: https://developer.apple.com/documentation/mattersupport/adding-matter-support-to-your-ecosystem
I met the error as below. 
Pls take a look at below image. 
Pls help
[image]
 me 
Thanks

bzbarsky-apple, 2024-06-04:
@iVoGia That looks like the device has its certs signed by a PAA that is not known to the commissioner (101 during device attestation is kPaaNotFound).

What PAA is being used here?

bzbarsky-apple, 2024-06-04:
And what list of PAAs did you provide when starting the controller or controller factory (depending on which exact controller startup API you are using)?

iVoGia, 2024-06-05:
Hi @bzbarsky-apple , 
The factory defined as below:
[image]

iVoGia, 2024-06-05:
> @iVoGia That looks like the device has its certs signed by a PAA that is not known to the commissioner (101 during device attestation is kPaaNotFound).
> 
> What PAA is being used here?

What's the PAA, can you explain for me ? 
The device is Start Lightbulb.

iVoGia, 2024-06-05:
How to I can get device list from Matter Accessories in the Setting ? 
Get all from Fabric ?

bzbarsky-apple, 2024-06-08:
OK, so in the factory params you are leaving `productAttestationAuthorityCertificates` as nil.  The docs say:
```
/*
 * The Product Attestation Authority certificates that are trusted to sign
 * device attestation information (and in particular to sign Product Attestation
 * Intermediate certificates, which then sign Device Attestation Certificates).
 *
 * Defaults to nil.
 */
```

So if you leave that nil, it will fall back to the test-only PAA being trusted, so you will not be able to verify attestation on any production devices.

> What's the PAA, can you explain for me ?

See document comment above, or the Matter specification, but basically it's a certificate signing root for attestation certificates.

The intent is that you select which PAAs you trust (possibly all the ones in the DCL, but it's up to you; it's a policy decision).

See also documentation at https://developer.apple.com/documentation/matter/onboarding_a_matter_device which says, as one of the things you need to do:

> Set the factory parameter object’s [productAttestationAuthorityCertificates](https://developer.apple.com/documentation/matter/mtrdevicecontrollerfactoryparams/4122941-productattestationauthoritycerti) to an array of trusted Matter certificates.

closed 2024-06-08 by bzbarsky-apple

iVoGia, 2024-06-08:
Hi @bzbarsky-apple  ,
Thank for your explained. I got it. 
I passed the paa cert for testing, and continues commission device without PAA in the AttestationDelegate. 
And now, I can commission device successfull for test. 
Pls take a look at below image. 
[image]
And I can establish a connection to device in a RequestHandle.swift. 
But, when I call MTRGetConnectedDevice method , it’s always return error: “ cannot establish a connection with timeout.”
Pls help me for above error. 
Thanks.

bzbarsky-apple, 2024-06-10:
> But, when I call MTRGetConnectedDevice method , it’s always return error: “ cannot establish a connection with timeout.”

I don't know what MTRGetConnectedDevice is, so hard to say anything useful about it.