project-chip/connectedhomeip #33728: [1.3] chip-energy-management-app report attribute forever after reboot the application before subscription timeout

wilbur-se, 2024-06-04:
### Reproduction steps

1.start chip-tool in interactive mode
2.pairing chip-energy-management-app using chip-tool
3.subscribe all attribute: any subscribe-all 0xffffffff,0xffffffff 0xffffffff 0xffffffff 1 20 1 0xffff,0xffff
4.reboot chip-energy-management-app before subscription timeout
5.chip-energy-management-app resume session and subscription
6.chip-energy-management-app report all attribute forever
[chip-energy-management-app.log](https://github.com/user-attachments/files/15546340/chip-energy-management-app.log)
[chip-tool.log](https://github.com/user-attachments/files/15546349/chip-tool.log)

### Bug prevalence

always

### GitHub hash of the SDK that was being used

matter 1.3

### Platform

core

### Type

Core SDK Performance Improvement

### Anything else?

From device log, LastReportGeneration and DirtyGeneration will never be same.

[1717494922.879495][1864160:1864160] CHIP:DMG: Building Reports for ReadHandler with LastReportGeneration = 0x0000000000000035 DirtyGeneration = 0x0000000000000036

bzbarsky-apple, 2024-06-04:
Something keeps marking attributes dirty.  What is that thing and why is it doing it?

@wilbur-se Can you try breakpointing in `ReadHandler::AttributePathIsDirty` while this loop is going on and seeing what the callstack looks like?

wilbur-se, 2024-06-05:
print endpoint,cluster,endpoint in AttributePathIsDirty, cluster 0x99 with attribute 0x41 always print. And always have error in IncreaseClusterDataVersion before the output

[1717550714.988927][1866900:1866900] CHIP:DMG: Endpoint 0, Cluster 0x0000_0099 not found in IncreaseClusterDataVersion!
[1717550714.988930][1866900:1866900] CHIP:DMG: AttributePathIsDirty ep 0 clt 99 attr 41

bzbarsky-apple, 2024-06-08:
Yes, I see that in the log.  The question is _why_ the data version is increasing.  What's marking things dirty all the time?

wilbur-se, 2024-06-19:
checked the code, call RecalculateSessionDuration (EvseSession in EnergyEvseDelegateImpl.cpp) with wrong mEndpointId (0 but should be 1).
For func EnergyEvseInit in EnergyEvseMain.cpp, ENERGY_EVSE_ENDPOINT will not synced to EvseSession which only initilized when EnergyEvseDelegate created.

jamesharrow, 2024-10-17:
I believe this has been fixed - see https://github.com/project-chip/connectedhomeip/pull/35339

jamesharrow, 2024-10-17:
Closing as fixed.

closed 2024-10-17 by jamesharrow