project-chip/connectedhomeip #33754: [BUG] Commissioning failed with src/app/CommandSender.cpp:328: CHIP Error 0x00000032: Timeout

Saravana-kr22, 2024-06-05:
### Reproduction steps

I have used python controller to done a stress test on my RPI sample dut (all-clusters-app) on Pair-Unpair operation for 10000 iteration. The iteration 2570 is passed without any error but the 2571 is failed to commission the dut during the SendNOC step with the below traceback error and the next iteration 2572 is also passed without any issues.

> Traceback (most recent call last):
  File "/Users/chaitanyajandhyala/code/python/matter-qa/src/matter_qa/library/helper_libs/matter_testing_support.py", line 1177, in pair_dut
    raise CommissionFailure(f"Pairing with DUT failed Reason {commission_device_result.__str__()} ")
chip.commissioning.CommissionFailure: CommissionFailure(Pairing with DUT failed Reason src/app/CommandSender.cpp:328: CHIP Error 0x00000032: Timeout )

### Bug prevalence

Whenever i do this

### GitHub hash of the SDK that was being used

3718e996b56f36a553c1b4571f77728396590776

### Platform

python, raspi

### Platform Version(s)

RPI-4, 8GB RAM

### Anything else?

Example of Failed iteartion details:

The iteartion number 2570 is passed below file for the same
TH log: [log_2024-04-26_04-12-01.log](https://github.com/user-attachments/files/15588137/log_2024-04-26_04-12-01.log)
DUT log: [Dut_log_2570_2024-04-26T04_12_11_598075.log](https://github.com/user-attachments/files/15588147/Dut_log_2570_2024-04-26T04_12_11_598075.log)

The iteration number 2571 is failed with this error below the logfile
TH log : [log_2024-04-26_04-12-21.log](https://github.com/user-attachments/files/15588107/log_2024-04-26_04-12-21.log)
DUT log: [Dut_log_2571_2024-04-26T04_17_27_934283.log](https://github.com/user-attachments/files/15588118/Dut_log_2571_2024-04-26T04_17_27_934283.log)

The iteartion number 2572 is passed below file for the same
TH log: [log_2024-04-26_04-17-36.log](https://github.com/user-attachments/files/15588171/log_2024-04-26_04-17-36.log)
Dut log: [Dut_log_2572_2024-04-26T04_17_45_908760.log](https://github.com/user-attachments/files/15588181/Dut_log_2572_2024-04-26T04_17_45_908760.log)

For the detailed summary please reference this json: [summary.json](https://github.com/user-attachments/files/15588197/summary.json)

Script reference: https://github.com/CHIP-Specifications/matter-qa/blob/main/src/matter_qa/scripts/reliability_scripts/TC_Pair.py

bzbarsky-apple, 2024-06-05:
OK, so on the DUT this is what I see:

```
[1714129942.591993][489230:489230] CHIP:EM: >>> [E:56552r S:51405 M:78484474] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
...
[1714129942.593713][489230:489230] CHIP:EM: <<< [E:56552r S:51405 M:44984139 (Ack:78484474)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[fe80::1887:8597:775f:fcc8%wlan0]:63338] --- Type 0000:10 (SecureChannel:StandaloneAck)
...
[1714129942.613441][489230:489230] CHIP:EM: <<< [E:56552r S:51405 M:44984140 (Ack:78484474)] (S) Msg TX to 1:FFFFFFFB00000000 [03F8] [UDP:[fe80::1887:8597:775f:fcc8%wlan0]:63338] --- Type 0001:09 (IM:InvokeCommandResponse)
...
[1714129942.960479][489230:489230] CHIP:EM: Retransmitting MessageCounter:44984140 on exchange 56552r Send Cnt 1
[1714129943.368354][489230:489230] CHIP:EM: Retransmitting MessageCounter:44984140 on exchange 56552r Send Cnt 2
[1714129943.975605][489230:489230] CHIP:EM: Retransmitting MessageCounter:44984140 on exchange 56552r Send Cnt 3
[1714129944.944039][489230:489230] CHIP:EM: Retransmitting MessageCounter:44984140 on exchange 56552r Send Cnt 4
[1714129946.426007][489230:489230] CHIP:EM: Failed to Send CHIP MessageCounter:44984140 on exchange 56552r sendCount: 4 max retries: 4
[1714130002.110713][489230:489230] CHIP:FS: Fail-safe timer expired
```

So it sent the response but never got an ack for it or any followup commands.

On the TH side, what I see is:

```
2024-04-26 04:12:22,509 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - <<< [E:56552i S:60591 M:78484474] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[fe80::dea6:32ff:feab:25e5%en1]:5540] --- Type 0001:08 (IM:InvokeCommandRequest)
...
2024-04-26 04:12:22,524 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - >>> [E:56552i S:60591 M:44984139 (Ack:78484474)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
...
2024-04-26 04:17:18,713 - chip.native.DMG - INFO - __init__.py:_RedirectToPythonLogging:40 - Time out! failed to receive invoke command response from Exchange: 56552i
```

So for some reason network communication broke down for a while there (at least 4s, since the DUT retransmitted over a 4s period), but without packet traces it's impossible to say whether the DUT failed to send, the TH failed to receive, or something in between failed to route.

cjandhyala, 2024-06-18:
closing the ticket for now, following-up with a new ticket..

closed 2024-06-18 by cjandhyala