project-chip/connectedhomeip #33776: [BUG] Buffer used by emberAfExternalAttributeWrite/Read callbacks is not cleared

kkasperczyk-no, 2024-06-06:
### Reproduction steps

I've faced the problem while testing the bridge application with CHIPTool. I created a dynamic endpoint and tried to modify it's node label:
1. Set node label to the TEST value $ bridgeddevicebasicinformation write node-label TEST 1 3
2. Verify that the label is correct $ bridgeddevicebasicinformation read node-label 1 3

The returned content is as expected:
[1717674111.287383][103802:103804] CHIP:TOO:   NodeLabel: TEST

3. Set node label to the value shorter than before, e.g - AB $ bridgeddevicebasicinformation write node-label AB 1 3
4. Verify the label value $ bridgeddevicebasicinformation read node-label 1 3

The returned content is incorrect
[1717674111.287383][103802:103804] CHIP:TOO:   NodeLabel: ABST

I investigated it a little bit and it seems the problem is that the same buffer is used for read and write operations: https://github.com/project-chip/connectedhomeip/blob/v1.3.0.0/src/app/util/ember-compatibility-functions.cpp#L160

What happens is that buffer is not cleared after the previous read attempt, so it contains TEST characters and then the next write operation overrides only first two bytes, because it does not add null terminator: https://github.com/project-chip/connectedhomeip/blob/v1.3.0.0/src/app/util/ember-compatibility-functions.cpp#L864. The problem is that `emberAfExternalAttributeWriteCallback` in such scenario does not provide information about the valid size.

The buffer content itself does not include null terminator, so we cannot just read the data until /0. The size is passed through attribute metadata, but it is set to 32 B (max node-label buffer size). I think it could be solved either by reducing the size passed through the metadata to actual data size or by adding null terminator at the end of the copied string.

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

v1.3.0.0 tag

### Platform

core

### Platform Version(s)

1.3

bzbarsky-apple, 2024-06-06:
> and then the next write operation overrides only first two bytes, because it does not add null terminator

It overrides 3 bytes: the buffer includes the size in it; first byte is the size, then there is data.  See https://github.com/project-chip/connectedhomeip/blob/v1.3.0.0/src/app/util/ember-compatibility-functions.cpp#L863

@kkasperczyk-no what does your code for handling the external attribute write look like?  If you are just memcopying the incoming data and then copying it back, it should all work (since the length will be correct at "2"), but it sounds like you are doing something different?

kkasperczyk-no, 2024-06-06:
It was my mistake, I treated the string in buffer as a C-string and I haven't noticed that this is pascal string with size encoded at the first byte. I think everything is fine. This issue can be closed.

closed 2024-06-06 by kkasperczyk-no