project-chip/connectedhomeip #33779: [BUG] Commission failed with src/app/ClusterStateCache.cpp:303: CHIP Error 0x00000010: Key not found

Saravana-kr22, 2024-06-06:
### Reproduction steps

I have used python controller to done a stress test on my RPI sample dut (all-clusters-app) on Pair-Unpair operation for 10000 iteration. The iteration 2763 is passed without any error but the 2764 is Failed to read the info such BasicCommissioningInfo, RegulatoryConfig , LocationCapability,  Breadcrumb during commission the dut with the below  error and the next iteration 2765 is also passed without any issues.

> 2024-04-26 05:40:48,629 - chip.native.CTL - ERROR - __init__.py:_RedirectToPythonLogging:38 - Failed to read BasicCommissioningInfo: src/app/ClusterStateCache.cpp:303: CHIP Error 0x00000010: Key not found
2024-04-26 05:40:48,630 - chip.native.CTL - ERROR - __init__.py:_RedirectToPythonLogging:38 - Failed to read RegulatoryConfig: src/app/ClusterStateCache.cpp:303: CHIP Error 0x00000010: Key not found
2024-04-26 05:40:48,630 - chip.native.CTL - ERROR - __init__.py:_RedirectToPythonLogging:38 - Failed to read LocationCapability: src/app/ClusterStateCache.cpp:303: CHIP Error 0x00000010: Key not found
2024-04-26 05:40:48,630 - chip.native.CTL - ERROR - __init__.py:_RedirectToPythonLogging:38 - Failed to read Breadcrumb: src/app/ClusterStateCache.cpp:303: CHIP Error 0x00000010: Key not found
2024-04-26 05:40:48,631 - chip.native.CTL - ERROR - __init__.py:_RedirectToPythonLogging:38 - Error parsing commissioning information
2024-04-26 05:40:48,631 - chip.native.CTL - ERROR - __init__.py:_RedirectToPythonLogging:38 - Error on commissioning step 'ReadCommissioningInfo2': 'src/app/ClusterStateCache.cpp:303: CHIP Error 0x00000010: Key not found'

### Bug prevalence

Whenver i do this

### GitHub hash of the SDK that was being used

3718e996b56f36a553c1b4571f77728396590776

### Platform

python, raspi

### Platform Version(s)

RPI-4, 8GB RAM

### Anything else?

Example of Failed iteartion details:

The iteartion number 2763 is passed below file for the same
TH log: [log_2024-04-26_05-36-23.log](https://github.com/user-attachments/files/15643667/log_2024-04-26_05-36-23.log)
DUT log: [Dut_log_2763_2024-04-26T05_36_32_752483.log](https://github.com/user-attachments/files/15643679/Dut_log_2763_2024-04-26T05_36_32_752483.log)

The iteration number 2764 is failed with this error below the logfile
TH log : [log_2024-04-26_05-36-46.log](https://github.com/user-attachments/files/15643452/log_2024-04-26_05-36-46.log)
DUT log: [Dut_log_2764_2024-04-26T05_40_57_865104.log](https://github.com/user-attachments/files/15643487/Dut_log_2764_2024-04-26T05_40_57_865104.log)

The iteartion number 2765 is passed below file for the same
TH log: [log_2024-04-26_05-41-07.log](https://github.com/user-attachments/files/15643527/log_2024-04-26_05-41-07.log)
Dut log: [Dut_log_2765_2024-04-26T05_41_16_933069.log](https://github.com/user-attachments/files/15643575/Dut_log_2765_2024-04-26T05_41_16_933069.log)

For the detailed summary please reference this json: [summary.json](https://github.com/user-attachments/files/15588197/summary.json)

Script reference: https://github.com/CHIP-Specifications/matter-qa/blob/main/src/matter_qa/scripts/reliability_scripts/TC_Pair.py

bzbarsky-apple, 2024-06-06:
The cluster state cache bit is just follow-on problems.  Here is the real issue, from the TH log:

```
2024-04-26 05:40:18,613 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - <<< [E:59991i S:60974 M:30600750] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[fe80::dea6:32ff:feab:25e4%en0]:5540] --- Type 0001:02 (IM:ReadRequest)
...
2024-04-26 05:40:18,621 - chip.native.DMG - ERROR - __init__.py:_RedirectToPythonLogging:38 - Time out! failed to receive report data from Exchange: 59991i
...
2024-04-26 05:40:19,177 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - Retransmitting MessageCounter:30600750 on exchange 59991i Send Cnt 1
2024-04-26 05:40:19,840 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - Retransmitting MessageCounter:30600750 on exchange 59991i Send Cnt 2
2024-04-26 05:40:20,765 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - Retransmitting MessageCounter:30600750 on exchange 59991i Send Cnt 3
2024-04-26 05:40:22,339 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - Retransmitting MessageCounter:30600750 on exchange 59991i Send Cnt 4
```

I don't know why we claim "time out" before we retransmit, but if you look at the DUT log, it didn't even manage to get an ack for its end of the PASE handshake:

```
[1714135007.042690][500980:500980] CHIP:EM: <<< [E:59990r S:0 M:84098057 (Ack:183265092)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[fe80::f:2352:63b5:7bd%eth0]:63338] --- Type 0000:40 (SecureChannel:StatusReport)
...
[1714135007.385579][500980:500980] CHIP:EM: Retransmitting MessageCounter:84098057 on exchange 59990r Send Cnt 1
[1714135007.742411][500980:500980] CHIP:EM: Retransmitting MessageCounter:84098057 on exchange 59990r Send Cnt 2
[1714135008.334451][500980:500980] CHIP:EM: Retransmitting MessageCounter:84098057 on exchange 59990r Send Cnt 3
[1714135009.307984][500980:500980] CHIP:EM: Retransmitting MessageCounter:84098057 on exchange 59990r Send Cnt 4
[1714135010.769943][500980:500980] CHIP:EM: Failed to Send CHIP MessageCounter:84098057 on exchange 59990r sendCount: 4 max retries: 4
```

The TH did get that StatusReport:

```
2024-04-26 05:36:46,945 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - >>> [E:59990i S:0 M:84098057 (Ack:183265092)] (U) Msg RX from 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
```

and sent a standalone ack (but much later?  Why?):

```
2024-04-26 05:40:18,621 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - <<< [E:59990i S:0 M:183265093 (Ack:84098057)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[fe80::dea6:32ff:feab:25e4%en0]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck)
```

Anyway, it looks like the TH is doing something really weird with packets here for some reason.

bzbarsky-apple, 2024-06-06:
And in any case, the DUT did not get any packets from TH after sending that StatusReport.  Did not get the ack, did not get the reads.  So commissioning failed.

woody-apple, 2024-06-17:
This same error is showing up outside the TH, I don't think this is a TH issue.

cjandhyala, 2024-06-18:
closing the ticket for now, following-up with a new ticket..

closed 2024-06-18 by cjandhyala