project-chip/connectedhomeip #33792: [BUG] Chip-Cert-Bins docker container script failing on TOT Master

raju-apple, 2024-06-06:
### Reproduction steps

docker buildx build --build-arg COMMITHASH=3d7e23e8f610adcf93b40fe3d94f86f7d3819a1c --platform linux/arm64 --tag connectedhomeip/chip-cert-bins:3d7e23e8f610adcf93b40fe3d94f86f7d3819a1c --push .

### Bug prevalence

Every time

### GitHub hash of the SDK that was being used

3d7e23e8f610adcf93b40fe3d94f86f7d3819a1c

### Platform

other

### Anything else?

Seeing this failure : 

ERROR: failed to solve: process "/bin/bash -c case ${TARGETPLATFORM} in     \"linux/amd64\")     set -x     && source scripts/activate.sh     && scripts/build/build_examples.py     --target linux-x64-chip-tool-ipv6only-platform-mdns     --target linux-x64-shell-ipv6only-platform-mdns     --target linux-x64-chip-cert-ipv6only-platform-mdns     --target linux-x64-all-clusters-ipv6only     --target linux-x64-all-clusters-ipv6only-nlfaultinject     --target linux-x64-all-clusters-minimal-ipv6only     --target linux-x64-bridge-ipv6only     --target linux-x64-tv-app-ipv6only     --target linux-x64-tv-casting-app-ipv6only     --target linux-x64-light-ipv6only     --target linux-x64-thermostat-ipv6only     --target linux-x64-ota-provider-ipv6only     --target linux-x64-ota-requestor-ipv6only     --target linux-x64-lock-ipv6only     --target linux-x64-simulated-app1-ipv6only     --target linux-x64-lit-icd-ipv6only     --target linux-x64-energy-management-ipv6only     --target linux-x64-microwave-oven-ipv6only     build     && mv out/linux-x64-chip-tool-ipv6only-platform-mdns/chip-tool out/chip-tool     && mv out/linux-x64-shell-ipv6only-platform-mdns/chip-shell out/chip-shell     && mv out/linux-x64-chip-cert-ipv6only-platform-mdns/chip-cert out/chip-cert     && mv out/linux-x64-all-clusters-ipv6only/chip-all-clusters-app out/chip-all-clusters-app     && mv out/linux-x64-all-clusters-ipv6only-nlfaultinject/chip-all-clusters-app out/chip-all-clusters-app-nlfaultinject     && mv out/linux-x64-all-clusters-minimal-ipv6only/chip-all-clusters-minimal-app out/chip-all-clusters-minimal-app     && mv out/linux-x64-bridge-ipv6only/chip-bridge-app out/chip-bridge-app     && mv out/linux-x64-tv-app-ipv6only/chip-tv-app out/chip-tv-app     && mv out/linux-x64-tv-casting-app-ipv6only/chip-tv-casting-app out/chip-tv-casting-app     && mv out/linux-x64-light-ipv6only/chip-lighting-app out/chip-lighting-app     && mv out/linux-x64-thermostat-ipv6only/thermostat-app out/thermostat-app     && mv out/linux-x64-ota-provider-ipv6only/chip-ota-provider-app out/chip-ota-provider-app     && mv out/linux-x64-ota-requestor-ipv6only/chip-ota-requestor-app out/chip-ota-requestor-app     && mv out/linux-x64-lock-ipv6only/chip-lock-app out/chip-lock-app     && mv out/linux-x64-simulated-app1-ipv6only/chip-app1 out/chip-app1     && mv out/linux-x64-lit-icd-ipv6only/lit-icd-app out/lit-icd-app     && mv out/linux-x64-energy-management-ipv6only/chip-energy-management-app out/chip-energy-management-app     && mv out/linux-x64-microwave-oven-ipv6only/chip-microwave-oven-app out/chip-microwave-oven-app     ;;     \"linux/arm64\")    set -x     && source scripts/activate.sh     && scripts/build/build_examples.py     --target linux-arm64-chip-tool-ipv6only-platform-mdns     --target linux-arm64-shell-ipv6only-platform-mdns     --target linux-arm64-chip-cert-ipv6only-platform-mdns     --target linux-arm64-all-clusters-ipv6only     --target linux-arm64-all-clusters-ipv6only-nlfaultinject     --target linux-arm64-all-clusters-minimal-ipv6only     --target linux-arm64-bridge-ipv6only     --target linux-arm64-tv-app-ipv6only     --target linux-arm64-tv-casting-app-ipv6only     --target linux-arm64-light-ipv6only     --target linux-arm64-thermostat-ipv6only     --target linux-arm64-ota-provider-ipv6only     --target linux-arm64-ota-requestor-ipv6only     --target linux-arm64-lock-ipv6only     --target linux-arm64-simulated-app1-ipv6only     --target linux-arm64-lit-icd-ipv6only     --target linux-arm64-energy-management-ipv6only     --target linux-arm64-microwave-oven-ipv6only     build     && mv out/linux-arm64-chip-tool-ipv6only-platform-mdns/chip-tool out/chip-tool     && mv out/linux-arm64-shell-ipv6only-platform-mdns/chip-shell out/chip-shell     && mv out/linux-arm64-chip-cert-ipv6only-platform-mdns/chip-cert out/chip-cert     && mv out/linux-arm64-all-clusters-ipv6only/chip-all-clusters-app out/chip-all-clusters-app     && mv out/linux-arm64-all-clusters-ipv6only-nlfaultinject/chip-all-clusters-app out/chip-all-clusters-app-nlfaultinject     && mv out/linux-arm64-all-clusters-minimal-ipv6only/chip-all-clusters-minimal-app out/chip-all-clusters-minimal-app     && mv out/linux-arm64-bridge-ipv6only/chip-bridge-app out/chip-bridge-app     && mv out/linux-arm64-tv-app-ipv6only/chip-tv-app out/chip-tv-app     && mv out/linux-arm64-tv-casting-app-ipv6only/chip-tv-casting-app out/chip-tv-casting-app     && mv out/linux-arm64-light-ipv6only/chip-lighting-app out/chip-lighting-app     && mv out/linux-arm64-thermostat-ipv6only/thermostat-app out/thermostat-app     && mv out/linux-arm64-ota-provider-ipv6only/chip-ota-provider-app out/chip-ota-provider-app     && mv out/linux-arm64-ota-requestor-ipv6only/chip-ota-requestor-app out/chip-ota-requestor-app     && mv out/linux-arm64-lock-ipv6only/chip-lock-app out/chip-lock-app     && mv out/linux-arm64-simulated-app1-ipv6only/chip-app1 out/chip-app1     && mv out/linux-arm64-lit-icd-ipv6only/lit-icd-app out/lit-icd-app     && mv out/linux-arm64-energy-management-ipv6only/chip-energy-management-app out/chip-energy-management-app     && mv out/linux-arm64-microwave-oven-ipv6only/chip-microwave-oven-app out/chip-microwave-oven-app     ;;     *) ;;     esac" did not complete successfully: exit code: 1

[docker_failure.txt](https://github.com/user-attachments/files/15687558/docker_failure.txt)

Attached full log to the bug

cecille, 2024-06-06:
Timeout maybe? #0 676.1 2024-06-06 23:14:32 INFO    aarch64-linux-gnu-g++: fatal error: Killed signal terminated program cc1plus

cecille, 2024-06-06:
Internet also suggests perhaps an oom

andy31415, 2024-06-07:
When we had killed gcc on android it was OOM due to templates.z

andy31415, 2024-06-07:
I was able to build ToT master on my own machine. It seems very likely that this is a out of memory issue on the machine that was used to try the build.

raju-apple, 2024-06-07:
Closing this issue , was able to run it successfully on TOT after re-starting docker as well

closed 2024-06-07 by raju-apple