project-chip/connectedhomeip #33793: [1.3] Building Darwin tool on latest commits giving error response

Garik098, 2024-06-07:
### Reproduction steps

While trying to build darwin tool on tot or recent commits is giving below error.

note: Run script build phase 'Run GN Build (chip_xcode_build_connector)' will be run during every build because the option to run the script phase "Based on dependency analysis" is unchecked. (in target 'Matter' from project 'Matter')
2024-06-06 17:30:04.742 xcodebuild[79779:49910172] DVTCoreDeviceEnabledState: DVTCoreDeviceEnabledState_Primary set via user default (DVTEnableCoreDevice=primary)
2024-06-06 17:30:04.744 xcodebuild[79779:49910172] [MT] DVTDevice: Adjusting available devices (inserting): added = {(
    <IDELinuxSupportCore.DVTRemoteLinuxComputer: 0x12a306910 (localhost:4000 | x86_64)>
)} removed = (null)
)}
** BUILD FAILED **

The following build commands failed:
	CompileC /Users/adk/Desktop/1.2/connectedhomeip/examples/darwin-framework-tool/out/debug/macos_framework_output/Build/Intermediates.noindex/Matter.build/Debug/Matter.build/Objects-normal/arm64/MTRDeviceConnectionBridge.o /Users/adk/Desktop/1.2/connectedhomeip/src/darwin/Framework/CHIP/MTRDeviceConnectionBridge.mm normal arm64 objective-c++ com.apple.compilers.llvm.clang.1_0.compiler (in target 'Matter' from project 'Matter')
(1 failure)

Build Framework Result: 65
ninja: build stopped: subcommand failed.

I went back to May 15th builds and I see the error to be appearing. 

This sha from Mar 29th seem to be building fine - d09b5ac98f4d7d8b9f2c307f55ab5462576623a5

I am using the below commands to build the binary,

gn gen out/debug --args='chip_mdns="platform" chip_inet_config_enable_ipv4=false' 

ninja -C out/debug

Did the source scripts/bootstrap.sh and source scripts/activate.sh before running the above command. Did the Zap related steps as well. 

Chip tool on the other hand is building fine on the latest commits. 

### Bug prevalence

always

### GitHub hash of the SDK that was being used

5dd912f755f6374be063d41f8e0caac7247e64d2

### Platform

darwin

### Type

Platform Issue

ksperling-apple, 2024-06-07:
Darwin-framework-tool builds fine for me using both the `gn` steps you have described and via Xcode (target "Darwin-framework-tool" in Matter.xcodeproj), for both 5dd912f755 and ToT (3d7e23e8f610adcf93b40fe3d94f86f7d3819a1c).

It's hard to tell what's going on on your system since the actual error isn't included in the excerpt of the build output above or which version of Xcode you're building with.

bzbarsky-apple, 2024-06-07:
The actual error (from Slack) is:

```
src/darwin/Framework/CHIP/MTRDeviceConnectionBridge.mm:34:94: error: use of undeclared identifier 'MSEC_PER_SEC'
        retryDelay = @(static_cast<double>(failureInfo.requestedBusyDelay.Value().count()) / MSEC_PER_SEC);
                                                                                             ^
1 error generated.
```

closed 2024-06-07 by mergify[bot]