project-chip/connectedhomeip #33807: [Documentation] Flag List Change Actions Replace and Delete as provisional and document write List behavior

Apollon77, 2024-06-07:
### Documentation issues

As follow up to https://github.com/project-chip/connectedhomeip/issues/33487 I would propose to update "8.9.2.6. AttributeDataIB" and also 10.6.4.3.1. Lists
* mark "List change processing actions MODIFY and DELETE" as provisional because they are not implemented by chip
* maybe also add information for  "List change processing action ADD" that this is basically only supported with clearing the list before becaus when I understand code correctly then the list is cleared before the first value of a list is added

In general these limitations should be clear in the specification and also to make sure that, if it ever gets implemented, it is clear that model revisions needs to be increased because adding these new types is not backward compatible.

The same goes for the requirements on how write interactions should be working withg chunked lists (see linked other issue for examples). Main thing is:
* single entries are processed one by one
* a single write path can fail which just means that this action to process fails but not complete writing
* validation happens per write interaction, the same for "logical processing of the write"
* when all write actions for a particular path are done then the value is updated and becomes effective for potential other parts of the logic (e.g. acl)

This is also not contained in the specification from my knowledge.

Thank you

Apollon77, 2024-06-07:
PS: And the comment in https://github.com/project-chip/connectedhomeip/blob/5db170117d3afd5b4c815c1c181cc8646779cc5a/src/app/WriteHandler.cpp#L314-L316 is wrong because the first item is set as mListOp = ReplaceAll some line above that and so the result of isListItemOperation is false and never true

Means that "emptying the list" is not ignored but processed as a "ReplaceAll" operation.

The code login would ignore btw any first "Add/Delete/Modify" action being sent without a "clear" package before that when I understand the implications correctly, so they would be ignored with a Busy response (which is also nowhere documented in the specs)

ReneJosefsen, 2024-06-12:
Is any of this is covered by https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/9118?

Apollon77, 2024-06-15:
I can not answer this because no access

Apollon77, 2024-06-19:
@ReneJosefsen I commented on the mentioned PR. Please tell me if you like me to open anew Spec issue internally describing it more complete rather then commenting on the closed PR.

Apollon77, 2024-06-19:
Ok I close here, lets discuss detsils in spec issue

closed 2024-06-19 by Apollon77