project-chip/connectedhomeip #33818: [BUG] Issue with RequestHandler.swift

iVoGia, 2024-06-10:
### Reproduction steps

In my RequestHandler.swift, this is extension of `MatterAddDeviceExtensionRequestHandler`
After commission device is completion.
I call method `MTRGetConnectedDevice`

```
func controller(_ controller: MTRDeviceController, commissioningComplete error: Error?, nodeID: NSNumber?) {
        if error != nil {
            os_log(.error, "TrinhVM: commissioningComplete error -> \(error!.localizedDescription)")
        } else {
            os_log(.error, "TrinhVM: commissioningComplete ->\(nodeID)")
            if MTRGetConnectedDevice({ (chipDevice, error) in
                if chipDevice == nil {
                    os_log(.debug, "Status: Failed to establish a connection with the device")
                } else {
                    os_log(.error, "Status: Success to establish a connection with the device: \(chipDevice?.description ?? "")")
                }
            }) {
                print("Status: Waiting for connection with the device")
            } else {
                print("Status: Failed to trigger the connection with the device")
            }
        }
}
```

It's OK, always `Status: Success to establish a connection with the device`. 

But, when I call which method on the screen of DemoApplication(App scheme) to get device to control it, it's always error with timeout.
```
"Waiting for connection with the device"

Mdns: Resolve failure (src/platform/Darwin/DnssdImpl.cpp:476: CHIP Error 0x00000074: The operation has been cancelled)
OperationalSessionSetup[1:0000000000000015]: operational discovery failed: src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:119: CHIP Error 0x00000032: Timeout
Creating NSError from src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:119: CHIP Error 0x00000032: Timeout (context: (null))
"Failed to establish a connection with the device Optional(Error Domain=MTRErrorDomain Code=9 \"Transaction timed out.\" UserInfo={NSLocalizedDescription=Transaction timed out.})"
```

I don't know why it's happened. 
Pls help me for this issue.

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

matter device, darwin, matter support, ios

### Platform

darwin

bzbarsky-apple, 2024-06-10:
Please file a feedback assistant ticket for this and report the ticket ID here.  But also, without knowing what `MTRGetConnectedDevice` does, it's impossible to say what this code is actually doing.

In any case, this does not seem like a Matter SDK problem.

closed 2024-06-10 by bzbarsky-apple

iVoGia, 2024-06-11:
Hi @bzbarsky-apple , 
I created a feedback assistant ticket with Id: "FB13828664", 
MTRGetConnectedDevice -> just call getBaseDevice method from MTRDeviceController.
Pls help me . 
Thanks .

iVoGia, 2024-06-11:
Call `getBaseDevice` from `MTRDeviceController `.
The message output is `Waiting for connection with the device` and final with timeout. 
Although, it's working well on the RequestHandler.swift (MatterAddDeviceExtensionRequestHandler)

bzbarsky-apple, 2024-06-13:
@iVoGia That ticket does not have any logs attached.  Please follow the instructions below to capture logs.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.