project-chip/connectedhomeip #33829: [BUG] Failedto read attribute with src/app/ReadClient.cpp:343: CHIP Error 0x00000077: Missing secure session

Saravana-kr22, 2024-06-10:
### Reproduction steps

I have used python controller to done a stress test on my RPI sample dut (all-clusters-app) on Read-Write operation for 100000 iteration. The iteration 12580 is passed without any error but the 12581 is Failed to read the node-label attribute with the below traceback error and the next iteration 12582 is also passed without any issues.

> Traceback (most recent call last):
  File "/Users/chaitanyajandhyala/code/python/matter-qa/src/matter_qa/scripts/reliability_scripts/TC_Read_Write.py", line 72, in verify_read_write
    nodelabel_read_value = await self.read_single_attribute_check_success(cluster=self.clusters,
  File "/Users/chaitanyajandhyala/code/python/matter-qa/src/matter_qa/library/helper_libs/matter_testing_support.py", line 876, in read_single_attribute_check_success
    result = await dev_ctrl.ReadAttribute(node_id, [(endpoint, attribute)], fabricFiltered=fabric_filtered)
  File "/Users/chaitanyajandhyala/code/python/matter-qa/.venv/lib/python3.9/site-packages/chip/ChipDeviceCtrl.py", line 1421, in ReadAttribute
    res = await self.Read(nodeid,
  File "/Users/chaitanyajandhyala/code/python/matter-qa/.venv/lib/python3.9/site-packages/chip/ChipDeviceCtrl.py", line 1352, in Read
    ClusterAttribute.Read(future=future, eventLoop=eventLoop, device=device.deviceProxy, devCtrl=self,
  File "/Users/chaitanyajandhyala/code/python/matter-qa/.venv/lib/python3.9/site-packages/chip/native/__init__.py", line 87, in raise_on_error
    raise self.to_exception()
chip.exceptions.ChipStackError: src/app/ReadClient.cpp:343: CHIP Error 0x00000077: Missing secure session

### Bug prevalence

Whenever i do this

### GitHub hash of the SDK that was being used

3718e996b56f36a553c1b4571f77728396590776

### Platform

python, raspi

### Platform Version(s)

RPI-4, 8GB RAM

### Anything else?

Example of Failed iteartion details:

The iteartion number 12580 is passed below file for the same
TH log: [log_2024-05-28_22-24-05.log](https://github.com/user-attachments/files/15775398/log_2024-05-28_22-24-05.log)

The iteration number 12581 is failed with this error below the logfile
TH log : [log_2024-05-28_22-24-31.log](https://github.com/user-attachments/files/15775397/log_2024-05-28_22-24-31.log)

The iteartion number 12582 is passed below file for the same
TH log: [log_2024-05-28_22-24-52.log](https://github.com/user-attachments/files/15775401/log_2024-05-28_22-24-52.log)

For the detailed summary please reference this json: [summary.json](https://github.com/user-attachments/files/15775309/summary.json)

Script reference: https://github.com/CHIP-Specifications/matter-qa/blob/main/src/matter_qa/scripts/reliability_scripts/TC_Read_Write.py

tcarmelveilleux, 2024-06-11:
Please re-execute and validate whether this always happens at 12580.

tcarmelveilleux, 2024-06-11:
Where are the DUT logs for these runs? We need the DUT logs to diagnose.

Saravana-kr22, 2024-06-11:
@tcarmelveilleux Currently we didn't have the dut log for this error will try to fetch the dut log from the next run

cjandhyala, 2024-06-17:
closing this issue for now and following it up in a different issue.

closed 2024-06-17 by cjandhyala