project-chip/connectedhomeip #33840: Microwave Oven Control Cluster - SetCookingParameters Command with valid PowerSetting  value shows the Invalid command error

Ashwinigrl, 2024-06-11:
### Feature Area

Other

### Test Case

MWOCTRL

### Reproduction steps

1. Execute the below mentioned command to put DUT into a commissionable state : `./chip-all-clusters-app`
2. Once DUT reach the commission able state sent below mentioned command on TH. : `./chip-tool pairing onnetwork 1 20202021`
3. Once after successful pairing provide the below commands:

4. **Read the MinPower value**

`./chip-tool microwaveovencontrol read min-power 1 1`

> [1717478153.817220][5912:5914] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_005F Attribute 0x0000_0003 DataVersion: 725732553
> [1717478153.820746][5912:5914] CHIP:TOO:   MinPower: 20

5. **Read the MaxPower value** 

`./chip-tool microwaveovencontrol read max-power 1 1`

> [1717478469.790067][5933:5935] CHIP:DMG: }
> [1717478469.790200][5933:5935] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_005F Attribute 0x0000_0004 DataVersion: 725732553
> [1717478469.790256][5933:5935] CHIP:TOO:   MaxPower: 90

6. **Send SetCookingParameters command to the DUT with Power setting field value is between MinPower to MaxPower.
Consider Power setting field value as 25.**

**Actual Outcome:**

`./chip-tool microwaveovencontrol set-cooking-parameters  1 1 --PowerSetting 25`

> [1718085750.218808][7110:7112] CHIP:DMG: 				StatusIB =
> [1718085750.218883][7110:7112] CHIP:DMG: 				{
> [1718085750.218959][7110:7112] CHIP:DMG: 					status = 0x85 (INVALID_COMMAND),
> [1718085750.219032][7110:7112] CHIP:DMG: 				},
> 

**Expected Outcome:**

`./chip-tool microwaveovencontrol set-cooking-parameters  1 1 --PowerSetting 25`

> [1718085862.583622][7637:7639] CHIP:DMG: 				StatusIB =
> [1718085862.583716][7637:7639] CHIP:DMG: 				{
> [1718085862.583795][7637:7639] CHIP:DMG: 					status = 0x00 (SUCCESS),
> [1718085862.583871][7637:7639] CHIP:DMG: 				},

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

 c15826050e744034ec9c90cf810fb2b34bb46c54

### Platform

raspi

### Anything else?

As per the Spec , In `SetCookingParameters` Command the `PowerSetting `Field value range is from Min Power to Max Power value.
In current SDK the `MinPower value is 20`, and `MaxPower is 90`.

I tried to send the  SetCookingParameters Command with PowerSetting field value as 25(Value between MinPower to MaxPower) the status value is **INVALID_COMMAND** in stead of **SUCCESS**.

**Reference log:**
[MWOCTRL.txt](https://github.com/user-attachments/files/15783566/MWOCTRL.txt)

**Spec References:**
-  https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/MicrowaveOvenControl.adoc#623-powersetting-field

**Test Plan References:**
- https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/MicrowaveOvenControl.adoc

tcarmelveilleux, 2024-06-11:
What is the value of `PowerStep` attribute? By default it's 10.

Per spec:

> 8.13.6.2.3. PowerSetting Field
>
> This field SHALL indicate the PowerSetting associated with the operation of the device. The value of this field SHALL be subject to the constraints of the PowerSetting attribute of this cluster. **If the PowerSetting field does not conform to the constraints of the PowerSetting attribute, the server SHALL return a CONSTRAINT_ERROR status.**

And then:

> 8.13.5.3. PowerSetting Attribute
> This attribute SHALL indicate the power level associated with the operation of the device.
> 
> If the MinPower, MaxPower, and PowerStep attributes are not supported:
> 
> The minimum value of this attribute SHALL be 10,
> 
> The maximum value of this attribute SHALL be 100,
> 
> The value SHALL be in even multiples of 10,
> 
> The default value SHALL be 100.
> 
> If the MinPower, MaxPower, and PowerStep attributes are supported:
> 
> The value of this attribute SHALL be between MinPower and MaxPower inclusive.
> 
> **The value of this attribute SHALL be such that (PowerSetting - MinPower) % PowerStep == 0**
> 

The actual bug is not that SUCCESS should have occurred, it's that CONSTRAINT_ERROR should have occurred instead of INVALID_COMMAND.

rbultman, 2024-06-11:
25 _may_ be valid depending on the value of PowerStep.  @Ashwinigrl which test case resulted in this failure?

cecille, 2024-06-11:
all-clusters doesn't have this cluster, assuming that this is running against the microwave-oven app, then the power step is indeed 10, and this should have returned a constraint error.

```
[1718134001.611476][3639415:3639417] CHIP:DMG: ReportDataMessage =
[1718134001.611488][3639415:3639417] CHIP:DMG: {
[1718134001.611498][3639415:3639417] CHIP:DMG: 	AttributeReportIBs =
[1718134001.611513][3639415:3639417] CHIP:DMG: 	[
[1718134001.611524][3639415:3639417] CHIP:DMG: 		AttributeReportIB =
[1718134001.611537][3639415:3639417] CHIP:DMG: 		{
[1718134001.611549][3639415:3639417] CHIP:DMG: 			AttributeDataIB =
[1718134001.611559][3639415:3639417] CHIP:DMG: 			{
[1718134001.611571][3639415:3639417] CHIP:DMG: 				DataVersion = 0xb845de10,
[1718134001.611582][3639415:3639417] CHIP:DMG: 				AttributePathIB =
[1718134001.611596][3639415:3639417] CHIP:DMG: 				{
[1718134001.611607][3639415:3639417] CHIP:DMG: 					Endpoint = 0x1,
[1718134001.611620][3639415:3639417] CHIP:DMG: 					Cluster = 0x5f,
[1718134001.611631][3639415:3639417] CHIP:DMG: 					Attribute = 0x0000_0005,
[1718134001.611644][3639415:3639417] CHIP:DMG: 				}
[1718134001.611658][3639415:3639417] CHIP:DMG: 					
[1718134001.611670][3639415:3639417] CHIP:DMG: 				Data = 10 (unsigned), 
[1718134001.611681][3639415:3639417] CHIP:DMG: 			},
[1718134001.611696][3639415:3639417] CHIP:DMG: 			
[1718134001.611707][3639415:3639417] CHIP:DMG: 		},
[1718134001.611719][3639415:3639417] CHIP:DMG: 		
[1718134001.611732][3639415:3639417] CHIP:DMG: 	],
[1718134001.611748][3639415:3639417] CHIP:DMG: 	
[1718134001.611758][3639415:3639417] CHIP:DMG: 	SuppressResponse = true, 
[1718134001.611770][3639415:3639417] CHIP:DMG: 	InteractionModelRevision = 11
[1718134001.611780][3639415:3639417] CHIP:DMG: }
```

It seems that the cert tests don't have any specific test for this right now, or the test is broken. If we want to change the spec, this will need either a cert test update or a cert test addition to check this condition so we don't end up with different behviour on non-sdk devices.

cecille, 2024-06-11:
Confirmed I do get a success error when I run the command with power setting that's a multiple of 10.
`./out/linux-x64-chip-tool/chip-tool microwaveovencontrol set-cooking-parameters  0x12344321 1 --PowerSetting 30`

```
[1718134887.158313][3650215:3650217] CHIP:DMG: InvokeResponseMessage =
[1718134887.158322][3650215:3650217] CHIP:DMG: {
[1718134887.158330][3650215:3650217] CHIP:DMG: 	suppressResponse = false, 
[1718134887.158337][3650215:3650217] CHIP:DMG: 	InvokeResponseIBs =
[1718134887.158351][3650215:3650217] CHIP:DMG: 	[
[1718134887.158359][3650215:3650217] CHIP:DMG: 		InvokeResponseIB =
[1718134887.158368][3650215:3650217] CHIP:DMG: 		{
[1718134887.158376][3650215:3650217] CHIP:DMG: 			CommandStatusIB =
[1718134887.158385][3650215:3650217] CHIP:DMG: 			{
[1718134887.158394][3650215:3650217] CHIP:DMG: 				CommandPathIB =
[1718134887.158403][3650215:3650217] CHIP:DMG: 				{
[1718134887.158411][3650215:3650217] CHIP:DMG: 					EndpointId = 0x1,
[1718134887.158419][3650215:3650217] CHIP:DMG: 					ClusterId = 0x5f,
[1718134887.158428][3650215:3650217] CHIP:DMG: 					CommandId = 0x0,
[1718134887.158437][3650215:3650217] CHIP:DMG: 				},
[1718134887.158448][3650215:3650217] CHIP:DMG: 				
[1718134887.158457][3650215:3650217] CHIP:DMG: 				StatusIB =
[1718134887.158466][3650215:3650217] CHIP:DMG: 				{
[1718134887.158476][3650215:3650217] CHIP:DMG: 					status = 0x00 (SUCCESS),
[1718134887.158483][3650215:3650217] CHIP:DMG: 				},
[1718134887.158493][3650215:3650217] CHIP:DMG: 				
[1718134887.158500][3650215:3650217] CHIP:DMG: 			},
[1718134887.158510][3650215:3650217] CHIP:DMG: 			
[1718134887.158518][3650215:3650217] CHIP:DMG: 		},
[1718134887.158529][3650215:3650217] CHIP:DMG: 		
[1718134887.158538][3650215:3650217] CHIP:DMG: 	],
[1718134887.158548][3650215:3650217] CHIP:DMG: 	
[1718134887.158556][3650215:3650217] CHIP:DMG: 	InteractionModelRevision = 11
[1718134887.158562][3650215:3650217] CHIP:DMG: },
```

rbultman, 2024-06-11:
This PR adds the negative checks to the test plan:
https://github.com/CHIP-Specifications/chip-test-plans/pull/4095/files

This PR fixes the SDK error and adds the negative tests:
https://github.com/project-chip/connectedhomeip/pull/33198/files

Ashwinigrl, 2024-06-27:
Issue fixed in Commit 07ac2f113c078b2659cf1b2b320776272070f99a. Hence closing this Issue.

closed 2024-06-27 by Ashwinigrl