project-chip/connectedhomeip #33842: [BUG] Can't connect Android CHPTVCastingApp with Android MatterTvServer

strangesource, 2024-06-11:
### Reproduction steps

1. Run the MatterTvAndroid `content-app`  and `platform-app` on a Google Chromecast HD (API 31)
2. Run the `CHIPTVCastingApp` on a Pixel 6a (API 34)
3. Observe Casting Player is discovered correctly
4. Click on observed casting player
5. Observe no further changes in the ui and 

### Bug prevalence

Every time

### GitHub hash of the SDK that was being used

c15826050e744034ec9c90cf810fb2b34bb46c54

### Platform

android

### Anything else?

I am trying to implement a PoC for using Matter casting with our Player SDK and I am running into issues when using the unmodified example projects.

strangesource, 2024-06-11:
The following logs can be observed on the platform app:
```
2024-06-11 09:17:55.171 11858-11858 DMG                     com.matter.tv.server                 D  Failed to read stored attribute (0, 0x0000_002C, 0x0000_0000: src/platform/android/KeyValueStoreManagerImpl.cpp:102: CHIP Error 0x000000A0: Value not found in the persisted storage
2024-06-11 09:17:55.172 11858-11858 KeyValueStoreManager    com.matter.tv.server                 D  Key 'g/a/0/2c/1' not found in shared preferences
2024-06-11 09:17:55.172 11858-11858 DMG                     com.matter.tv.server                 D  Failed to read stored attribute (0, 0x0000_002C, 0x0000_0001: src/platform/android/KeyValueStoreManagerImpl.cpp:102: CHIP Error 0x000000A0: Value not found in the persisted storage
2024-06-11 09:17:55.172 11858-11858 KeyValueStoreManager    com.matter.tv.server                 D  Key 'g/a/0/2d/0' not found in shared preferences
2024-06-11 09:17:55.172 11858-11858 DMG                     com.matter.tv.server                 D  Failed to read stored attribute (0, 0x0000_002D, 0x0000_0000: src/platform/android/KeyValueStoreManagerImpl.cpp:102: CHIP Error 0x000000A0: Value not found in the persisted storage
2024-06-11 09:17:55.174 11858-11858 KeyValueStoreManager    com.matter.tv.server                 D  Key 'g/a/2/8/4000' not found in shared preferences
2024-06-11 09:17:55.174 11858-11858 DMG                     com.matter.tv.server                 D  Failed to read stored attribute (2, 0x0000_0008, 0x0000_4000: src/platform/android/KeyValueStoreManagerImpl.cpp:102: CHIP Error 0x000000A0: Value not found in the persisted storage
2024-06-11 09:17:55.174 11858-11858 DMG                     com.matter.tv.server                 D  AccessControlCluster: initializing
2024-06-11 09:17:55.174 11858-11858 ZCL                     com.matter.tv.server                 D  Initiating Admin Commissioning cluster.
2024-06-11 09:17:55.174 11858-11858 DMG                     com.matter.tv.server                 D  Endpoint 0, Cluster 0x0000_0030 update version to 23bee4b7
2024-06-11 09:17:55.174 11858-11858 ZCL                     com.matter.tv.server                 E  DeviceInfoProvider is not registered
2024-06-11 09:17:55.174 11858-11858 ZCL                     com.matter.tv.server                 E  Trying to write invalid Calendar Type
2024-06-11 09:17:55.174 11858-11858 ZCL                     com.matter.tv.server                 E  Failed to write calendar type with error: 0x87
2024-06-11 09:17:55.174 11858-11858 ZCL                     com.matter.tv.server                 D  TV Android App::OnOff::PostClusterInit
2024-06-11 09:17:55.175 11858-11858 TvApp                   com.matter.tv.server                 D  postClusterInit for 6 at 1
2024-06-11 09:17:55.176 11858-11858 ZCL                     com.matter.tv.server                 D  TV Android App: OnOffManager::NewManager
2024-06-11 09:17:55.176 11858-11858 ZCL                     com.matter.tv.server                 D  TV Android App: WakeOnLan::PostClusterInit
2024-06-11 09:17:55.177 11858-11858 TvApp                   com.matter.tv.server                 D  postClusterInit for 1283 at 1
2024-06-11 09:17:55.178 11858-11858 ZCL                     com.matter.tv.server                 D  TV Android App: WakeOnLan::SetDefaultDelegate
2024-06-11 09:17:55.178 11858-11858 ZCL                     com.matter.tv.server                 D  TV Android App: Channel::PostClusterInit
2024-06-11 09:17:55.179 11858-11858 TvApp                   com.matter.tv.server                 D  postClusterInit for 1284 at 1
2024-06-11 09:17:55.187 11858-11858 ZCL                     com.matter.tv.server                 D  TV Android App: Channel::SetDefaultDelegate
2024-06-11 09:17:55.187 11858-11858 ZCL                     com.matter.tv.server                 D  TV Android App: TargetNavigator::SetDefaultDelegate
2024-06-11 09:17:55.188 11858-11858 ZCL                     com.matter.tv.server                 D  TV Android App: MediaPlayback::PostClusterInit
2024-06-11 09:17:55.189 11858-11858 TvApp                   com.matter.tv.server                 D  postClusterInit for 1286 at 1
2024-06-11 09:17:55.192 11858-11858 ZCL                     com.matter.tv.server                 D  TV Android App: MediaPlayback::SetDefaultDelegate
2024-06-11 09:17:55.192 11858-11858 ZCL                     com.matter.tv.server                 D  TV Android App: MediaInput::PostClusterInit
2024-06-11 09:17:55.193 11858-11858 TvApp                   com.matter.tv.server                 D  postClusterInit for 1287 at 1
2024-06-11 09:17:55.196 11858-11858 ZCL                     com.matter.tv.server                 D  TV Android App: MediaInput::SetDefaultDelegate
2024-06-11 09:17:55.196 11858-11858 ZCL                     com.matter.tv.server                 D  TV Android App: LowPower::PostClusterInit
2024-06-11 09:17:55.197 11858-11858 TvApp                   com.matter.tv.server                 D  postClusterInit for 1288 at 1
2024-06-11 09:17:55.198 11858-11858 ZCL                     com.matter.tv.server                 D  TV Android App: LowPower::SetDefaultDelegate
2024-06-11 09:17:55.198 11858-11858 ZCL                     com.matter.tv.server                 D  TV Android App: KeypadInput::PostClusterInit
2024-06-11 09:17:55.199 11858-11858 TvApp                   com.matter.tv.server                 D  postClusterInit for 1289 at 1
[... cut ...]
2024-06-11 09:19:29.739 11858-12120 IN                      com.matter.tv.server                 D  CDC msg sent
2024-06-11 09:19:29.739 11858-12120 CTL                     com.matter.tv.server                 D  ------PROMPT USER: Matter Casting Video Client is requesting permission to cast to this TV, approve? [0x0000_FFF1,0x0000_8003,973687D9930504AA,1C003F12A0AB3208E557F9BFC525593F0556]
2024-06-11 09:19:29.742 11858-12120 UserPrompterResolver    com.matter.tv.server                 D  Received prompt for OK permission vendor id:65521 productId:32771. Commissionee: Matter Casting Video Client
```

strangesource, 2024-06-11:
The following logs on the CHIPTVCastingApp:
```
'B1514E223C40CBBC' to /192.168.1.19, type : ._matterd._udp
2024-06-11 09:19:20.456 20075-27652 NsdManager...ceResolver com.chip.casting                     D  resolve: Timing out
2024-06-11 09:19:20.459 20075-27652 NsdManager...AvailState com.chip.casting                     D  Signaling NsdManager Resolver as free
2024-06-11 09:19:20.460 20075-20075 NsdManager...AvailState com.chip.casting                     D  Found NsdManager Resolver free, using it and marking it as busy
2024-06-11 09:19:23.463 20075-27657 NsdManager...ceResolver com.chip.casting                     D  resolve: Timing out
2024-06-11 09:19:23.465 20075-27657 NsdManager...AvailState com.chip.casting                     D  Signaling NsdManager Resolver as free
2024-06-11 09:19:23.492 20075-20115 NsdService...ndResolver com.chip.casting                     D  Service discovery started. regType: _matterd._udp
2024-06-11 09:19:23.493 20075-20075 DIS                     com.chip.casting                     D   ----- entry [0] : DN Küche
2024-06-11 09:19:23.493 20075-20075 DIS                     com.chip.casting                     D   ----- entry [1] : DT 35
2024-06-11 09:19:23.493 20075-20075 DIS                     com.chip.casting                     D   ----- entry [2] : VP 65521+32769
2024-06-11 09:19:23.493 20075-20075 DIS                     com.chip.casting                     D  Discovered commissionable/commissioner node:
2024-06-11 09:19:23.493 20075-20075 DIS                     com.chip.casting                     D  	Hostname: 192.168.1.19
2024-06-11 09:19:23.493 20075-20075 DIS                     com.chip.casting                     D  	IP Address #1: 192.168.1.19
2024-06-11 09:19:23.493 20075-20075 DIS                     com.chip.casting                     D  	Port: 5550
2024-06-11 09:19:23.494 20075-20075 DIS                     com.chip.casting                     D  	Mrp Interval idle: not present
2024-06-11 09:19:23.494 20075-20075 DIS                     com.chip.casting                     D  	Mrp Interval active: not present
2024-06-11 09:19:23.494 20075-20075 DIS                     com.chip.casting                     D  	Mrp Active Threshold: not present
2024-06-11 09:19:23.494 20075-20075 DIS                     com.chip.casting                     D  	TCP Supported: 0
2024-06-11 09:19:23.494 20075-20075 DIS                     com.chip.casting                     D  	ICD: not present
2024-06-11 09:19:23.494 20075-20075 DIS                     com.chip.casting                     D  	Device Name: Küche
2024-06-11 09:19:23.494 20075-20075 DIS                     com.chip.casting                     D  	Vendor ID: 65521
2024-06-11 09:19:23.494 20075-20075 DIS                     com.chip.casting                     D  	Product ID: 32769
2024-06-11 09:19:23.494 20075-20075 DIS                     com.chip.casting                     D  	Device Type: 35
2024-06-11 09:19:23.494 20075-20075 DIS                     com.chip.casting                     D  	Instance Name: B1514E223C40CBBC
2024-06-11 09:19:23.494 20075-20075 DIS                     com.chip.casting                     D  	Commissioning Mode: 0
2024-06-11 09:19:23.494 20075-20075 DIS                     com.chip.casting                     D  	Supports Commissioner Generated Passcode: false
2024-06-11 09:19:23.494 20075-20075 DIS                     com.chip.casting                     D  DeviceDiscoveryDelegateImpl::OnDiscoveredDevice() called
2024-06-11 09:19:23.494 20075-20075 SVR                     com.chip.casting                     D  Updated Casting Player
2024-06-11 09:19:23.494 20075-20075 SVR                     com.chip.casting                     D  CastingPlayerDiscovery-JNI DiscoveryDelegateImpl::HandleOnUpdated() called with CastingPlayer, ID: 192.168.1.195550
2024-06-11 09:19:23.494 20075-20075 SVR                     com.chip.casting                     D  convertCastingPlayerFromCppToJava() called
2024-06-11 09:19:23.494 20075-20115 NsdService...ndResolver com.chip.casting                     D  onServiceFound: found service not a target for resolution, ignoring name: B1514E223C40CBBC, type: _matterd._udp., subtypes: _T35, hostAddresses: , hostname: null, port: 0, network: 317, expirationTime: 2024-06-11T07:19:20Z, txtRecord: 
2024-06-11 09:19:23.495 20075-20075 CastingPla...geListener com.chip.casting                     I  onChanged() Discovered changes to CastingPlayer with deviceId: 192.168.1.195550
2024-06-11 09:19:23.496 20075-20115 NsdService...ndResolver com.chip.casting                     D  onServiceFound: found target service name: CD1F67B0FE4C95C5, type: _matterd._udp., subtypes: _T35, _V65521, hostAddresses: , hostname: null, port: 0, network: 317, expirationTime: 2024-06-11T07:19:20Z, txtRecord: 
2024-06-11 09:19:23.502 20075-20115 NsdService...ndResolver com.chip.casting                     I  Discovery stopped: _matterd._udp
2024-06-11 09:19:23.508 20075-20115 NsdService...ndResolver com.chip.casting                     I  Resolved service 'CD1F67B0FE4C95C5' to /192.168.3.87, type : ._matterd._udp
2024-06-11 09:19:29.523 20075-20075 DIS                     com.chip.casting                     D   ----- entry [0] : CP 1
2024-06-11 09:19:29.523 20075-20075 DIS                     com.chip.casting                     D   ----- entry [1] : DN Test TV Lukas
[... cut ...]
                                                                                                    CommissionerDeclaration:commissionerPasscode:     false
                                                                                                    CommissionerDeclaration::qRCodeDisplayed:         false
2024-06-11 09:19:32.231 20075-20118 MatterCallback          com.chip.casting                     E  MatterCallback::Caught an unhandled Throwable from the client: java.lang.NullPointerException: Attempt to invoke virtual method 'void androidx.fragment.app.FragmentActivity.runOnUiThread(java.lang.Runnable)' on a null object reference
```

Hiwensen, 2024-07-08:
Hi, any progress on this?
I encountered the same issue when trying to connect to a FireTV device from a phone, but I ran the `content-app`,
and didn't run the `platform-app` on TV
@strangesource Do you know if I should run the `platform-app` on the TV?

strangesource, 2024-07-08:
Unfortunately no progress, I haven't spend more time with this though. 

In general both, the content app and the platform app, are required. I am unsure though if that is the case for Fire TV devices if they already come with matter casting support build in. This ~scenario~ is unfortunately not very well documented.

Hiwensen, 2024-07-10:
@strangesource Thanks for your reply!
Yeah, the developer documentation is not good. It doesn't clearly state whether you need to run the tv-platform app on the TV devices to connect your phone to the TV. 
I encountered some build errors when running the tv-platform app. 
Planning to fix the errors and run the TV-platform app.

strangesource, 2024-07-10:
I did manage to get the `tv-platform` app to build so if you have any questions I might be able to help. :) 
Unfortunately I did not get much further, so if you figure something out kindly let me know.

Hiwensen, 2024-07-10:
Sure thing, will let you know when I find something useful. I'm also trying to implement a Matter Casting POC.
I built and ran `tv-casting` and `tv-content` apps successfully.
But the IDE prompts can't find some classes when I run the `tv-platform` app, looks like the missed classes are from some `.jar` or `.so` file, but the developer doc didn't say how to build those files.
The directory I opened with AndroidStudio is `connectedhomeip/examples/tv-app/android/App`
Do you know how to fix the build errors?

[image]

strangesource, 2024-07-10:
Did you use the `./scripts/build/build_examples.py` script with the correct target and architecture?

Hiwensen, 2024-07-11:
I used `./scripts/build/build_examples.py --target android-arm64-tv-casting-app build` to build the tv-casting app successfully.
Working on `./scripts/build/build_examples.py --target android-arm64-chip-tool build` to make it a success, now the error message is 
```
Class xxx was compiled with an incompatible version of Kotlin. The binary version of its metadata is 2.0.0, expected version is 1.8.0.
```
Should I run any other command?

strangesource, 2024-07-11:
I think you will need the `tv-server` target, not the chip tool.

The error hints towards Kotlin version incompatibility. Please make sure your project uses the same Koltin version (1.8.0 I think) as the project.

Hiwensen, 2024-07-11:
Many thanks!
I run `./scripts/build/build_examples.py --target android-arm-tv-server build` successfully.
Some fundings:
* A Mibox TV device that can't be discovered before can be discovered after installing the tv-platform app but still can't connect to the TV device with a phone. It's the same error as the one you posted.
* Failed install to the FireTV, I thinks that's because FTV has already implemented the Matter Service.
`Failure [INSTALL_FAILED_DUPLICATE_PERMISSION: Package com.matter.tv.server attempting to redeclare permission com.matter.tv.app.api.permission.SEND_DATA already owned by com.amazon.tv.devicecontrol]
`
> I think you will need the tv-server target, not the chip tool.

I didn't see any places that say we should use the TV-server target. Do you know where the document that describes the target is?
Just in case I miss something.

strangesource, 2024-07-11:
Thanks for the insights! So we are both blocked by the same issue now I guess.

> I didn't see any places that say we should use the TV-server target. Do you know where the document that describes the target is?

I think tv-server is an old name for the `tv-platform` but it's not documented anywhere.

Hiwensen, 2024-07-18:
Hi @strangesource , do you know what does [this sentence](https://project-chip.github.io/connectedhomeip-doc/examples/tv-casting-app/android/README.html#preparing-for-build) mean?
[image]

I tried to add `CHIP_DEVICE_CONFIG_DEVICE_VENDOR_ID`and `CHIP_DEVICE_CONFIG_DEVICE_PRODUCT_ID`, didn't add `CHIP_DEVICE_CONFIG_DEVICE_NAME`,  not sure the value I should pass to these configs.
```
#define CHIP_DEVICE_CONFIG_DEVICE_VENDOR_ID 65521
#define CHIP_DEVICE_CONFIG_DEVICE_PRODUCT_ID 32769
```

pgregorr-amazon, 2024-10-10:
Hi @Hiwensen and @strangesource

This PR(https://github.com/project-chip/connectedhomeip/pull/35984) fixes the issue you we're seeing in June 2024: Commissioner Declaration with error code: 13 being sent from the platform app (tv-app) to the CHIPTVCastingApp (connectedhomeip/examples/tv-casting-app) and not being handled appropriately. 

The tv-app has also been updated since June, it no longer sends the Commissioner Declaration with error code: 13.

You should be able to connect now. Please verify. 

@lazarkov can you please look into adding documentation around the following (https://github.com/project-chip/connectedhomeip/tree/master/examples/tv-app/android): 

> the IDE prompts can't find some classes when I run the tv-platform app, looks like the missed classes are from some .jar or .so file, but the developer doc didn't say how to build those files.

closed 2024-10-10 by sharadb-amazon

Hiwensen, 2024-11-22:
> Hi @Hiwensen and @strangesource
>
> This PR(#35984) fixes the issue you we're seeing in June 2024: Commissioner Declaration with error code: 13 being sent from the platform app (tv-app) to the [...]

Thank you @pgregorr-amazon! I will verify in the next following days and get back to you.

Hiwensen, 2024-11-27:
@pgregorr-amazon I tried with the latest code by running `git pull & git submodule update --init`, but I still can't connect with a FireTV device from an Android phone device. Below are the screenshot and the adb log.
The error looks the same as the one I posted at https://github.com/project-chip/connectedhomeip/issues/34224

[image]
[matter_connect_log.txt](https://github.com/user-attachments/files/17930398/matter_connect_log.txt)