project-chip/connectedhomeip #33850: [Documentation]

ndyck14, 2024-06-11:
### Documentation issues

instructions provided on https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/linux

mentioned to use target as `darwin-x64-all-clusters-no-ble-asan-clang` for macOS.

This results in
```
Error: Invalid value for '--target': 'darwin-x64-all-clusters-no-ble-asan-clang' is not a valid target name.
```

It seems the correct target name is actually `darwin-arm64-all-clusters-no-ble-asan-clang` (x64 --> arm64)

This seems potentially trivial, although darwin-x64 is used in a number of places in yaml files. Not sure if it was changed somewhere along the way, just not everywhere? Or if there is otherwise a proper disctinction. Perhaps @bzbarsky-apple or @jtung-apple have insight?

### Platform

darwin

bzbarsky-apple, 2024-06-13:
> It seems the correct target name is actually darwin-arm64-all-clusters-no-ble-asan-clang (x64 --> arm64)

That depends on whether you're on an Intel or ARM Mac, right?

CI uses darwin-x64 because the things it's running on have Intel processors.  If you're on an Intel mbp/air/whatever, darwin-x64 is the right thing.  If you're on an M1/2/3/etc laptop darwin-arm64 is the right thing.

@ndyck14 I agree the docs should be clearer about this.  Do you want to fix them up, or should I do it?

ndyck14, 2024-07-01:
@bzbarsky-apple, if you are able that would be great.

bzbarsky-apple, 2024-07-04:
@ndyck14 So I went to do this, and the instructions at https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/linux/README.md already have different commands for "Intel Mac" and "Arm [sic] Mac".... and have for 18+ months.

I'm happy to adjust the documentation as needed, but at this point I'd like to understand what part was unclear in what we have right now.  I had thought above that we only documented the Intel build string, but we do document both....

ndyck14, 2024-07-11:
yup, looks like the sunglasses were on the top of the head here. 

Perhaps a message could be rephrased "...is not a valid target name for this system"

But either way, not quite sure how i missed that. 

sorry for the run around.

closed 2024-07-11 by ndyck14