project-chip/connectedhomeip #33857: [1.3] All values of FanModeSequenceEnum in FanControl cluster are Off

devnaruka, 2024-06-12:
### Reproduction steps

In this commit, all values in the `FanModeSequenceEnum` got changed to `Off`. I don’t think it is intentional and seems like a bug.

https://github.com/project-chip/connectedhomeip/commit/9c0cd9eb3656d3c7e40ca811257270db50384d61#diff-5e76a5680a663888acfb0e00e8da9465bfe3ab6f98a92e7a05248fc80abe2401

### Bug prevalence

Haven't tested, but should be persistent

### GitHub hash of the SDK that was being used

Not tested

### Platform

other

### Type

Spec Compliance Issue

bzbarsky-apple, 2024-06-12:
@cecille

bzbarsky-apple, 2024-06-12:
Spec change to avoid this in the future: https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/9369

Corresponding SDK change to the non-generated XML: https://github.com/project-chip/connectedhomeip/pull/33874

The generated XML bits pointed to in this issue will get updated at some point after the spec PR merges.

PR #33874 (2024-06-12): [FANCTRL] Fix enums
This PR removes the slashes in the enum names for fan control.

Spec PR here: https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/9369

Fixes #33857
closed 2024-06-12 by mergify[bot]