project-chip/connectedhomeip #33868: Sample All-Clusters App need to be updated by removing Provisional Clusters, Non-Spec Features, Non-Spec Standard Attributes, and Non-Spec Standard Commands

Rajashreekalmane, 2024-06-12:
**Description:**
While validating the TC-IDM-10.2 test case using   DUT as master and V1.3 as controller, errors were identified, including Provisional Clusters, Non-Spec Features, Non-Spec Standard Attributes, and Non-Spec Standard Commands. Therefore, these elements need to be removed from the sample app (all-clusters-app).
- [ ] Command with unknown direction in the DoorLock cluster.
- [ ] Attributes included but disallowed by conformance in the NetworkCommissioning cluster.
- [ ] Command included but disallowed by conformance in the AdministratorCommissioning cluster.
- [ ] Standard cluster found on the device but not present in spec data in the OnOffSwitchConfiguration and BinaryInputBasic clusters.
- [ ] Attributes included but disallowed by conformance in the LevelControl cluster.
- [ ] Attributes included but disallowed by conformance in the LaundryWasherMode and RefrigeratorAndTemperatureControlledCabinetMode clusters.
- [ ] Attributes included but disallowed by conformance in the DishwasherMode cluster.
- [ ] Command included but disallowed by conformance in the RvcOperationalState cluster.
- [ ] Standard attribute found on the device but not in spec in the DeviceEnergyManagement cluster.
- [ ] Disallowed feature with a mask in the EnergyEvse cluster.
- [ ] Attributes and commands included but disallowed by conformance in the WindowCovering cluster.
- [ ] Standard attribute found on the device but not in spec in the Thermostat cluster.
- [ ] Standard cluster found on the device but not present in spec data in the BarrierControl and ElectricalMeasurement clusters.

Reproduction steps
lunch the all cluster app by sending below mentioned command
./chip-all-clusters-app

Command to execute the [TC-IDM-10.2]
python3 TC_DeviceConformance.py --manual-code 34970112332  --storage-path admin_storage.json --tests test_TC_IDM_10_2 --paa-trust-store-path /credentials/development/paa-root-certs/

**SDK that was being used** : 47097e023a960f2cf5cb2d4dc4f00a2a9c02727f

**Platform** : RPI

PFA log below  : 
**[TC-IDM-10.2** V1.3 cntr and master dut.txt](https://github.com/user-attachments/files/15800638/TC-IDM-10.2.V1.3.cntr.and.master.dut.txt)

bzbarsky-apple, 2024-06-12:
The point of all-clusters-app is to serve as a testbed for testing all clusters and cluster features, including ones that are provisional, in-development, etc.  This app is NOT expected to pass tests like TC-IDM-10.2.

If this test needs to be validated, it presumably needs to be validated against some other example app.

@cecille

bzbarsky-apple, 2024-06-12:
That said, going down your list:

> Command with unknown direction in the DoorLock cluster.

This is a bug in the test or harness.  It's basically complaining about not finding commands that are not part of Matter at all (though the 1.3 spec messed this up, which might be what you are seeing).

> Attributes included but disallowed by conformance in the NetworkCommissioning cluster.

This is a general issue with the Network Commissioning cluster, not with all-clusters-app per se.  It's tracked in https://github.com/project-chip/connectedhomeip/issues/31405

>  Command included but disallowed by conformance in the AdministratorCommissioning cluster.

This is just a bug in #30004, which added the BC feature from the spec but did not set it in the places where it should be set.  The feature bit should be set; nothing needs to be removed.  Filed https://github.com/project-chip/connectedhomeip/issues/33881

>  Standard cluster found on the device but not present in spec data in the OnOffSwitchConfiguration and BinaryInputBasic clusters.

This is the "this thing enables all the clusters, even provisional/etc ones".

That said, these specific clusters we could consider disabling in this app.  Two separate issues needed, one per cluster.

>  Attributes included but disallowed by conformance in the LevelControl cluster.

The problem is that other parts of the test harness expect all attributes to be exercised via all-clusters-app, and specifically asked us to enable all attributes, even ones that are not normally allowed by the conformance/features.  Please figure out what you all actually want here, inside the cert team.

> Attributes included but disallowed by conformance in the LaundryWasherMode and RefrigeratorAndTemperatureControlledCabinetMode clusters.
> Attributes included but disallowed by conformance in the DishwasherMode cluster.

Those look like bugs, which should be filed as separate issues (the appliances folks need to deal with those).

>  Command included but disallowed by conformance in the RvcOperationalState cluster.

Should be a separate issue for the RVC folks to deal with.

>  Standard attribute found on the device but not in spec in the DeviceEnergyManagement cluster.

This seems like a bug in the test?  The spec has attribute 0 on this cluster just fine; it's ESAType.

>  Attributes and commands included but disallowed by conformance in the WindowCovering cluster.

See above about cert in the past explicitly asking for these to be enabled here.

>  Standard attribute found on the device but not in spec in the Thermostat cluster.

This is the preset stuff.  It's definitely in the spec, albeit provisional.

>  Standard cluster found on the device but not present in spec data in the BarrierControl and ElectricalMeasurement clusters.

We could consider removing these if there is no plan to actually standardize them.  Two separate issues needed.

Rajashreekalmane, 2024-06-25:
Here are the separate issues for the above-mentioned clusters: Laundry Washer Mode Cluster, Refrigerator and Temperature Controlled Cabinet Mode, and Dishwasher Mode Cluster.

Links to the issues:

https://github.com/project-chip/connectedhomeip/issues/30384
https://github.com/project-chip/connectedhomeip/issues/30383
https://github.com/project-chip/connectedhomeip/issues/30382

bzbarsky-apple, 2024-06-26:
Those are about the XML, which is rather unrelated to what happens in the example app.

Rajashreekalmane, 2024-06-28:
Issues raised for RefrigeratorAndTemperatureControlledCabinetMode, DishwasherMode, LaundryWasherMode clusters

https://github.com/project-chip/connectedhomeip/issues/34129
https://github.com/project-chip/connectedhomeip/issues/34131
https://github.com/project-chip/connectedhomeip/issues/34127

closed 2024-10-17 by Rajashreekalmane