project-chip/connectedhomeip #33881: [CERT-TEST-FAILURE] Need to set BC feature bit in Administrator Commissioning cluster as needed

bzbarsky-apple, 2024-06-12:
### Feature Area

Other

### Test Case

TC-IDM-10.2

### Reproduction steps

The issue is that https://github.com/project-chip/connectedhomeip/pull/30004 added the BC feature from the spec, but did not actually set that feature bit in the places that need it set.  Which is probably all the .zap files that correspond to the .matter files listed by `git grep -l "handle command OpenBasicCommissioningWindow"`

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

47097e023a960f2cf5cb2d4dc4f00a2a9c02727f

### Platform

core

cecille, 2024-06-13:
Did that PR actually add that command? It doesn't appear as changed in the .matter files, but there's also a LOT of them and I could just be missing it.

I recall going through these a while ago and removing BC command and feature specifically. The fix that was applied then was to remove the command rather than set the feature bit. BC is mentioned in the thread model, so the sense was that we shouldn't have it in the examples as there's a lot of copy-paste into real products where this could be a problem.

bzbarsky-apple, 2024-06-13:
> Did that PR actually add that command?

No, the command was there before.

We could turn off the command in various places too, but that didn't happen either, apparently?