project-chip/connectedhomeip #33890: DEM Modes don't state that they should be mandatory

jamesharrow, 2024-06-12:
The intention of the DEM Mode cluster was to address a 1.3 security threat model review, in that consumers should explicitly have control on if they want their device to be remotely controllable.

Hence we added the DEM Mode cluster with some 'example' tags.

However the DEM Spec now expects all commands to be checked against the OptOutStateEnum which is a Read-only attribute.

The intent of the DEM Modes was to allow the user to change the opt-out mode, but it looks like nothing in the spec mandates the inclusion of these specific modes.

I believe we should add this to tighten up and ensure products don't ship without these modes.
Potentially we need a test case to validate that these modes SHALL be included - but not sure how to mandate specific modes are included. Does this go in the Device Type? Or the Mode cluster (remove the word 'examples' might help).

ReneJosefsen, 2024-06-13:
Should this have been a spec issue?

jamesharrow, 2024-06-13:
> Should this have been a spec issue?

Yes - I added it using the Github project view and it accidentally ended up here.
I'll remove and repost it.

closed 2024-06-13 by jamesharrow