project-chip/connectedhomeip #33892: [Build]

cybortd, 2024-06-13:
### Build issue(s)

after installing environment in esp-idf i did this 
1.cd connectedhomeip/examples/lighting-app/esp32 
2. idf.py build 
then received: 
ARNING: Python interpreter "/home/ubutv/work/connectedhomeip/.environment/pigweed-venv/bin/python" used to start idf.py is not from installed venv "/home/ubutv/.espressif/python_env/idf5.4_py3.10_env"
Executing action: all (aliases: build)
Running cmake in directory /home/ubutv/work/connectedhomeip/examples/lighting-app/esp32/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/ubutv/work/connectedhomeip/.environment/pigweed-venv/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/ubutv/work/connectedhomeip/examples/lighting-app/esp32"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- Component directory /home/ubutv/work/connectedhomeip/examples/lighting-app/esp32/third_party/connectedhomeip/examples/common/cmake does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/ubutv/work/connectedhomeip/examples/lighting-app/esp32/third_party/connectedhomeip/examples/common/credentials does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/ubutv/work/connectedhomeip/examples/lighting-app/esp32/third_party/connectedhomeip/examples/common/imgui_ui does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/ubutv/work/connectedhomeip/examples/lighting-app/esp32/third_party/connectedhomeip/examples/common/m5stack-tft does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/ubutv/work/connectedhomeip/examples/lighting-app/esp32/third_party/connectedhomeip/examples/common/pigweed does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/ubutv/work/connectedhomeip/examples/lighting-app/esp32/third_party/connectedhomeip/examples/common/tracing does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/ubutv/work/connectedhomeip/examples/lighting-app/esp32/third_party/connectedhomeip/examples/common/websocket-server does not contain a CMakeLists.txt file. No component will be added
CMake Error at /home/ubutv/work/esp-idf/tools/cmake/project.cmake:465 (message):
  Directory specified in EXTRA_COMPONENT_DIRS doesn't exist:
  /home/ubutv/work/connectedhomeip/examples/lighting-app/esp32/third_party/connectedhomeip/examples/common/m5stack-tft/repo/components
Call Stack (most recent call first):
  /home/ubutv/work/esp-idf/tools/cmake/project.cmake:549 (__project_init)
  CMakeLists.txt:37 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1, output of the command is in the /home/ubutv/work/connectedhomeip/examples/lighting-app/esp32/build/log/idf_py_stderr_output_15320 and /home/ubutv/work/connectedhomeip/examples/lighting-app/esp32/build/log/idf_py_stdout_output_15320

- I try to build an examples of matter( lighting-app). how can i solve this problem.
thks

### Platform

esp32

bzbarsky-apple, 2024-06-13:
@shubhamdp @dhrishi

dhrishi, 2024-06-16:
@shripad621git Please take a look.

@cybortd Meanwhile, can you please confirm if you have followed the set of steps [here](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/setup_idf_chip.md#setup-esp-idf-and-matter-environment), in the same sequence?

cybortd, 2024-06-16:
yes, i have followed that. i did it again three times but always same error

shripad621git, 2024-06-17:
@cybortd , from the python environment I can see  from the logs, I guess you are on a higher esp-idf version than recommended 
[here](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/setup_idf_chip.md). Can you please confirm the idf version or the idf commit id you are using.
Also can you please confirm that "connectedhomeip/examples/common/m5stack-tft/repo/components" exists in your setup or not.

cybortd, 2024-06-18:
@shripad621git  i used esp-idf v5.1.2 as recommeded

shripad621git, 2024-07-10:
@cybortd , please can you please confirm that during connecectedhomeip setup for esp32, have you executed the below command  
` ./scripts/checkout_submodules.py --platform esp32 linux --shallow.`

dhrishi, 2024-10-25:
Closing this now. Please provide the requested information if you are still seeing this issue

closed 2024-10-25 by dhrishi