project-chip/connectedhomeip #33901: Build Issue[1.0]

aniketkumar04, 2024-06-13:
### Reproduction steps

I was trying to build the project and followed the steps mention in the build guide: https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md#installing-prerequisites-on-linux

Below are the errors I am getting after export command:

export ANDROID_HOME=~/home/aniket.kumar/matter/android2/platform-tools
export ANDROID_NDK_HOME=~/home/aniket.kumar/matter/android2/android-ndk-r26d

aniket.kumar@si:~/matter/connectedhomeip$ ./scripts/build/build_examples.py --target android-arm64-chip-tool build
2024-06-13 12:29:34 INFO    Building targets: android-arm64-chip-tool
2024-06-13 12:29:34 INFO    Preparing builder 'android-arm64-chip-tool'
2024-06-13 12:29:34 INFO    Generating /home/aniket.kumar/matter/connectedhomeip/out/android-arm64-chip-tool
2024-06-13 12:29:34 INFO    Setting up Android deps through Gradle
2024-06-13 12:29:34 INFO    > Task :buildSrc:compileJava NO-SOURCE
2024-06-13 12:29:34 INFO    > Task :buildSrc:compileGroovy UP-TO-DATE
2024-06-13 12:29:34 INFO    > Task :buildSrc:processResources NO-SOURCE
2024-06-13 12:29:34 INFO    > Task :buildSrc:classes UP-TO-DATE
2024-06-13 12:29:34 INFO    > Task :buildSrc:jar UP-TO-DATE
2024-06-13 12:29:34 INFO    > Task :buildSrc:assemble UP-TO-DATE
2024-06-13 12:29:34 INFO    > Task :buildSrc:compileTestJava NO-SOURCE
2024-06-13 12:29:34 INFO    > Task :buildSrc:compileTestGroovy NO-SOURCE
2024-06-13 12:29:34 INFO    > Task :buildSrc:processTestResources NO-SOURCE
2024-06-13 12:29:34 INFO    > Task :buildSrc:testClasses UP-TO-DATE
2024-06-13 12:29:34 INFO    > Task :buildSrc:test NO-SOURCE
2024-06-13 12:29:34 INFO    > Task :buildSrc:check UP-TO-DATE
2024-06-13 12:29:34 INFO    > Task :buildSrc:build UP-TO-DATE
2024-06-13 12:29:34 INFO    > Task :copyArtifacts UP-TO-DATE
2024-06-13 12:29:34 INFO    > Task :setUpAndroidDeps
2024-06-13 12:29:34 INFO
2024-06-13 12:29:34 INFO    BUILD SUCCESSFUL in 731ms
2024-06-13 12:29:34 INFO    4 actionable tasks: 1 executed, 3 up-to-date
2024-06-13 12:29:35 INFO    Command ['python3', 'third_party/android_deps/set_up_android_deps.py'] completed
2024-06-13 12:29:35 INFO    Setting up Java deps
2024-06-13 12:29:35 INFO    Command ['third_party/java_deps/set_up_java_deps.sh'] completed
Traceback (most recent call last):
  File "/home/aniket.kumar/matter/connectedhomeip/./scripts/build/build_examples.py", line 234, in <module>
    main(auto_envvar_prefix='CHIP')
  File "/home/aniket.kumar/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/aniket.kumar/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/aniket.kumar/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    rv.append(sub_ctx.command.invoke(sub_ctx))
  File "/home/aniket.kumar/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/aniket.kumar/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/aniket.kumar/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/aniket.kumar/matter/connectedhomeip/./scripts/build/build_examples.py", line 224, in cmd_build
    context.obj.Build()
  File "/home/aniket.kumar/matter/connectedhomeip/scripts/build/build/__init__.py", line 67, in Build
    self.Generate()
  File "/home/aniket.kumar/matter/connectedhomeip/scripts/build/build/__init__.py", line 62, in Generate
    builder.generate()
  File "/home/aniket.kumar/matter/connectedhomeip/scripts/build/builders/android.py", line 355, in generate
    self.validate_build_environment()
  File "/home/aniket.kumar/matter/connectedhomeip/scripts/build/builders/android.py", line 177, in validate_build_environment
    raise Exception(
Exception: '/home/aniket.kumar/home/aniket.kumar/matter/android2/platform-tools/tools/bin/sdkmanager' and '/home/aniket.kumar/home/aniket.kumar/matter/android2/platform-tools/cmdline-tools/latest/bin/sdkmanager' is not executable by the current user

Folder structure:
SDK
aniket.kumar@si:~/matter/android2/platform-tools/cmdline-tools/latest/bin$ ls
apkanalyzer  avdmanager  lint  profgen  resourceshrinker  retrace  screenshot2  sdkmanager

aniket.kumar@si~/matter/android2/platform-tools/tools/bin$ ls
sdkmanager

Please let me know what went wrong.

### Bug prevalence

2

### GitHub hash of the SDK that was being used

android-ndk-r26d

### Platform

android

### Type

Platform Issue

### Anything else?

NA

Kartikgupta1502, 2024-08-16:
I am getting the same error. Were you able to solve this?

andy31415, 2024-10-21:
what are the permissions for `/home/aniket.kumar/home/aniket.kumar/matter/android2/platform-tools/cmdline-tools/latest/bin/sdkmanager` ?

It seems like you are unable to execute a binary file, so maybe those are somehow owned by root or the path is invalid.

You can also try to use the build docker images instead, if you need a pre-build environment.

yunhanw-google, 2025-02-03:
not able to be reproduced in my side, from your log, it seems tobe 
platform-tools/cmdline-tools/latest/bin/sdkmanager is missing, please double-check the path, thanks

closed 2025-02-03 by yunhanw-google