project-chip/connectedhomeip #33919: [1.3] [TC-DRLK-2.6] Verification of test step is not generic

doublemis1, 2024-06-14:
### Reproduction steps

Test case's step 6 expect specific failure.
The test case specify that the device supports at least 11 holiday schedules.
If the device support less than 11 holiday schedules (index 10), the test case fail.
Based on the specification the device may support form 0 to 255 holiday schedules.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

main

### Platform

core

### Platform Version(s)

1.3

### Type

Test Improvement

bzbarsky-apple, 2024-06-14:
@doublemis1 test plan issues should get filed in https://github.com/CHIP-Specifications/chip-test-plans

doublemis1, 2024-06-17:
@bzbarsky-apple thanks for it pointing out.
Created in right place:
https://github.com/CHIP-Specifications/chip-test-plans/issues/4228

closed 2024-06-17 by doublemis1