project-chip/connectedhomeip #33936: [CERT-TEST-FAILURE] Test_TC_IDM_5_2.yaml test cases 3 and 4 should expect TIMEOUT?

Apollon77, 2024-06-14:
### Feature Area

Other

### Test Case

Test_TC_IDM_5_2

### Reproduction steps

The test cases 3 and 4 are frommmy perspective testing a Invoke/Write interaction with an expired TimedInteraction before.

https://github.com/project-chip/connectedhomeip/blob/a12a0c070e2ad6e0347e617ace1e8098b4ccd8d3/src/app/tests/suites/certification/Test_TC_IDM_5_2.yaml#L90

When I check specs (1.1+) it tells e.g.

> Incoming Write Request Action
> If this action is part of a Timed Write transaction, and the Timeout has expired from the preced­ ing Timed Request action, then a Status Response action with the TIMEOUT Status Code SHALL be submitted to the message layer and this interaction SHALL terminate.

and for read:
> If this action is part of a Timed Invoke transaction, and the Timeout has expired from the pre­ ceding Timed Request action, then a Status Response action with the TIMEOUT Status Code SHALL be submitted to the message layer and this interaction SHALL terminate.

This test here expects UnsupportedAccess as error.

Do I miss anything or is the test defined wrong or the specticification?

### Bug prevalence

came back as error from a current 1.1 certification

### GitHub hash of the SDK that was being used

master

### Platform

core

cecille, 2024-06-15:
This appears to be quite deliberate in the code - see https://github.com/project-chip/connectedhomeip/blob/master/src/app/TimedHandler.cpp#L69.

@bzbarsky-apple - I know this was three years ago, but do you recall what part of the spec was being referenced here? Do we have a spec conflict?

cecille, 2024-06-15:
Found it - please see https://github.com/CHIP-Specifications/connectedhomeip-spec/commit/0bd0e4f937ebaf84d6ebe7dc22e79c5af605bc2a. Error was changed, it would appear that the SDK and tests were not updated.

bzbarsky-apple, 2024-06-17:
Yeah, code was written 3 years ago, spec stealth-changed 2 years ago....  :(

cjandhyala, 2024-08-19:
@Apollon77 closing the issue given SDK and test plan issues are closed. Feel free to reopen if you still think this issue needs to be addressed.

closed 2024-08-19 by cjandhyala