project-chip/connectedhomeip #33950: [Cirque] use virtualenv

andy31415, 2024-06-17:
After upgrading to 24.04 it becomes very obvious that cirque should use a virtualenv instead of dirtying the system python packages.

We should instead create a virtualenv for every pip package install and python usage. We also seem to have significant copy&paste throughout the tests for pip install, those should be consolidated.

FergusHuang, 2024-10-25:
@andy31415 could you review the PR #36163 ? I don't know why there is no reviewers in this PR.

andy31415, 2024-12-16:
> @andy31415 could you review the PR #36163 ? I don't know why there is no reviewers in this PR.

Reviewers are not added automatically anymore as that did not seem to actually encourage reviews (we kept adding the same 20-30 people on every PR). I expect reviews to happen and that PR does have one review and probably is a candidate for fast track.

I did have a comment on it though: it requires a"how is this tested" section (or if this just changes cirque tests, a statement that cirque tests uses this and as such it is tested). I would also prefer creating things into `out` instead of separating out yet another folder in the top level source tree.