project-chip/connectedhomeip #33958: [1.3] chef-rvc-mode-delegate.cpp:115:115: error: ‘kOnOff’ is not a member of ‘chip::app::Clusters::RvcRunMode::Feature’

lboue, 2024-06-17:
### Reproduction steps

Hello,

I am facing error building Chef RVC Linux app:
```
cd connectedhomeip/
source scripts/activate.sh
cd examples/chef

./chef.py -t linux -d rootnode_roboticvacuumcleaner_1807ff0c49 -b
```

Regards

### Bug prevalence

always

### GitHub hash of the SDK that was being used

d83caf7fefeaa91ebc549863552050311abe46ae

### Platform

other

### Platform Version(s)

Linux

### Type

Platform Issue

### Anything else?

user@Matter:~/connectedhomeip/examples/chef$ ./chef.py -t linux -d rootnode_roboticvacuumcleaner_1807ff0c49 -b

  ______  __    __   _______  _______
 /      ||  |  |  | |   ____||   ____|
|  ,----'|  |__|  | |  |__   |  |__
|  |     |   __   | |   __|  |   __|
|  `----.|  |  |  | |  |____ |  |
 \______||__|  |__| |_______||__|

Target is set to rootnode_roboticvacuumcleaner_1807ff0c49
Setting up environment...
/home/user/connectedhomeip/examples/chef/../..//scripts/activate.sh: line 116: scripts/setup/gen_pigweed_cipd_json.py: No such file or directory

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
 ▀▀    █   █    ▀▀

  ACTIVATOR! This sets your shell environment variables.

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Project actions........skipped
  Setting environment variables for Python environment.....done
  Setting environment variables for pw packages............skipped
  Setting environment variables for Host tools.............done
[... 31 lines cut ...]
/home/user/connectedhomeip/examples/chef/common/chef-rvc-operational-state-delegate.cpp:164:131: error: no matching function for call to ‘chip::app::Clusters::RvcOperationalState::Instance::Instance(chip::app::Clusters::RvcOperationalState::RvcOperationalStateDelegate*&, chip::EndpointId&)’
  164 |     gRvcOperationalStateInstance        = new RvcOperationalState::Instance(gRvcOperationalStateDelegate, operationalStateEndpoint);
      |                                                                                                                                   ^
In file included from /home/user/connectedhomeip/examples/chef/common/chef-rvc-operational-state-delegate.h:22,
                 from /home/user/connectedhomeip/examples/chef/common/chef-rvc-operational-state-delegate.cpp:22:
../third_party/connectedhomeip/src/app/clusters/operational-state-server/operational-state-server.h:384:5: note: candidate: ‘chip::app::Clusters::RvcOperationalState::Instance::Instance(chip::app::Clusters::RvcOperationalState::Delegate*, chip::EndpointId)’
  384 |     Instance(Delegate * aDelegate, EndpointId aEndpointId) :
      |     ^~~~~~~~
../third_party/connectedhomeip/src/app/clusters/operational-state-server/operational-state-server.h:384:25: note:   no known conversion for argument 1 from ‘chip::app::Clusters::RvcOperationalState::RvcOperationalStateDelegate*’ to ‘chip::app::Clusters::RvcOperationalState::Delegate*’
  384 |     Instance(Delegate * aDelegate, EndpointId aEndpointId) :
      |              ~~~~~~~~~~~^~~~~~~~~
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-unknown-warning-option’ may have been intended to silence earlier diagnostics
[5/8] c++ obj/rootnode_roboticvacuumcleaner_1807ff0c49.main.cpp.o
ninja: build stopped: subcommand failed.
Traceback (most recent call last):
  File "/home/user/connectedhomeip/examples/chef/./chef.py", line 1011, in <module>
    sys.exit(main())
  File "/home/user/connectedhomeip/examples/chef/./chef.py", line 872, in main
    shell.run_cmd("ninja -C out")
  File "/home/user/connectedhomeip/examples/chef/stateful_shell.py", line 139, in run_cmd
    raise RuntimeError(
RuntimeError: Error. Nonzero return code.
Returncode: 1
Cmd: ninja -C out

bzbarsky-apple, 2024-06-18:
This feature value was removed from the spec.

lboue, 2024-06-18:
@bzbarsky-apple Shouldn't the code be changed accordingly when the specialisation evolves?
Moreover, why were there any changes in the v1.3 branch when the specifications are now public?

bzbarsky-apple, 2024-06-21:
> Shouldn't the code be changed accordingly when the specialisation evolves?

Yes.  But it wasn't in this case, which is why you are hitting this problem.

> Moreover, why were there any changes in the v1.3 branch when the specifications are now public?

That's a question for the RVC people and @robszewczyk

lboue, 2024-08-28:
This now works with the latest commit:

```
$ ./examples/chef/chef.py -t linux -d rootnode_roboticvacuumcleaner_1807ff0c49 -b

  ______  __    __   _______  _______
 /      ||  |  |  | |   ____||   ____|
|  ,----'|  |__|  | |  |__   |  |__
|  |     |   __   | |   __|  |   __|
|  `----.|  |  |  | |  |____ |  |
 \______||__|  |__| |_______||__|

Target is set to rootnode_roboticvacuumcleaner_1807ff0c49
Setting up environment...

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
 ▀▀    █   █    ▀▀

  ACTIVATOR! This sets your shell environment variables.

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Project actions........skipped
  Setting environment variables for Python environment.....done
  Setting environment variables for pw packages............skipped
  Setting environment variables for Host tools.............done

Checking the environment:

20240828 07:47:18 INF Environment passes all checks!

Environment looks good, you are ready to go!

RPC PW disabled
Software Version String: ""
Product ID 0x8000 / Vendor ID 0xFFF1
Building...
x86_64
Generating compile_commands took 41ms
Done. Made 3665 targets from 295 files in 1313ms
ninja: Entering directory `out'
[70/70] stamp obj/default.stamp
Done
```

closed 2024-08-28 by lboue