project-chip/connectedhomeip #33966: How to solve the problem of slow Commissioning process？

chuckrui, 2024-06-18:
In the process of production, we found that when we carried out application testing (add the device to Apple home app-> Test onoff and level control -> Remove the device), we found that it took about 5 minutes to successfully add the device to Apple home app each time. What's the reason, I wonder? Or is there anything that can optimize the speed of the distribution network? It is worth mentioning that our speed will not be so slow at the beginning, but when more and more devices are added, you can obviously feel the speed is getting slower and slower.
I was wondering what else could be done to make things faster in terms of production testing? Because so far we've only tested on Apple home, we haven't tried any other ecosystem

we use Nordic52840 and the matter certification version is 1.0.
The following is my network log, can you help me see where the response is slow?
[Commissioning log.txt](https://github.com/user-attachments/files/15879204/Commissioning.log.txt)

bzbarsky-apple, 2024-06-18:
So looking at that log, the device starts advertising over BLE here:

```
00> I: 30270 [DL]Chuck:CHIPoBLE advertising started
```

The BLE connection is established 17 seconds later (assuming those are millisecond timestamps):

```
00> I: 47306 [DL]BLE connection established (ConnId: 0x00)
```

Then commissioning starts here:

```
00> I: 49201 [EM]>>> [E:36424r M:249843416] (U) Msg RX from 0:10EABFD526874AF4 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
```

and completes pretty quickly:

```
00> I: 60088 [FS]GeneralCommissioning: Received CommissioningComplete
```

Then there is a second commissioning (one for keychain, one for Home), which completes here:

```
00> I: 66459 [FS]GeneralCommissioning: Received CommissioningComplete
```

Then the device hears nothing from the other site for a while until:

```
00> I: 162148 [EM]>>> [E:36467r M:120959253] (S) Msg RX from 1:00000000457B8B14 [E86A] --- Type 0001:02 (IM:ReadRequest)
```

Then the other side tries to subscribe to the device:

```
00> I: 165060 [EM]>>> [E:77r M:77935753] (S) Msg RX from 1:00000000457B8B14 [E86A] --- Type 0001:03 (IM:SubscribeRequest)
```

and the priming data gets sent, about one packet per 100ms:

```
00> I: 165620 [EM]<<< [E:77r M:23793268 (Ack:77935764)] (S) Msg TX to 1:00000000457B8B14 [E86A] --- Type 0001:05 (IM:ReportData)
00> I: 165731 [EM]>>> [E:77r M:77935765 (Ack:23793268)] (S) Msg RX from 1:00000000457B8B14 [E86A] --- Type 0001:01 (IM:StatusResponse)
00> I: 165742 [EM]<<< [E:77r M:23793269 (Ack:77935765)] (S) Msg TX to 1:00000000457B8B14 [E86A] --- Type 0001:05 (IM:ReportData)
00> I: 165860 [EM]>>> [E:77r M:77935766 (Ack:23793269)] (S) Msg RX from 1:00000000457B8B14 [E86A] --- Type 0001:01 (IM:StatusResponse)
...etc
```

This continues until:

```
00> I: 167645 [EM]<<< [E:77r M:23793286 (Ack:77935782)] (S) Msg TX to 1:00000000457B8B14 [E86A] --- Type 0001:04 (IM:SubscribeResponse)
```

By this point:

```
00> I: 191064 [EM]>>> [E:87r M:77935795] (S) Msg RX from 1:00000000457B8B14 [E86A] --- Type 0001:08 (IM:InvokeCommandRequest)
```

the other side is sending lighting "on" commands to the device.

So it's taking 30s to start advertising over BLE, another 17s to establish the BLE connection, then 20s to do the commissioning, then for close to 100s the client goes off an does ... something.  Then subscription takes about 2s, and I would expect the device to be considered "added" at or before that point, at about the 3 minute mark.

bzbarsky-apple, 2024-06-18:
In terms of why there's that 100s gap: I recommend filing a feedback assistant ticket per the directions below, which would include logs from the Apple device in question that would perhaps allow that to be pinned down.  Please mention the ticket ID in this issue, and I will see about getting someone to look at it.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

closed 2025-04-02 by woody-apple