project-chip/connectedhomeip #33974: [1.2] The endpoint device is not displayed in the Apple app

maenkai, 2024-06-18:
### Reproduction steps

```shell
1. Matter bridge device create and commision
2. The endpoint device is not displayed in the Apple app
3. chiptool can look up the endpoint
4. Restart homepod or restart the Matter bridge device
5. Still have problems
6. Appple app re-commision the bridge device
7. Have a chance of recovery

### Bug prevalence

indeterminacy

### GitHub hash of the SDK that was being used

46e1ad8478c55ec20273da9d40a9c6fecf5c6b38

### Platform

esp32

### Platform Version(s)

1.2

### Type

Platform Issue

### Anything else?

[chiptool.log](https://github.com/user-attachments/files/15881044/chiptool.log)

maenkai, 2024-06-18:
[chiptool.log](https://github.com/user-attachments/files/15881050/chiptool.log)

bzbarsky-apple, 2024-06-18:
@maenkai Please file a feedback assistant ticket following the instructions below and post the ticket ID here so I can follow up on it.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

closed 2025-04-02 by woody-apple