project-chip/connectedhomeip #33988: [SDK Quality Test] #1: Commissioning failed with src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:119: CHIP Error 0x00000032: Timeout

Saravana-kr22, 2024-06-18:
**Description:**
I have used python controller to done a stress test on my RPI sample dut (all-clusters-app) on Pair-Unpair operation for 10000 iteration. The iteration 8022 is passed without any error but the 8023 is Failed to commission the dut during FindOperational step with the below traceback error 

> Traceback (most recent call last):
> File "/Users/chaitanyajandhyala/code/python/matter-qa/src/matter_qa/library/helper_libs/matter_testing_support.py", line 1177, in pair_dut
> raise CommissionFailure(f"Pairing with DUT failed Reason {commission_device_result.str()} ")
> chip.commissioning.CommissionFailure: CommissionFailure(Pairing with DUT failed Reason src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:119: CHIP Error 0x00000032: Timeout )

GitHub hash of the SDK that was being used: https://github.com/project-chip/connectedhomeip/commit/3718e996b56f36a553c1b4571f77728396590776

**Platform**
python, raspi

**Platform Version(s)**
RPI-4, 8GB RAM

**Reference log:**
Example of Failed iteartion details:

The iteartion number 8022 is passed below file for the same
TH log: [log_2024-04-28_03-31-58.log](https://github.com/user-attachments/files/15646584/log_2024-04-28_03-31-58.log)
DUT log: [Dut_log_8022_2024-04-28T03_32_08_256032.log](https://github.com/user-attachments/files/15646605/Dut_log_8022_2024-04-28T03_32_08_256032.log)

The iteration number 8023 is failed with this error below the logfile
TH log : [log_2024-04-28_03-32-30.log](https://github.com/user-attachments/files/15646419/log_2024-04-28_03-32-30.log)
DUT log: [Dut_log_8023_2024-04-28T03_40_32_688669.log](https://github.com/user-attachments/files/15646448/Dut_log_8023_2024-04-28T03_40_32_688669.log)

For the detailed summary please reference this json: [summary.json](https://github.com/user-attachments/files/15588197/summary.json)

Script reference: https://github.com/CHIP-Specifications/matter-qa/blob/main/src/matter_qa/scripts/reliability_scripts/TC_Pair.py

bzbarsky-apple, 2024-07-18:
>  controller - python controller running on raspi,

Per log, TH is clearly running on Darwin.

Anyway, from the TH log:

```
2024-04-28 03:32:31,023 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - <<< [E:22372i S:5863 M:152467288] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[fe80::dea6:32ff:feab:25e5%en1]:5540] --- Type 0001:08 (IM:InvokeCommandRequest)
...
2024-04-28 03:32:31,068 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - >>> [E:22372i S:5863 M:225529247 (Ack:152467288)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
...
2024-04-28 03:32:31,103 - chip.native.ATM - INFO - __init__.py:_RedirectToPythonLogging:40 - 				{
2024-04-28 03:38:17,340 - chip.native.ATM - INFO - __init__.py:_RedirectToPythonLogging:40 - 					"Anonymous<0>" : 
...
2024-04-28 03:38:17,345 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - >>> [E:22372i S:5863 M:225529248 (Ack:152467288)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
```

Note that we lost timeslice in the middle of tracing an incoming packet (the InvokeCommandResponse) for almost 6 minutes.  In the meantime the fail-safe expires on the DUT, so when we try to do the next thing (operational resolve) it fails, because the DUT is not advertising.

Looks like the same issue as https://github.com/project-chip/connectedhomeip/issues/33993

Saravana-kr22, 2024-07-19:
@bzbarsky-apple Below are the logs of the last week run on commit: https://github.com/project-chip/connectedhomeip/commit/d033f8b917faa17b82a432e0c895f96199217dcb

TH-log:
[log_2024-07-05_14-43-36.log](https://github.com/user-attachments/files/16307010/log_2024-07-05_14-43-36.log)

DUT-log: 
[Dut_log_199_2024-07-05T14_43_36_141592.log](https://github.com/user-attachments/files/16307022/Dut_log_199_2024-07-05T14_43_36_141592.log)

Platform
controller - python controller running on raspi,
DUT - all-cluster-app running on Nordic-dk

Provisioned using Ble-thread

Platform Version(s)
RPI-4, 8GB RAM
Nordic-boorad- nrf52840-DK

TCPdump file: [tcpdump.zip](https://github.com/user-attachments/files/16307076/200.zip)

bzbarsky-apple, 2024-07-19:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/33993

closed 2024-07-19 by bzbarsky-apple