project-chip/connectedhomeip #33992: [SDK Quality Test] #2:Unpairing function is failed with src/app/CommandSender.cpp:328: CHIP Error 0x00000032

Saravana-kr22, 2024-06-18:
**Description:**
I have used python controller to run a stress test on my RPI sample dut (all-clusters-app) on Pair-Unpair test scenario for 10,000 iterations. The test passed in iteration 5150 without any error but the iteration 5151 is Failed to commission the dut  during the Remove Current Fabric step with the below traceback error

> Traceback (most recent call last):
> File "/Users/chaitanyajandhyala/code/python/matter-qa/src/matter_qa/library/helper_libs/matter_testing_support.py", line 1187, in unpair_dut
> controller.UnpairDevice(node_id)
> File "/Users/chaitanyajandhyala/code/python/matter-qa/.venv/lib/python3.9/site-packages/chip/ChipDeviceCtrl.py", line 452, in UnpairDevice
> return self._ChipStack.CallAsync(
> File "/Users/chaitanyajandhyala/code/python/matter-qa/.venv/lib/python3.9/site-packages/chip/native/init.py", line 87, in raise_on_error
> raise self.to_exception()
> chip.exceptions.ChipStackError: src/app/CommandSender.cpp:328: CHIP Error 0x00000032: Timeout
> 

GitHub hash of the SDK that was being used: https://github.com/project-chip/connectedhomeip/commit/3718e996b56f36a553c1b4571f77728396590776

**Platform**
controller - python controller running on raspi,
DUT - all-cluster-app running on raspi

Provisioned using on-network

**Platform Version(s)**
RPI-4, 8GB RAM

**Test Scenario**

Pair/Provision the DUT (All-cluster-app running on raspi) using on-network
Read device information after the device is provisioned
sleep for 2 secs
Unpair the DUT (i.e. removing the fabric)
factory reset the DUT
repeat step1 - 6 for 10,000 iterations
Script reference: https://github.com/CHIP-Specifications/matter-qa/blob/main/src/matter_qa/scripts/reliability_scripts/TC_Pair.py

**Reference log:**
Example of Passed iteration details:

The iteration number 5150 is passed, below log files for the same
TH log: [log_2024-04-27_00-03-36.log](https://github.com/user-attachments/files/15645130/log_2024-04-27_00-03-36.log)
DUT log: [Dut_log_5150_2024-04-27T00_03_46_696841.log](https://github.com/user-attachments/files/15645146/Dut_log_5150_2024-04-27T00_03_46_696841.log)

Example of Failed iteration details:

The iteration number 5151 is failed with this error below the logfile
TH log : [log_2024-04-27_00-04-02.log](https://github.com/user-attachments/files/15644923/log_2024-04-27_00-04-02.log)
DUT log: [Dut_log_5151_2024-04-27T00_08_18_324214.log](https://github.com/user-attachments/files/15644934/Dut_log_5151_2024-04-27T00_08_18_324214.log)g)

For the detailed summary please reference this json: [summary.json](https://github.com/user-attachments/files/15588197/summary.json)

bzbarsky-apple, 2024-07-18:
From the TH log:

```
2024-04-27 00:04:06,739 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - <<< [E:37036i S:182 M:76519479] (S) Msg TX to 1:0000000012344321 [03F8] [UDP:[fe80::dea6:32ff:feab:25e4%en0]:5540] --- Type 0001:08 (IM:InvokeCommandRequest)
2024-04-27 00:04:06,739 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:42 - ICR moving to [AwaitingRe]
2024-04-27 00:04:06,739 - chip.native.ATM - INFO - __init__.py:_RedirectToPythonLogging:40 - {
2024-04-27 00:08:18,155 - chip.native.ATM - INFO - __init__.py:_RedirectToPythonLogging:40 - 	"event" : "MessageSend",
```

So TH lost timeslice for 4+ minutes.  The other side had in fact sent a standalone ack in the meantime (which we now get), but then we time out our exchange (because it's been 4+ minutes) before the response or any of its 4 retransmits can get delivered to us.

Looks like the same issue as https://github.com/project-chip/connectedhomeip/issues/33993

bzbarsky-apple, 2024-07-19:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/33993

closed 2024-07-19 by bzbarsky-apple