project-chip/connectedhomeip #33993: [SDK Quality Test] #3: Commission failed with src/app/ClusterStateCache.cpp:303: CHIP Error 0x00000010: Key not found

Saravana-kr22, 2024-06-18:
**Description:**
I have used python controller to run a stress test on my RPI sample dut (all-clusters-app) on Pair-Unpair test scenario for 10,000 iterations. The test passed in iteration 2763 without any error but the iteration 2764 is Failed to read the info such BasicCommissioningInfo, RegulatoryConfig , LocationCapability, Breadcrumb during commission the dut with the below error

> 2024-04-26 05:40:48,629 - chip.native.CTL - ERROR - init.py:_RedirectToPythonLogging:38 - Failed to read BasicCommissioningInfo: src/app/ClusterStateCache.cpp:303: CHIP Error 0x00000010: Key not found
> 2024-04-26 05:40:48,630 - chip.native.CTL - ERROR - init.py:_RedirectToPythonLogging:38 - Failed to read RegulatoryConfig: src/app/ClusterStateCache.cpp:303: CHIP Error 0x00000010: Key not found
> 2024-04-26 05:40:48,630 - chip.native.CTL - ERROR - init.py:_RedirectToPythonLogging:38 - Failed to read LocationCapability: src/app/ClusterStateCache.cpp:303: CHIP Error 0x00000010: Key not found
> 2024-04-26 05:40:48,630 - chip.native.CTL - ERROR - init.py:_RedirectToPythonLogging:38 - Failed to read Breadcrumb: src/app/ClusterStateCache.cpp:303: CHIP Error 0x00000010: Key not found
> 2024-04-26 05:40:48,631 - chip.native.CTL - ERROR - init.py:_RedirectToPythonLogging:38 - Error parsing commissioning information
> 2024-04-26 05:40:48,631 - chip.native.CTL - ERROR - init.py:_RedirectToPythonLogging:38 - Error on commissioning step 'ReadCommissioningInfo2': 'src/app/ClusterStateCache.cpp:303: CHIP Error 0x00000010: Key not found'

GitHub hash of the SDK that was being used: https://github.com/project-chip/connectedhomeip/commit/3718e996b56f36a553c1b4571f77728396590776

**Platform**
controller - python controller running on raspi,
DUT - all-cluster-app running on raspi

Provisioned using on-network

**Platform Version(s)**
RPI-4, 8GB RAM

**Test Scenario**

Pair/Provision the DUT (All-cluster-app running on raspi) using on-network
Read device information after the device is provisioned
sleep for 2 secs
Unpair the DUT (i.e. removing the fabric)
factory reset the DUT
repeat step1 - 6 for 10,000 iterations
Script reference: https://github.com/CHIP-Specifications/matter-qa/blob/main/src/matter_qa/scripts/reliability_scripts/TC_Pair.py

**Reference log:**
Example of Passed iteration details:

The iteration number 2763 is passed, below log files for the same
TH log: [log_2024-04-26_05-36-23.log](https://github.com/user-attachments/files/15643667/log_2024-04-26_05-36-23.log)
DUT log: [Dut_log_2763_2024-04-26T05_36_32_752483.log](https://github.com/user-attachments/files/15643679/Dut_log_2763_2024-04-26T05_36_32_752483.log)

Example of Failed iteration details:

The iteration number 2764 is failed with this error below the logfile
TH log : [log_2024-04-26_05-36-46.log](https://github.com/user-attachments/files/15643452/log_2024-04-26_05-36-46.log)
DUT log: [Dut_log_2764_2024-04-26T05_40_57_865104.log](https://github.com/user-attachments/files/15643487/Dut_log_2764_2024-04-26T05_40_57_865104.log)

For the detailed summary please reference this json: [summary.json](https://github.com/user-attachments/files/15588197/summary.json)

bzbarsky-apple, 2024-07-18:
The "Key not found" thing is a bit of a red herring.  The actual failure is here, from the TH log:

```
2024-04-26 05:40:18,613 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - <<< [E:59991i S:60974 M:30600750] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[fe80::dea6:32ff:feab:25e4%en0]:5540] --- Type 0001:02 (IM:ReadRequest)
...
2024-04-26 05:40:18,621 - chip.native.DMG - ERROR - __init__.py:_RedirectToPythonLogging:38 - Time out! failed to receive report data from Exchange: 59991i

2024-04-26 05:40:18,628 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - <<< [E:59992i S:60974 M:30600751] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[fe80::dea6:32ff:feab:25e4%en0]:5540] --- Type 0001:02 (IM:ReadRequest)
...
2024-04-26 05:40:19,177 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - Retransmitting MessageCounter:30600750 on exchange 59991i Send Cnt 1
2024-04-26 05:40:19,265 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - Retransmitting MessageCounter:30600751 on exchange 59992i Send Cnt 1
2024-04-26 05:40:19,840 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - Retransmitting MessageCounter:30600750 on exchange 59991i Send Cnt 2
2024-04-26 05:40:19,882 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - Retransmitting MessageCounter:30600751 on exchange 59992i Send Cnt 2
2024-04-26 05:40:20,765 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - Retransmitting MessageCounter:30600750 on exchange 59991i Send Cnt 3
2024-04-26 05:40:20,925 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - Retransmitting MessageCounter:30600751 on exchange 59992i Send Cnt 3
2024-04-26 05:40:22,339 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - Retransmitting MessageCounter:30600750 on exchange 59991i Send Cnt 4
2024-04-26 05:40:22,415 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - Retransmitting MessageCounter:30600751 on exchange 59992i Send Cnt 4
2024-04-26 05:40:24,725 - chip.native.EM - ERROR - __init__.py:_RedirectToPythonLogging:38 - Failed to Send CHIP MessageCounter:30600750 on exchange 59991i sendCount: 4 max retries: 4
2024-04-26 05:40:24,784 - chip.native.EM - ERROR - __init__.py:_RedirectToPythonLogging:38 - Failed to Send CHIP MessageCounter:30600751 on exchange 59992i sendCount: 4 max retries: 4
2024-04-26 05:40:48,625 - chip.native.DMG - ERROR - __init__.py:_RedirectToPythonLogging:38 - Time out! failed to receive report data from Exchange: 59992i
```

So our reads timed out and things failed.  I am not 100% sure why the read on exchange `59991i` timed out so quickly, by the way; that's a bit odd.

Looking at the other side, at the DUT log, we have:

```
[1714135007.042325][500980:500980] CHIP:EM: >>> [E:59990r S:0 M:183265092 (Ack:84098056)] (U) Msg RX from 0:6949780F7315A42B [0000] --- Type 0000:24 (SecureChannel:PASE_Pake3)
[1714135007.042690][500980:500980] CHIP:EM: <<< [E:59990r S:0 M:84098057 (Ack:183265092)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[fe80::f:2352:63b5:7bd%eth0]:63338] --- Type 0000:40 (SecureChannel:StatusReport)
...
[1714135007.043119][500980:500980] CHIP:IN: New secure session activated for device <FFFFFFFB00000000, 0>, LSID:37850 PSID:60974!
...
[1714135007.385579][500980:500980] CHIP:EM: Retransmitting MessageCounter:84098057 on exchange 59990r Send Cnt 1
[1714135007.742411][500980:500980] CHIP:EM: Retransmitting MessageCounter:84098057 on exchange 59990r Send Cnt 2
[1714135008.334451][500980:500980] CHIP:EM: Retransmitting MessageCounter:84098057 on exchange 59990r Send Cnt 3
[1714135009.307984][500980:500980] CHIP:EM: Retransmitting MessageCounter:84098057 on exchange 59990r Send Cnt 4
[1714135010.769943][500980:500980] CHIP:EM: Failed to Send CHIP MessageCounter:84098057 on exchange 59990r sendCount: 4 max retries: 4
[1714135067.101502][500980:500980] CHIP:FS: Fail-safe timer expired
...
[1714135218.736037][500980:500980] CHIP:EM: >>> [E:59990r S:0 M:183265093 (Ack:84098057)] (U) Msg RX from 0:6949780F7315A42B [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1714135218.742592][500980:500980] CHIP:IN: Data received on an unknown session (LSID=37850). Dropping it!
```

OK, so we established PASE, then we waited 60s on the DUT, then the fail-safe expired, then another 151 seconds later (!) we started getting packets on our PASE session, as well as an ack for the StatusReport we had sent a _long_ time before.  What happened during those 200-plus seconds?

Well, going back to the TH log we have:

```
2024-04-26 05:36:46,946 - chip.native.CTL - INFO - __init__.py:_RedirectToPythonLogging:40 - Sending read request for commissioning information
2024-04-26 05:36:46,946 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:42 - SendReadRequest ReadClient[0x299d109f0]: Sending Read Request
...
2024-04-26 05:36:46,948 - chip.native.ATM - INFO - __init__.py:_RedirectToPythonLogging:40 - 					"Anonymous<4>" : 
2024-04-26 05:40:18,601 - chip.native.ATM - INFO - __init__.py:_RedirectToPythonLogging:40 - 					{
...
2024-04-26 05:40:18,613 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - <<< [E:59991i S:60974 M:30600750] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[fe80::dea6:32ff:feab:25e4%en0]:5540] --- Type 0001:02 (IM:ReadRequest)
```

Note the two consecutive tracing log lines that claim timestamps that are 304s apart.

So as far as I can tell at least from the data here, the process running on the TH "just" lost the timeslice for 5 minutes or so, if the timestamps are not lying to us.  This seems like an odd thing to have happen.  If it's running on a laptop, I would guess "laptop went to sleep and then got woken up".  That doesn't seem like something a raspi would do, though....

bzbarsky-apple, 2024-07-18:
@cjandhyala can you please look into what's going on with the TH here?

bzbarsky-apple, 2024-07-18:
> That doesn't seem like something a raspi would do, though....

Yeah, so.  Issue says "controller - python controller running on raspi," but the TH log says:

```
2024-04-26 05:36:46,897 - chip.native.DIS - ERROR - __init__.py:_RedirectToPythonLogging:38 - Mdns: Browse failure (src/platform/Darwin/DnssdImpl.cpp:412: CHIP Error 0x00000074: The operation has been cancelled)
```

so this is probably not so much a raspi.  And if it's an actual laptop, then who knows what it did in terms of sleep.