project-chip/connectedhomeip #33994: [SDK Quality Test] #3: commissiong failed with src/controller/python/ChipDeviceController-ScriptPairingDeviceDiscoveryDelegate.h:59: CHIP Error 0x00000032: Timeout

Saravana-kr22, 2024-06-18:
**Description:**
I have used python controller to run a stress test on my RPI sample dut (all-clusters-app) on Pair-Unpair test scenario for 10,000 iterations. The test passed in iteration 6126 without any error but the iteration 6127is Failed to establish secure session to device during commission the dut with the below error

> 2024-04-27 09:39:17,885 - root - WARNING - ChipDeviceCtrl.py:HandlePASEEstablishmentComplete:307 - Failed to establish secure session to device: src/controller/python/ChipDeviceController-ScriptPairingDeviceDiscoveryDelegate.h:59: CHIP Error 0x00000032: Timeout
> 2024-04-27 09:39:17,885 - root - WARNING - ChipDeviceCtrl.py:HandleCommissioningComplete:271 - Failed to commission: src/controller/python/ChipDeviceController-ScriptPairingDeviceDiscoveryDelegate.h:59: CHIP Error 0x00000032: Timeout

GitHub hash of the SDK that was being used: https://github.com/project-chip/connectedhomeip/commit/3718e996b56f36a553c1b4571f77728396590776

**Platform**
controller - python controller running on raspi,
DUT - all-cluster-app running on raspi

Provisioned using on-network

**Platform Version(s)**
RPI-4, 8GB RAM

**Test Scenario**

Pair/Provision the DUT (All-cluster-app running on raspi) using on-network
Read device information after the device is provisioned
sleep for 2 secs
Unpair the DUT (i.e. removing the fabric)
factory reset the DUT
repeat step1 - 6 for 10,000 iterations
Script reference: https://github.com/CHIP-Specifications/matter-qa/blob/main/src/matter_qa/scripts/reliability_scripts/TC_Pair.py

**Reference log:**
Example of Passed iteration details:

The iteration number 6126 is passed, below log files for the same
TH log: [log_2024-04-27_09-31-00.log](https://github.com/user-attachments/files/15641015/log_2024-04-27_09-31-00.log)
DUT log:[Dut_log_6126_2024-04-27T09_31_10_889913.log](https://github.com/user-attachments/files/15641109/Dut_log_6126_2024-04-27T09_31_10_889913.log)

Example of Failed iteration details:

The iteration number 6127 is failed with this error below the logfile
TH log : [log_2024-04-27_09-31-26.log](https://github.com/user-attachments/files/15640913/log_2024-04-27_09-31-26.log)
DUT log: [Dut_log_6127_2024-04-27T09_39_18_048163.log](https://github.com/user-attachments/files/15640934/Dut_log_6127_2024-04-27T09_39_18_048163.log)

For the detailed summary please reference this json: [summary.json](https://github.com/user-attachments/files/15588197/summary.json)

bzbarsky-apple, 2024-07-18:
>  controller - python controller running on raspi,

TH log says:
```
2024-04-27 09:31:26,433 - chip.native.DIS - ERROR - __init__.py:_RedirectToPythonLogging:38 - Mdns: Browse failure (src/platform/Darwin/DnssdImpl.cpp:412: CHIP Error 0x00000074: The operation has been cancelled)
```

so this is not a raspi.  It also says, two consecutive log lines:

```
2024-04-27 09:31:26,444 - chip.native.DIS - INFO - __init__.py:_RedirectToPythonLogging:40 - Mdns : OnNewInterface hostname:DCA632AB25E50000.local. fullname:808B416A5901702B._matterc._udp.local. interface: 13 port: 5540 TXT:"VP=65521+32769,DT=65535,D=3790,CM=1,PH=33,PI="
2024-04-27 09:39:17,880 - chip.native.DIS - INFO - __init__.py:_RedirectToPythonLogging:40 - Mdns: OnResolve flags: 0, interface: 6, fullname: 808B416A5901702B._matterc._udp.local., hostname: DCA632AB25E50000.local., port: 5540
```

so we lost the timeslice for almost 8 minutes there.  And then of course discovery timed out.

Same thing as https://github.com/project-chip/connectedhomeip/issues/33993

bzbarsky-apple, 2024-07-19:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/33993

closed 2024-07-19 by bzbarsky-apple