project-chip/connectedhomeip #33995: [SDK Quality Test] #5: Commissioning failed with src/controller/python/ChipDeviceController-ScriptDevicePairingDelegate.cpp:89: CHIP Error 0x00000003: Incorrect state

Saravana-kr22, 2024-06-18:
**Description:**
I have used python controller to run a stress test on my RPI sample dut (all-clusters-app) on Pair-Unpair test scenario for 10,000 iterations. The test passed in iteration 3732 without any error but the iteration 3733 is Failed to establish secure session to device during commission the dut with the below traceback error 

> Traceback (most recent call last):
> File "/Users/chaitanyajandhyala/code/python/matter-qa/src/matter_qa/library/helper_libs/matter_testing_support.py", line 1177, in pair_dut
> raise CommissionFailure(f"Pairing with DUT failed Reason {commission_device_result.str()} ")
> chip.commissioning.CommissionFailure: CommissionFailure(Pairing with DUT failed Reason src/controller/python/ChipDeviceController-ScriptDevicePairingDelegate.cpp:89: CHIP Error 0x00000003: Incorrect state )

GitHub hash of the SDK that was being used: https://github.com/project-chip/connectedhomeip/commit/3718e996b56f36a553c1b4571f77728396590776

**Platform**
controller - python controller running on raspi,
DUT - all-cluster-app running on raspi

Provisioned using on-network

**Platform Version(s)**
RPI-4, 8GB RAM

**Test Scenario**

Pair/Provision the DUT (All-cluster-app running on raspi) using on-network
Read device information after the device is provisioned
sleep for 2 secs
Unpair the DUT (i.e. removing the fabric)
factory reset the DUT
repeat step1 - 6 for 10,000 iterations
Script reference: https://github.com/CHIP-Specifications/matter-qa/blob/main/src/matter_qa/scripts/reliability_scripts/TC_Pair.py

**Reference log:**
Example of Passed iteration details:

The iteration number 3732 is passed, below log files for the same
TH log: [log_2024-04-26_12-34-20.log](https://github.com/user-attachments/files/15589758/log_2024-04-26_12-34-20.log)
DUT log: [Dut_log_3732_2024-04-26T12_34_30_676217.log](https://github.com/user-attachments/files/15589751/Dut_log_3732_2024-04-26T12_34_30_676217.log)

Example of Failed iteration details:

The iteration number 3733 is failed with this error below the logfile
TH log : [log_2024-04-26_12-34-44.log](https://github.com/user-attachments/files/15589721/log_2024-04-26_12-34-44.log)
DUT log: [Dut_log_3733_2024-04-26T12_39_18_541072.log](https://github.com/user-attachments/files/15589740/Dut_log_3733_2024-04-26T12_39_18_541072.log)

For the detailed summary please reference this json: [summary.json](https://github.com/user-attachments/files/15588197/summary.json)

cecille, 2024-07-15:
if you look in the TH log, the PBKDF request is sent and then almost immediately times out

```
2024-04-26 12:39:18,382 - chip.native.SC - DEBUG - __init__.py:_RedirectToPythonLogging:42 - Sent PBKDF param request
2024-04-26 12:39:18,383 - chip.native.CTL - INFO - __init__.py:_RedirectToPythonLogging:40 - Setting attestation nonce to random value
2024-04-26 12:39:18,383 - chip.native.CTL - INFO - __init__.py:_RedirectToPythonLogging:40 - Setting CSR nonce to random value
2024-04-26 12:39:18,383 - chip.native.CTL - INFO - __init__.py:_RedirectToPythonLogging:40 - Commission called for node ID 0x0000000012344321
2024-04-26 12:39:18,384 - chip.native.CTL - ERROR - __init__.py:_RedirectToPythonLogging:38 - Unknown filter type; all matches will fail
2024-04-26 12:39:18,384 - chip.native.SC - ERROR - __init__.py:_RedirectToPythonLogging:38 - PASESession timed out while waiting for a response from the peer. Expected message type was 33
2024-04-26 12:39:18,384 - chip.native.IN - DEBUG - __init__.py:_RedirectToPythonLogging:42 - SecureSession[0x2ab0cd870]: Released - Type:1 LSID:62902
2024-04-26 12:39:18,385 - chip.native.ZCL - ERROR - __init__.py:_RedirectToPythonLogging:38 - Secure Pairing Failed
```

It then later receives the PBKDF response and drops it as unexpected
```
2024-04-26 12:39:18,435 - chip.native.EM - ERROR - __init__.py:_RedirectToPythonLogging:38 - Dropping unexpected message of type 0x21 with protocolId (0, 0) and MessageCounter:141543206 on exchange 11771i
```

NOTE: 0x21 == 33

bzbarsky-apple, 2024-07-18:
Looking at one more previous line on the TH side:

```
2024-04-26 12:34:44,562 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - <<< [E:11771i S:0 M:183271843] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[fe80::dea6:32ff:feab:25e5%en1]:5540] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
2024-04-26 12:39:18,382 - chip.native.SC - DEBUG - __init__.py:_RedirectToPythonLogging:42 - Sent PBKDF param request
```

Note that almost-5-minute (!) gap between when we log that we sent the packet and when we log that we returned from the function that sent the packet.  That's why it immediately times out: the timeout time was set before that "Msg TX" log line, and 274s is longer than that timeout.

This looks like the same problem as https://github.com/project-chip/connectedhomeip/issues/33993 and https://github.com/project-chip/connectedhomeip/issues/33997

Saravana-kr22, 2024-07-19:
@bzbarsky-apple The below are the logs of the last week run on commit: d033f8b917faa17b82a432e0c895f96199217dcb

TH-log:  [log_2024-07-12_10-37-53.log](https://github.com/user-attachments/files/16304631/log_2024-07-12_10-37-53.log)
DUT-log: [Dut_log_197_2024-07-12T10_38_44_407316.log](https://github.com/user-attachments/files/16304675/Dut_log_197_2024-07-12T10_38_44_407316.log)

**Platform**
controller - python controller running on raspi,
DUT - all-cluster-app running on raspi

Provisioned using Ble-wifi

**Platform Version(s)**
RPI-4, 8GB RAM

TCPdump file: [tcpdump.zip](https://github.com/user-attachments/files/16304793/197.zip)

bzbarsky-apple, 2024-07-19:
@Saravana-kr22 so that's different TH hardware and different commissioning method than the original issue here, right?  Seems like it should be a separate issue, because it's got a completely different thing going on.  In particular, the BLE connection is not working right in that log.

bzbarsky-apple, 2024-07-19:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/33993

closed 2024-07-19 by bzbarsky-apple