project-chip/connectedhomeip #33996: [SDK Quality Test] #6: Unpairing function is failed with src/app/ReadClient.cpp:682: CHIP Error 0x00000032: Timeout

Saravana-kr22, 2024-06-18:
**Description:**
I have used python controller to run a stress test on my RPI sample dut (all-clusters-app) on Pair-Unpair test scenario for 10,000 iterations. The test passed in iteration 5688 without any error but the iteration 5689 is failed to Unpair the dut during the OnReadAttribute step for the currentFabricIndex attribute with the below traceback error

> Traceback (most recent call last):
> File "/Users/chaitanyajandhyala/code/python/matter-qa/src/matter_qa/library/helper_libs/matter_testing_support.py", line 1187, in unpair_dut
> controller.UnpairDevice(node_id)
> File "/Users/chaitanyajandhyala/code/python/matter-qa/.venv/lib/python3.9/site-packages/chip/ChipDeviceCtrl.py", line 452, in UnpairDevice
> return self._ChipStack.CallAsync(
> File "/Users/chaitanyajandhyala/code/python/matter-qa/.venv/lib/python3.9/site-packages/chip/native/init.py", line 87, in raise_on_error
> raise self.to_exception()
> chip.exceptions.ChipStackError: src/app/ReadClient.cpp:682: CHIP Error 0x00000032: Timeout)

GitHub hash of the SDK that was being used: https://github.com/project-chip/connectedhomeip/commit/3718e996b56f36a553c1b4571f77728396590776

**Platform**
controller - python controller running on raspi,
DUT - all-cluster-app running on raspi

Provisioned using on-network

**Platform Version(s)**
RPI-4, 8GB RAM

**Test Scenario**

Pair/Provision the DUT (All-cluster-app running on raspi) using on-network
Read device information after the device is provisioned
sleep for 2 secs
Unpair the DUT (i.e. removing the fabric)
factory reset the DUT
repeat step1 - 6 for 10,000 iterations
Script reference: https://github.com/CHIP-Specifications/matter-qa/blob/main/src/matter_qa/scripts/reliability_scripts/TC_Pair.py

**Reference log:**
Example of Passed iteration details:

The iteration number 5688 is passed, below log files for the same
TH log: [log_2024-04-27_04-43-23.log](https://github.com/user-attachments/files/15589034/log_2024-04-27_04-43-23.log)
DUT log: [Dut_log_5688_2024-04-27T04_43_33_846904.log](https://github.com/user-attachments/files/15589021/Dut_log_5688_2024-04-27T04_43_33_846904.log)

Example of Failed iteration details:

The iteration number 5689 is failed with this error below the logfile
TH log : [log_2024-04-27_04-43-44.log](https://github.com/user-attachments/files/15588965/log_2024-04-27_04-43-44.log)
DUT log: [Dut_log_5689_2024-04-27T04_48_18_251279.log](https://github.com/user-attachments/files/15588979/Dut_log_5689_2024-04-27T04_48_18_251279.log)

For the detailed summary please reference this json: [summary.json](https://github.com/user-attachments/files/15588197/summary.json)

bzbarsky-apple, 2024-07-18:
>  controller - python controller running on raspi,

TH log says:

```
2024-04-27 04:43:44,477 - chip.native.DIS - ERROR - __init__.py:_RedirectToPythonLogging:38 - Mdns: Browse failure (src/platform/Darwin/DnssdImpl.cpp:412: CHIP Error 0x00000074: The operation has been cancelled)
```

so this is not a raspi.  The log also says:

```
2024-04-27 04:43:48,353 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:42 - SendReadRequest ReadClient[0x2df35b480]: Sending Read Request
...
2024-04-27 04:43:48,354 - chip.native.ATM - INFO - __init__.py:_RedirectToPythonLogging:40 - 	"payload" : 
2024-04-27 04:48:18,065 - chip.native.ATM - INFO - __init__.py:_RedirectToPythonLogging:40 - 	{
```

so we once again lost timesliace in mid-trace for about 4.5 minutes. Just like https://github.com/project-chip/connectedhomeip/issues/33993
```

bzbarsky-apple, 2024-07-19:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/33993

closed 2024-07-19 by bzbarsky-apple