project-chip/connectedhomeip #33997: [SDK Quality Test] #7: Commissioning failed with src/app/CommandSender.cpp:328: CHIP Error 0x00000032: Timeout

Saravana-kr22, 2024-06-18:
**Description:**
I have used python controller to run a stress test on my RPI sample dut (all-clusters-app) on Pair-Unpair test scenario for 10,000 iterations. The test passed in iteration 2570 without any error but the iteration 2571 is failed to commission the dut during the SendNOC step with the below traceback error

> Traceback (most recent call last):
> File "/Users/chaitanyajandhyala/code/python/matter-qa/src/matter_qa/library/helper_libs/matter_testing_support.py", line 1177, in pair_dut
> raise CommissionFailure(f"Pairing with DUT failed Reason {commission_device_result.str()} ")
> chip.commissioning.CommissionFailure: CommissionFailure(Pairing with DUT failed Reason src/app/CommandSender.cpp:328: CHIP Error 0x00000032: 

GitHub hash of the SDK that was being used: https://github.com/project-chip/connectedhomeip/commit/3718e996b56f36a553c1b4571f77728396590776

**Platform**
controller - python controller running on raspi,
DUT - all-cluster-app running on raspi

Provisioned using on-network

Platform Version(s)
RPI-4, 8GB RAM

**Test Scenario**

Pair/Provision the DUT (All-cluster-app running on raspi) using on-network
Read device information after the device is provisioned
sleep for 2 secs
Unpair the DUT (i.e. removing the fabric)
factory reset the DUT
repeat step1 - 6 for 10,000 iterations
Script reference: https://github.com/CHIP-Specifications/matter-qa/blob/main/src/matter_qa/scripts/reliability_scripts/TC_Pair.py

**Reference log:**
Example of Passed iteration details:

The iteration number 2570 is passed, below log files for the same
TH log: [log_2024-04-26_04-12-01.log](https://github.com/user-attachments/files/15588137/log_2024-04-26_04-12-01.log)
DUT log: [Dut_log_2570_2024-04-26T04_12_11_598075.log](https://github.com/user-attachments/files/15588147/Dut_log_2570_2024-04-26T04_12_11_598075.log)

Example of Failed iteration details:

The iteration number 2571 is failed with this error below the logfile
TH log : [log_2024-04-26_04-12-21.log](https://github.com/user-attachments/files/15588107/log_2024-04-26_04-12-21.log)
DUT log: [Dut_log_2571_2024-04-26T04_17_27_934283.log](https://github.com/user-attachments/files/15588118/Dut_log_2571_2024-04-26T04_17_27_934283.log)

For the detailed summary please reference this json: [summary.json](https://github.com/user-attachments/files/15588197/summary.json)

bzbarsky-apple, 2024-07-18:
From the DUT log:

```
[1714129942.591993][489230:489230] CHIP:EM: >>> [E:56552r S:51405 M:78484474] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
...
[1714129942.593493][489230:489230] CHIP:ZCL: OpCreds: Received an AddNOC command
[1714129942.593713][489230:489230] CHIP:EM: <<< [E:56552r S:51405 M:44984139 (Ack:78484474)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[fe80::1887:8597:775f:fcc8%wlan0]:63338] --- Type 0000:10 (SecureChannel:StandaloneAck)
...
[1714129942.613441][489230:489230] CHIP:EM: <<< [E:56552r S:51405 M:44984140 (Ack:78484474)] (S) Msg TX to 1:FFFFFFFB00000000 [03F8] [UDP:[fe80::1887:8597:775f:fcc8%wlan0]:63338] --- Type 0001:09 (IM:InvokeCommandResponse)
...
[1714129942.960479][489230:489230] CHIP:EM: Retransmitting MessageCounter:44984140 on exchange 56552r Send Cnt 1
[1714129943.368354][489230:489230] CHIP:EM: Retransmitting MessageCounter:44984140 on exchange 56552r Send Cnt 2
[1714129943.484887][489230:489230] CHIP:DL: Avahi group established
[1714129943.975605][489230:489230] CHIP:EM: Retransmitting MessageCounter:44984140 on exchange 56552r Send Cnt 3
[1714129944.944039][489230:489230] CHIP:EM: Retransmitting MessageCounter:44984140 on exchange 56552r Send Cnt 4
[1714129946.426007][489230:489230] CHIP:EM: Failed to Send CHIP MessageCounter:44984140 on exchange 56552r sendCount: 4 max retries: 4
[1714130002.110713][489230:489230] CHIP:FS: Fail-safe timer expired
```

So the DUT got an invoke request, sent an ack, then tried to send an invoke response almost immediately, heard nothing for 55 more seconds, and then the fail-safe expired.

On the TH side, the corresponding bit looks like this:

```
2024-04-26 04:12:22,509 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - <<< [E:56552i S:60591 M:78484474] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[fe80::dea6:32ff:feab:25e5%en1]:5540] --- Type 0001:08 (IM:InvokeCommandRequest)
...
2024-04-26 04:12:22,524 - chip.native.EM - INFO - __init__.py:_RedirectToPythonLogging:40 - >>> [E:56552i S:60591 M:44984139 (Ack:78484474)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
2024-04-26 04:17:18,712 - chip.native.EM - DEBUG - __init__.py:_RedirectToPythonLogging:42 - Found matching exchange: 56552i, Delegate: 0x2957abbc8
2024-04-26 04:17:18,713 - chip.native.EM - DEBUG - __init__.py:_RedirectToPythonLogging:42 - Rxd Ack; Removing MessageCounter:78484474 from Retrans Table on exchange 56552i
2024-04-26 04:17:18,713 - chip.native.DMG - INFO - __init__.py:_RedirectToPythonLogging:40 - Time out! failed to receive invoke command response from Exchange: 56552i
2024-04-26 04:17:18,713 - chip.native.CTL - INFO - __init__.py:_RedirectToPythonLogging:40 - Device failed to receive the operational certificate Response: src/app/CommandSender.cpp:328: CHIP Error 0x00000032: Timeout
```

Why is there a nearly 5-minute gap between the `Found matching exchange:` log line and the `Rxd Ack` log line that is logged as soon as the incoming message is dispatched to that exchange?

Looks to me like TH lost the timeslice for 5 minutes there.  Just like in https://github.com/project-chip/connectedhomeip/issues/33993

bzbarsky-apple, 2024-07-18:
@cjandhyala Why the 5-minute gap between things that should happen milliseconds apart on the TH side?

bzbarsky-apple, 2024-07-19:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/33993

closed 2024-07-19 by bzbarsky-apple