project-chip/connectedhomeip #34008: [BUG] Bug found by bugprone-casting-through-void check in .clang-tidy

jlatusek, 2024-06-19:
### Reproduction steps

1. Add `bugprone-casting-through-void` to `.clang-tidy`
2. Open latest build container based on ubuntu:24.04
3. Run `Clang-tidy validation` and previous step from [build.yaml](https://github.com/project-chip/connectedhomeip/blob/master/.github/workflows/build.yaml)

### Bug prevalence

always

### GitHub hash of the SDK that was being used

4cdce52a5fa25542d4fbeb3cdff6e75b3aa1c379

### Platform

other

### Anything else?

Log file with compilation error:
[1_Build on Linux (fake, gcc_release, clang, simulated).txt](https://github.com/user-attachments/files/15898096/1_Build.on.Linux.fake.gcc_release.clang.simulated.txt)

bzbarsky-apple, 2024-06-20:
@jlatusek Those are all from using the GLib API macros, right?  Sounds like we need to add suppressions for those cases or something.

jlatusek, 2024-06-26:
@bzbarsky-apple This error is caused by using glib macros to do the casting. We can try to do the casting another way, but glib macros also check that the object being casted is of the correct type.

bzbarsky-apple, 2024-06-26:
Right, so it sounds like we need to suppress the clang-tidy check for these known-safe cases....

andy31415, 2024-10-21:
We should add supression comments or fix the code.
At the same time, I believe we updated to ubuntu 24.04 and evetyhing should be updated now.

Closing as likely stale - we support ubuntu 24.04 today.

closed 2024-10-21 by andy31415